NVIDIA/open-gpu-kernel-modules #187: Electron applications do not redraw properly to reflect latest input (reuse old frames)

Gert-dev, 2022-05-14:
**NVIDIA Driver Version**
515.43.04

**GPU**
NVIDIA RTX 2070 Super Max-Q

**Describe the bug**
Electron apps such as Visual Studio Code or the Tidal Hi-Fi application regularly reuse old frames or their visual contents seem to "lag behind" the actual state of the application. This only happens on Wayland (possibly only with GNOME, I'm not sure), not on X11.

**To Reproduce**
1. Start a GNOME 42.1 Wayland session (using GBM).
2. Start Visual Studio Code (tested with 1.67).
3. Open the terminal, type some text, such as `This is a very long command`.
4. Use the left arrow key to move the cursor in the terminal back, one position at a time.
5. Notice how often the cursor temporarily quickly jumps back to the right a fraction of a second. Sometimes this also happens when typing input; the last-typed character is not immediately reflected until the window must redraw due to another event, such as typing another character.

Alternatively:

1. Start a GNOME 42.1 Wayland session (using GBM).
2. Start the [Tidal Hi-Fi application](https://github.com/Mastermindzh/tidal-hifi).
3. Type something in the search bar above.
4. Notice the contents of the search bar lagging behind what has actually been typed sometimes.

This bug is also present in the proprietary drivers, and has been since at least GBM support was added (November 2021, I believe), and IIRC even when only EGLStreams was present. It only seems to happen with various Electron-based applications.

**Expected behavior**
There is no lag in redrawing and input is immediately properly reflected in the application.

**Please reproduce the problem, run nvidia-bug-report.sh, and attach the resulting nvidia-bug-report.log.gz.**

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8690924/nvidia-bug-report.log.gz)

iczero, 2022-05-15:
See related: https://gitlab.freedesktop.org/xorg/xserver/-/issues/1317

nonetrix, 2022-06-30:
This also seems to happen in Anki with it using QTWebEngine to display cards although the issue seems slightly different hovering over the QTWebEngine widget seems to force it to show the new frame also just waiting will do the same

erik-kz, 2022-07-12:
> See related: https://gitlab.freedesktop.org/xorg/xserver/-/issues/1317

Sigh, yes, this does seem to be yet another instance of the general Xwayland synchronization problem. I'm prioritizing trying to get this fixed. As mentioned in my last comment on the above issue, I intend to propose a patch adding the necessary infrastructure to Xwayland.

For the time being, it may be possible to work around the issue by setting __GL_MaxFramesAllowed=1. For graphics-intensive applications like games this can have a significant performance impact (30% to 50% lower FPS in some cases), but for something like VS Code or Anki that's probably not a huge issue.

Gert-dev, 2022-07-14:
> For the time being, it may be possible to work around the issue by setting __GL_MaxFramesAllowed=1. For graphics-intensive applications like games this can have a significant performance impact (30% [...]

FWIW I tried this from a terminal, but it had no effect for me. VSCode does launch its own process outside the terminal, so perhaps the environment variables aren't properly being passed on.

Bananaman, 2022-08-07:
@erik-kz Seriously, thank you and everyone else at NVIDIA who works on Wayland. You are brilliant.

Thanks to your hard work I was able to switch to Wayland a few months ago and only encounter a few minor issues. Mainly the judder due to lack of explicit sync in XWayland apps (Discord and Vscode).

But I saw your comments from a week ago on your linked ticket, about experimenting with explicit sync in X Wayland. It is clearly the right way forward (rather than trying to hack implicit sync into the NVIDIA driver). And I appreciate that you went above and beyond to solve this the right way. That is perfection and genius!

There are only a few things with NVIDIA Wayland that still hurt. The experience overall is good but the issues I'd say are:

- Sync issues, as mentioned here. They are pretty severe but I can live with it for now.
- Lack of support for the Wayland color temperature setting API, so color profiles for color correction and night light are not possible.
- The Gamescope compositor still crashes on NVIDIA even with the latest NVIDIA driver and Gamescope version.

That list is honestly it. It's a small list. Gaming in general works very well, and so do most applications.

And considering that you guys achieved thisucb Wayland compatibility in such a short time after you began to implement GBM is stunning.

Clearly NVIDIA hires absolute geniuses!

I'll keep using Wayland since the upsides outweigh the downsides. And I look forward to the results of your awesome developments in the future! 😊❤️

darlanjunior, 2022-12-13:
> > > See related: https://gitlab.freedesktop.org/xorg/xserver/-/issues/1317
> > 
> > 
> > Sigh, yes, this does seem to be yet another instance of the general Xwayland synchronization problem. I'm prioritizing trying to get this fixed. As mentioned in my last comment on the above issue, I intend to propose a patch adding the necessary infrastructure to Xwayland.
> > For the time being, it may be possible to work around the issue by setting __GL_MaxFramesAllowed=1. For graphics-intensive applications like games this can have a significant performance impact (30% to 50% lower FPS in some cases), but for something like VS Code or Anki that's probably not a huge issue.
> 
> Only thing purely NVIDIA related(to my knowledge at least) stopping me from using Wayland keep up the great work
> 
> Only other issues I haven't tested on non NVIDIA cards so I don't feel like making a issue, if anyone has AMD or Intel hardware and is willing to test might be worth looking into these
> 
> * Android studio emulator doesn't start
> * Krita image canvas will not render
> * KDenLive color picker will not let you pick a color only returns dark blue
> * weird shadow under mouse cursor that gets darker over time under cursor in KDE when hovering on blurred taskbar
> * Alacritty on GNOME title bar sometimes glitches out when moving window
> * Hyprland default wallpaper looks strange moving the mouse will fix the area around it in a box shape
> * KDE start menu under `right click -> show alternatives -> application menu` doesn't change category sometimes when changing focus to another category or when it does sometimes menu will be half blurred and half transparent
> * Chrome fails to use hardware compositor when using Wayland natively works on xwayland

About Android Studio Emulator, once I switched to SwiftShader it started working. 
[image]

mhalano, 2022-12-29:
I saw this with my note application Obsidian that also uses Electron platform. When I'm typing something fast (and I'm a fast typer), there are strange blinks and lags.

benlc, 2023-03-16:
Definitely happening in several Electron apps with no known workaround from what I can tell:
- Obsidian
- Notion
- Slack

emko, 2023-07-02:
any work around ? other then switching to x11 ?

mhalano, 2023-07-24:
Yeah, I have this a lot on Slack.

skyne98, 2023-09-24:
It's very annoying 💀 Happens in Godot too!

z411, 2023-09-27:
This most probably has to do with Xwayland:

https://gitlab.freedesktop.org/xorg/xserver/-/issues/1317

You see it the most in Electron apps because they use Xwayland by default, not native Wayland. I suggest forcing your Electron apps to use native Wayland by adding the following parameters to them at launch:

`--enable-features=UseOzonePlatform --ozone-platform=wayland`

gremo, 2023-09-29:
Same problem here on Fedora 38, Nvidia RTX 4070, in some apps: Visual Studio Code, Mattermost, Obsidian. I think they are all electron apps.

CrabbyDisk, 2024-02-16:
I have the same issue happening in Discord. Nvidia RTX 3060 Ti driver: 545.29.06 kernel: 6.7.4-zen1-1-zen

cwegener, 2024-02-16:
You can help testing the initial implementation of explicit sync https://gitlab.freedesktop.org/xorg/xserver/-/merge_requests/967

If you are on Arch, this AUR package makes installing the patch easy https://aur.archlinux.org/packages/xorg-xwayland-explicit-sync-git

erik-kz, 2024-02-16:
> You can help testing the initial implementation of explicit sync https://gitlab.freedesktop.org/xorg/xserver/-/merge_requests/967
> 
> If you are on Arch, this AUR package makes installing the patch easy https://aur.archlinux.org/packages/xorg-xwayland-explicit-sync-git

Note that to fully solve the problem, the above patch will need to be accompanied by an upcoming NVIDIA driver release. The patch by itself __might__ help in some cases, but corruption is still possible.

nonetrix, 2024-02-17:
Good to see this being added finally, puts NVIDIA cards back on the table for me perhaps next time

Flameopathic, 2024-06-06:
> I have the same issue happening in Discord. Nvidia RTX 3060 Ti driver: 545.29.06 kernel: 6.7.4-zen1-1-zen

Turning off hardware acceleration in Discord worked for me, but I'm on the 555 beta now. Issue persists in other electron apps despite 555 beta. Consider trying it.

nonetrix, 2024-06-06:
Yes desktops need to support explicit sync, this can be closed. Everything is fixed on NVIDIA end, just need to wait for desktops. I think for now GNOME has a MR for it, not sure state of KDE etc. I know it's planned for Hyprland once Vaxry rewrites all of the wlroots implementations and replaces them with Hyprland ones

xuanruiqi, 2024-06-06:
> Yes desktops need to support explicit sync, this can be closed. Everything is fixed on NVIDIA end, just need to wait for desktops. I think for now GNOME has a MR for it, not sure state of KDE etc. I [...]

GNOME does not have a MR for it; it already has it. Explicit sync support is available in the latest stable version of Mutter (>= 46.1). You also need XWayland >= 24.1.

I don't know for sure about KDE, but it seems that explicit sync support has been added in KWin 6.1 beta, and the stable release is coming soon, too. So, you either already have it (or almost), of course assuming you are using the latest version of software.

Gert-dev, 2024-08-02:
I don't have this problem any more now that explicit sync is in since GNOME 46.1 :tada: . Closing this.

closed 2024-08-02 by Gert-dev

nonetrix, 2024-08-02:
Hyprland you can enable it in experimental option on git branch