NVIDIA/open-gpu-kernel-modules #188: Failed to install on Azure VM.

xkszltl, 2022-05-14:
**NVIDIA Driver Version**

515.43.04

**GPU**

V100, but on Azure. AFAIK their hypervisor doesn't support GSP.

**Describe the bug**

We're having trouble installing 515 driver via DKMS.
510 was fine.

```
2022-05-14T00:00:57.8274090Z Setting up nvidia-settings (515.43.04-1) ...
2022-05-14T00:00:57.8292897Z Setting up nvidia-smi (515.43.04-1) ...
2022-05-14T00:00:57.8310522Z Setting up libgles-nvidia2:amd64 (515.43.04-1) ...
2022-05-14T00:00:57.8333871Z Setting up nvidia-driver-bin (515.43.04-1) ...
2022-05-14T00:00:57.8363352Z Setting up libnvcuvid1:amd64 (515.43.04-1) ...
2022-05-14T00:00:57.8381486Z Setting up nvidia-persistenced (515.43.04-1) ...
2022-05-14T00:00:57.9140961Z Warning: The home dir /var/run/nvpd/ you specified can't be accessed: No such file or directory
2022-05-14T00:00:57.9145946Z Adding system user `nvpd' (UID 108) ...
2022-05-14T00:00:57.9148403Z Adding new group `nvpd' (GID 114) ...
2022-05-14T00:00:57.9404618Z Adding new user `nvpd' (UID 108) with group `nvpd' ...
2022-05-14T00:00:58.0063595Z Not creating home directory `/var/run/nvpd/'.
2022-05-14T00:00:58.2563485Z Created symlink /etc/systemd/system/multi-user.target.wants/nvidia-persistenced.service → /lib/systemd/system/nvidia-persistenced.service.
2022-05-14T00:00:58.5343308Z Job for nvidia-persistenced.service failed because the control process exited with error code.
2022-05-14T00:00:58.5345041Z See "systemctl status nvidia-persistenced.service" and "journalctl -xe" for details.
2022-05-14T00:00:58.5364124Z Setting up libnvidia-opticalflow1:amd64 (515.43.04-1) ...
2022-05-14T00:00:58.5384589Z Setting up nvidia-egl-icd:amd64 (515.43.04-1) ...
2022-05-14T00:00:58.5427294Z Setting up libnvidia-encode1:amd64 (515.43.04-1) ...
2022-05-14T00:00:58.5446958Z Setting up nvidia-driver-libs:amd64 (515.43.04-1) ...
2022-05-14T00:00:58.6457793Z Processing triggers for nvidia-alternative (515.43.04-1) ...
2022-05-14T00:00:58.6815900Z update-alternatives: updating alternative /usr/lib/nvidia/current because link group nvidia has changed slave links
2022-05-14T00:00:58.6909923Z Setting up nvidia-kernel-dkms (515.43.04-1) ...
2022-05-14T00:00:58.7524183Z Loading new nvidia-current-515.43.04 DKMS files...
2022-05-14T00:00:58.8821742Z Building for 5.10.0-14-cloud-amd64
2022-05-14T00:00:58.9464342Z Building initial module for 5.10.0-14-cloud-amd64
2022-05-14T00:01:07.5776525Z Error! Bad return status for module build on kernel: 5.10.0-14-cloud-amd64 (x86_64)
2022-05-14T00:01:07.5777316Z Consult /var/lib/dkms/nvidia-current/515.43.04/build/make.log for more information.
2022-05-14T00:01:07.5823079Z dpkg: error processing package nvidia-kernel-dkms (--configure):
2022-05-14T00:01:07.5823791Z  installed nvidia-kernel-dkms package post-installation script subprocess returned error exit status 10
2022-05-14T00:01:07.5833911Z dpkg: dependency problems prevent configuration of nvidia-driver:
2022-05-14T00:01:07.5836121Z  nvidia-driver depends on nvidia-kernel-dkms (= 515.43.04-1) | nvidia-kernel-515.43.04 | nvidia-kernel-open-dkms (= 515.43.04-1); however:
2022-05-14T00:01:07.5836902Z   Package nvidia-kernel-dkms is not configured yet.
2022-05-14T00:01:07.5837418Z   Package nvidia-kernel-515.43.04 is not installed.
2022-05-14T00:01:07.5838186Z   Package nvidia-kernel-dkms which provides nvidia-kernel-515.43.04 is not configured yet.
2022-05-14T00:01:07.5838774Z   Package nvidia-kernel-open-dkms is not installed.
2022-05-14T00:01:07.5841101Z 
2022-05-14T00:01:07.5842830Z dpkg: error processing package nvidia-driver (--configure):
2022-05-14T00:01:07.5843376Z  dependency problems - leaving unconfigured
2022-05-14T00:01:07.5844028Z dpkg: dependency problems prevent configuration of cuda-drivers-515:
2022-05-14T00:01:07.5844627Z  cuda-drivers-515 depends on nvidia-driver (>= 515.43.04); however:
2022-05-14T00:01:07.5845181Z   Package nvidia-driver is not configured yet.
2022-05-14T00:01:07.5845355Z 
2022-05-14T00:01:07.5845818Z dpkg: error processing package cuda-drivers-515 (--configure):
2022-05-14T00:01:07.5846563Z  dependency problems - leaving unconfigured
2022-05-14T00:01:07.5848749Z dpkg: dependency problems prevent configuration of cuda-drivers:
2022-05-14T00:01:07.5849400Z  cuda-drivers depends on cuda-drivers-515 (= 515.43.04-1); however:
2022-05-14T00:01:07.5849942Z   Package cuda-drivers-515 is not configured yet.
2022-05-14T00:01:07.5850126Z 
2022-05-14T00:01:07.5850560Z dpkg: error processing package cuda-drivers (--configure):
2022-05-14T00:01:07.5851083Z  dependency problems - leaving unconfigured
2022-05-14T00:01:07.5851691Z Processing triggers for libgdk-pixbuf-2.0-0:amd64 (2.42.2+dfsg-1) ...
2022-05-14T00:01:07.6020275Z Processing triggers for libc-bin (2.31-13+deb11u3) ...
2022-05-14T00:01:07.6099733Z Processing triggers for initramfs-tools (0.140) ...
2022-05-14T00:01:07.6301617Z update-initramfs: Generating /boot/initrd.img-5.10.0-14-cloud-amd64
2022-05-14T00:01:10.8598813Z Processing triggers for update-glx (1.2.1~deb11u1) ...
2022-05-14T00:01:10.8662037Z Processing triggers for glx-alternative-nvidia (1.2.1~deb11u1) ...
2022-05-14T00:01:10.9032521Z update-alternatives: using /usr/lib/nvidia to provide /usr/lib/glx (glx) in auto mode
[... 201 lines cut ...]
2022-05-14T00:01:52.9132535Z Errors were encountered while processing:
2022-05-14T00:01:52.9133011Z  nvidia-kernel-dkms
2022-05-14T00:01:52.9133369Z  nvidia-driver
2022-05-14T00:01:52.9133736Z  cuda-drivers-515
2022-05-14T00:01:52.9134085Z  cuda-drivers
2022-05-14T00:01:53.0105790Z E: Sub-process /usr/bin/dpkg returned an error code (1)
2022-05-14T00:01:53.0128828Z [31m[ERROR] Failed to install packages.[0m
2022-05-14T00:01:53.0181684Z ##[error]Bash exited with code '1'.
2022-05-14T00:01:53.0198482Z ##[section]Finishing: Install GPU driver
```

Given Azure doesn't support GSP, we had to inject `NVreg_EnableGpuFirmware=0` previously after installation of 510.
```
sudo modprobe -r nvidia{,_{modeset,uvm}} || true
sudo modprobe nvidia{,_{modeset,uvm}} NVreg_EnableGpuFirmware=0
```

Because based on what I heard, the new release (or at least the open source part) requires GSP.
I wonder if that's causing the issue.

**To Reproduce**

- Get a Azure VM with T4 (NCasT4v3 SKU) + Debian 11.
- Set up nvidia repo.
- `apt-get install cuda-drivers`.

PAR2020, 2022-06-08:
@xkszltl, can you please try this again with the new driver that was dropped and report back?
I believe Azure has completed their roll-out of the new 515 drivers.
Thanks.

PAR2020, 2022-07-12:
@xkszltl checking in on this issue...any luck?

xkszltl, 2022-09-22:
Sorry for the long wait, yes it works.

During the past months we had several blocking issue and that's why it took so long to confirm.
- We're using Debian 11, so have to blend in ubuntu2004's repo to use cudnn/tensorrt. But there's 510.85 on ubuntu only, but up to 510.84 on debian side. The repo-level metadata is identical between them so there's no good way for apt priority pinning to work. We had to do a lot of nasty version pinning after the release of 510.85, making driver update very hard.
- Azure hypervisor doesn't support GSP but new driver automatically use that on T4. Had to do a lot of nasty version detection and pinning there as well.

Would be great if nvidia can make that experience better in general.