NVIDIA/open-gpu-kernel-modules #194: New Feature Request: Proper TILEd display support

mkoloberdin, 2022-05-15:
**Is your feature request related to a problem? Please describe.**
Some high-resolution displays (5K, 8K) require to be connected via two DP cables to drive them in their native resolution due to DP bandwidth restrictions. These kind of displays are currently represented by the driver as TWO separate side-by-side displays to the userspace, labeled with "TILE" properties, which userspace is then supposed to join back into one display/surface. There is no good reason for this approach, since these "tiles" are parts of ONE physical display, and this externalizes the problem to userspace software, and therefore a massive effort and code duplication is expected. It's unreasonable to expect every possible piece of userspace software (x11 servers and compositors, wayland compositors, games and other full-screen software etc.) to implement each their own solution.

**Describe the solution you'd like**
This should be addressed at the driver level (drm, mode setting). The tiles of the same physical display should be transparently joined behind the scenes by the driver and only the joined display/surface should be exposed to the userspace. 

**Describe alternatives you've considered**
For X11 there is a working userspace solution:
https://github.com/CyberShadow/hax11
Here is the description of how this is handled in X11/randr in general (a bad idea for reasons described above): https://keithp.com/blogs/MST-monitors/

There does not seem to be any working solution for wayland.

**Additional context**
- https://forums.developer.nvidia.com/t/feature-request-proper-support-of-tiled-displays-4k-5k-etc-in-linux/46850
- https://gitlab.com/kwinft/kwinft/-/issues/259
- kwin (WiP): https://invent.kde.org/plasma/kwin/-/merge_requests/1174
- wlroots (WiP): https://gitlab.freedesktop.org/wlroots/wlroots/-/merge_requests/3344

mkoloberdin, 2022-05-15:
[image]

edisionnano, 2022-05-16:
> [image]

Is this with Xinerama?

mkoloberdin, 2022-05-17:
> Is this with Xinerama?

No.