NVIDIA/open-gpu-kernel-modules #198: Thank You NVIDIA <3

martinval11, 2022-05-15:
[image]

databoose, 2022-05-15:
They need to show their commits still, point by point is not what git is meant for. It also makes it hard to show contributor's work.

MartinK7, 2022-05-15:
This "issue" has been already discussed.
`https://github.com/NVIDIA/open-gpu-kernel-modules/issues/1#issuecomment-1124323144`

THEGOLDENPRO, 2022-05-15:
😂editing the middle finger, what amazing art

niv, 2022-05-16:
Thank you for your kindness and humour. Pretty good edit, but the lighting is a bit off - highlight on the back of the hand seems to come from a different angle.

7 out of 10, should probably have used RTX

closed 2022-05-16 by niv