NVIDIA/open-gpu-kernel-modules #202: Segmentation fault (core dumped) while doing prime-run

B0BAA, 2022-05-15:
**NVIDIA Driver Version**
nvidia-dkms 515.43.04 (Arch Linux)

**GPU**
GTX 1650 (TU117M)

**Describe the bug**
when using prime-run I am unable to run anything on dGPU as it shows the error /usr/bin/prime-run: line 2: 25341 Segmentation fault      (core dumped) __NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only __GLX_VENDOR_LIBRARY_NAME=nvidia "$@"

**To Reproduce**
prime-run obs(or any appication) or __NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only __GLX_VENDOR_LIBRARY_NAME=nvidia obs

**Expected behavior**
should run it on the dGPU without any errors or Segmentation fault(core dumps) everything was working yesterday but suddenly this is now happening.

amrit1711, 2022-05-16:
@B0BAA Pease confirm if you are getting Segmentation fault with Open source driver only or with Nvidia proprietary driver as well.

B0BAA, 2022-05-16:
Ive only gotten the error with the nvidia open source driver the arch repository has updated all the drivers to this version(515.43.04) I might have to manually get the proprietary drivers and then tell you

V1del, 2022-05-16:
@B0BAA the nvidia-dkms package is the closed driver, nvidia-open-dkms  would be the driver from here. That the versions happen to be matching should be irrelevant as to which source the package comes from

B0BAA, 2022-05-16:
It somehow automatically fixed after a reboot

closed 2022-05-16 by B0BAA