NVIDIA/open-gpu-kernel-modules #219: undefined reference to `nvKmsKapiGetFunctionsTable` at kernel-open/nvidia-drm/nvidia-drm.c

worthless443, 2022-05-17:
### Describe the bug
when building nvidia-drm.ko, undefined reference to `nvKmsKapiGetFunctionsTable` is encountered. Upon a bit of examination,  I see that `nvKmsKapiGetFunctionsTable` is called at nvidia-drm/nvidia-drm.c:41 `if (!nvKmsKapiGetFunctionsTable(&nvKmsFuncsTable))`, but there are no referefences of `nvKmsKapiGetFunctionsTable` within `kernel-open/nvidia-drm/`, thus this linking problem comes up with building like this 
```
make LD=ld CC=cc OBJDUMP=objdump KBUILD_OUTPUT=/lib/modules/5.15.32-gentoo-r1-x86_64/build -C /lib/modules/5.15.32-gentoo-r1-x86_64/source M=/home/aissy/c_cpp/kernel/kmod/kernel-open ARCH=x86_64 NV_KERNEL_SOURCES=/lib/modules/5.15.32-gentoo-r1-x86_64/source NV_KERNEL_OUTPUT=/lib/modules/5.15.32-gentoo-r1-x86_64/build NV_KERNEL_MODULES=nvidia-drm INSTALL_MOD_DIR=kernel/drivers/video NV_SPECTRE_V2=0 modules
```

**To Reproduce**
use the above make flags but with `NV_KERNEL_MODULES` set to `nvidia-drm`
**Expected behavior**
should exit without errors and produce a `nvidia-drm.ko` module

aritger, 2022-05-17:
Ultimately, nvidia-drm.ko depends on nvidia-modeset.ko (the latter is what defines `nvKmsKapiGetFunctionsTable`).  Similarly, nvidia-modeset.ko depends on nvidia.ko.  I don't think it is possible to surgically build individual kernel modules out of this repository without building the prerequisite kernel modules.

So, unfortunately, I don't think we can fix this.  Please let me know if I'm missing something, or if there is a particular reason this functionality is needed, and we can try to look harder for a solution.

closed 2022-05-17 by aritger

worthless443, 2022-05-18:
thanks for clarifying this.