NVIDIA/open-gpu-kernel-modules #221: Guix port

phodina, 2022-05-17:
**NVIDIA Driver Version**
Hi, 
I'm porting 515.43.04 nvidia open gpu kernel module for [Guix](https://guix.gnu.org/). However I face some issues in compiling the kernel module.

**GPU**
 NVIDIA Corporation TU117GLM

**Describe the bug**
The build fails.
First if I don't override the build phase it fails to find the `utils.mk` file as 
it does not run the make command from the root of the repo.

That links to second issue where the linux kernel version of the driver fails:

```
make -C kernel-open modules
make[1]: Entering directory 
'/tmp/guix-build-nvidia-gpu-linux-module-515.43.04.drv-0/source/kernel-open'
make[2]: Entering directory 
'/gnu/store/gcwhb3632xh588ihsddi2ss3y3v45zlz-nvidia-gpu-linux-module-515.43.04/lib/modules/5.17.5/build'make[2]:
 *** No rule to make target 'modules'. Stop.
```

**To Reproduce**
Install guix and attempt to build the package
```
From d800feb8a3d9ba1663b99ccfb2de1d620ce23f8b Mon Sep 17 00:00:00 2001
From: Petr Hodina <phodina@protonmail.com>
Date: Thu, 12 May 2022 09:45:35 +0200
Subject: [PATCH] gnu: Add nvidia-gpu-linux-module.

* gnu/packages/linux.scm (nvidia-gpu-linux-module): New variable.

diff --git a/gnu/packages/linux.scm b/gnu/packages/linux.scm
index 9bb2314ebd..ebeb59dcea 100644
--- a/gnu/packages/linux.scm
+++ b/gnu/packages/linux.scm
@@ -1345,6 +1345,56 @@ (define-public librem-ec-acpi-linux-module
 and the notification, WiFi, and Bluetooth LED.")
     (license license:gpl2)))
 
+(define-public nvidia-gpu-linux-module
+  (package
+    (name "nvidia-gpu-linux-module")
+    (version "515.43.04")
+    (home-page "https://github.com/NVIDIA/open-gpu-kernel-modules")
+    (source (origin
+              (method git-fetch)
+              (uri (git-reference
+                    (url home-page)
+                    (commit version)))
+              (file-name (git-file-name name version))
+              (sha256
+               (base32
+                "1byk8rah79h0diwvdx4nggc71f7y6fhjadjavpc3kqlrspjjpc6m"))))
+    (build-system linux-module-build-system)
+    (arguments
+     (list #:tests?
+           #f ;no tests
+           #:make-flags
+           #~(list (string-append "CC="
+                                  #$(cc-for-target)))
+           #:phases
+           #~(modify-phases %standard-phases
+               (add-after 'unpack 'fix-modules-prefix
+                 (lambda* _
+                   (substitute* "kernel-open/Makefile"
+                     (("\\$\\(shell uname -r\\)") #$(package-version
+                                                     linux-libre))
+                     (("/lib/modules/") (string-append #$output
+                                                       "/lib/modules/")))))
+               (replace 'build
+                 (lambda* (#:key (make-flags '())
+                           (parallel-build? #t) #:allow-other-keys)
+                   (mkdir-p "build")
+                   (mkdir-p (string-append #$output "/lib/modules/"
+                                           #$(package-version linux-libre)
+                                           "/build"))
+                   (apply invoke "make"
+                          (string-append "-C"
+                                         (getenv "PWD") "/source")
+                          `(,@(if parallel-build?
+                                  `("-j" ,(number->string (parallel-job-count)))
+                                  '()) ,@make-flags)))))))
+    (native-inputs (list inetutils)) ;hostname
+    (synopsis "Linux kernel module for NVIDIA GPU")
+    (description
+     "This package provides Linux kernel module for NVIDIA GPUs based on Turing
+and newer models.")
+    (license (list license:gpl2+ license:expat))))
+
 (define-public rtl8821ce-linux-module
   (let ((commit "dce62b9bbb7c76339b514153fcebb3a19bafdcc7")
         (revision "5"))
-- 
2.34.0

```

**Expected behavior**
The package builds and the module is loaded into the kernel

princesstwi, 2022-05-17:
Guix by default doesn't allow closed firmware. If you get it to build you still need a way to fetch the installer for the firmware and closed source userspace

aritger, 2022-05-17:
Thanks for your interest.  I'm afraid none of us from NVIDIA have much experience with guix to be able to provide assistance.  I'm going to convert this issue to a discussion, and hopefully others with more guix familiarity can help you.