NVIDIA/open-gpu-kernel-modules #223: Not Resuming From Sleep On Quadro T1000 Mobile

dylanmtaylor, 2022-05-18:
**NVIDIA Driver Version**
Please write the version of the NVIDIA driver you are using.
nvidia-open 515.43.04-6 package on Arch Linux 

**GPU**
Please write the particular model of NVIDIA GPU you are using.
Nvidia Quadro T1000 Mobile on a Dell Precision 5540 Mobile Workstation

**Describe the bug**
Please write a clear and concise description of what the bug is.

When my laptop runing Arch Linux with the latest 5.17 kernel goes into a sleep state (fully asleep, not just monitor off), with nvidia-open, the system does not resume from sleep no matter what combinations of keys or the power button are pressed. This requires pressing and holding power to shut off. Sometimes, the capslock key light is blinking intermittently, making me assume there may have been a kernel panic. I checked `sudo journalctl` and did not see any output at all between it entering the sleep state and the output from the next system boot

**To Reproduce**
Please write the steps to reproduce the behavior.
Install nvidia-open in a similar system configuration and allow the system to fully sleep. I am using the KDE desktop environment.

**Expected behavior**
Please write a clear and concise description of what you expected to happen.
The system should resume from sleep.

**Please reproduce the problem, run nvidia-bug-report.sh, and attach the resulting nvidia-bug-report.log.gz.**
I cannot run this _while_ the system is in a sleep state. Would it be helpful to generate this file?

amrit1711, 2022-05-18:
Thanks dylanmtaylor for reporting the issue.
Both suspend/resume and notebook support in general are not supported in the initial release and were not part of the test matrix.
But those features are coming in one of the future releases.

dylanmtaylor, 2023-02-26:
> Thanks dylanmtaylor for reporting the issue. Both suspend/resume and notebook support in general are not supported in the initial release and were not part of the test matrix. But those features are coming in one of the future releases.

Hello,

This issue still remains and is problematic for end users especially if Nvidia's open-source driver is ever going to replace the proprietary blob. Do you have an ETA on this? I imagine even on desktops, this will be a huge deal for lots of users.

Thank you.

jsrobson10, 2023-07-09:
yup. on Wayland this is extremely problematic especially, as any graphical issues doesn't just "sort itself out." literally everything breaks after resuming from suspend, which is a shame, because Wayland otherwise works so well.

je-vv, 2024-01-24:
Has this been solved?  Current release is 545.29.06, which BTW is the one used by Arch.