NVIDIA/open-gpu-kernel-modules #228: Simple DRM bug on Fedora 36

RyzenDew, 2022-05-18:
**NVIDIA Driver Version**
515.43.04

**GPU**
3070 Max-q/GTX 1070

**Describe the bug**
https://fedoraproject.org/wiki/Changes/ReplaceFbdevDrivers
These Changes have broken TTY on fedora 36 and it causes a black screen for most if not all users.
more info here https://www.reddit.com/r/Fedora/comments/ucnqet/fedora_36_nvidia_blank_ttys/

Another issue is nvidia users are also getting a black screen at boot some having to wait 5 mins for the screen to kick in.

**To Reproduce**
install any driver on fedora 36 and try to switch to TTY 

**Expected behavior**
No black screen on TTY

gabrc52, 2022-05-20:
The issue I'm experiencing is not only TTYs, but everything after boot finishes is completely black.

RyzenDew, 2022-05-21:
> The issue I'm experiencing is not only TTYs, but everything after boot finishes is completely black.
> 
> This is an example of such a user: https://www.reddit.com/r/Fedora/comments/ueq4jm/internal_laptop_screen_black_in_discrete_graphics/

Also part of the same issue

SamPingu, 2022-05-31:
Let me add this here. This is a major problem when you run into issues with your fstab or one of your disks refuse to mount for some reason. https://www.youtube.com/watch?v=SS0A-XvM8RI

RyzenDew, 2022-05-31:
Might do a video covering this as well to warn the fedora nvidia community

PAR2020, 2022-06-08:
Apparently Fedora has a fix out per below thread:
level 2
[paidhi](https://www.reddit.com/user/paidhi/)
·
[4 days ago](https://www.reddit.com/r/Fedora/comments/ueq4jm/comment/ib4so4y/?utm_source=reddit&utm_medium=web2x&context=3)
Fedora has implemented a fix with kernel package 5.17.11-300.fc36.

Also the [Rpmfusion](https://rpmfusion.org/Howto/NVIDIA) akmod-nvidia package was updated to reflect this. You will need akmod-nvidia version 3:510.68.02-2.fc36 which is now in the rpmfusion-nonfree-updates repository.

SamPingu, 2022-06-12:
Yup! Can confirm. The 5.17.11-300 along with 510.68.02-02 fixed the issue.

Evernow, 2022-06-12:
Do note the fedora kernel does not fix the issue per say, simply bypasses it.

RyzenDew, 2022-08-23:
reminder this still needs to be fixed and is very important

RyzenDew, 2022-08-23:
@YusufKhan-gamedev can you maybe work on this if possible?

YusufKhan-gamedev, 2022-08-23:
> @YusufKhan-gamedev can you maybe work on this if possible?

I *think* this might be that this might be something to do with ogk(these modules) not support fbdev emulation? My fbcon patches would probably fix the issue if thats the case(remember to set the kernel module param). I dont fully understand the problem though, im more into getting userland drivers for ogk but if I knew what the underlying issue is i would at least look into it.

RyzenDew, 2022-08-23:
The issue is @fedora screwed up and replaced the drm they where using with simpledrm some weird emulation thing and amd supports it and intel but not nvidia

YusufKhan-gamedev, 2022-08-23:
> The issue is @fedora screwed up and replaced the drm they where using with simpledrm some weird emulation thing and amd supports it and intel but not nvidia

Simpledrm is just a drm driver, so is amdgpu, nvidia_drm and i915(intel), there has to be some bug. Could  this be related to https://github.com/NVIDIA/open-gpu-kernel-modules/issues/341? In that case you may have a conflicting drm driver, perhaps it could be simpledrm? If its simpledrm you might want to blacklist it until the kernel issue it fixed. There is also probably some way you can get simplefb or vesafb on fedora, perhaps an external repo?

RyzenDew, 2022-08-24:
Nobara added a fix in kernel but it doesn’t always work and for regular fedora its just screwed the blacklist doesn’t always work 
Rpm fusion added a fix but that breaks just hoping someone adds support for it

RyzenDew, 2022-08-24:
I will try your patches

YusufKhan-gamedev, 2022-08-24:
> 

> Nobara added a fix in kernel but it doesn’t always work and for regular fedora its just screwed the blacklist doesn’t always work

Could I see the fix?

RyzenDew, 2022-08-24:
They just reverted simpledrm Sent from my iPhone

YusufKhan-gamedev, 2022-08-24:
> They just reverted simpledrm

So they put `simpledrm.modeset=0 ` in the kernel boot options? That should fix any conflicts with simpledrm...

YusufKhan-gamedev, 2022-08-24:
I *think* it could get a fix up though...

GloriousEggroll, 2022-08-24:
Hi @YusufKhan-gamedev I'm one of the maintainers of Nobara. To clarify, we disabled simpledrm in the kernel config and re-enabled fb_simple and fb_vesa, essentially reverting Fedora's changes:

# CONFIG_DRM_SIMPLEDRM is not set
CONFIG_FB_SIMPLE=y
CONFIG_FB_VESA=y

This allows us to freely swap between rpmfusion and negativo17 nvidia drivers without messing with people's boot options (other than still needing the nouveau blacklist ofc.)

YusufKhan-gamedev, 2022-08-24:
Ah that makes sense, im *guessing* simpledrm is a conflicting framebuffer that messes with nvidia-drm though.

YusufKhan-gamedev, 2022-08-24:
Either fbdev emulation is required or there is another driver trying to take control of the framebuffer, cant think of anything else.

YusufKhan-gamedev, 2022-08-24:
@RyzenDew you might wanna checkout my `drm/aperture_control` branch to see if it fixes your issue.

https://github.com/YusufKhan-gamedev/open-gpu-kernel-modules/tree/drm/aperture_control

YusufKhan-gamedev, 2022-08-25:
I can reproduce the issue

YusufKhan-gamedev, 2022-08-25:
Yeah, none of them solve it, perhaps its a duplicate of #341 ? In that case maybe try a older kernel??

Evernow, 2023-01-16:
This has once again become a problem on the 6.1 kernel on Fedora: https://bugzilla.redhat.com/show_bug.cgi?id=2161104

Apparently given the comment on the Fedora dev their goal is not to ship any fbdev, and won't block any release on the NVIDIA driver, classic Fedora behavior, as they seem to be hoping for Nouveau to improve.

cutephoton, 2023-01-23:
sigh. Been nothing but problems with the drivers since moving to the 6.x kernels. It appears fedora has stopped supporting the nvidia workaround with the latest kernels. There is a [patch in testing](https://bodhi.fedoraproject.org/updates/FEDORA-2023-47cff193ec) linked to the bug. I couldn't tell if works or what needs to be done to make it work (per some comments). Might just have to give this GPU the Office Space treatment.

cam-rod, 2023-01-23:
Fedora devs readded the kernel hack (it was accidentally dropped during an update), and the Nvidia driver works again as of `6.1.7-200.fc37.x86_64`. Hopefully the driver will be properly fixed before this happens again.

birdie-github, 2023-10-24:
This must have been fixed in 545.23.06 beta drivers.

C0rn3j, 2024-01-28:
Wasn't this fixed?

Judging by this comment https://github.com/negativo17/nvidia-driver/issues/129#issuecomment-1445376122, support was silently added in 525.60.11.