NVIDIA/open-gpu-kernel-modules #229: [Sway] The mouse cursor is not visible

nvidiaLinuxUser4, 2022-05-18:
**NVIDIA Driver Version**
495+.

**GPU**
Any.

**Describe the bug**
The mouse cursor is not visible until WLR_NO_HARDWARE=1 is set.

**To Reproduce**
Install and run sway or another wm based on wlroots.

**Expected behavior**
The mouse cursor will be visible.

amrit1711, 2022-05-19:
Thanks for reporting the issue. However after reading the issue description, it appears that it is not specific to open gpu kernel module components.
If you think otherwise, please let us know.
Else I would request you to post such issues on Linux forum using below URL or report through email "[linux-bugs@nvidia.com](mailto:linux-bugs@nvidia.com)" where more generic linux issues are addressed and worked upon.

nvidiaLinuxUser, 2022-05-19:
Ok.

closed 2022-05-19 by nvidiaLinuxUser