NVIDIA/open-gpu-kernel-modules #230: install_modules target did not work on Arch

Joshua-Ashton, 2022-05-18:
Hello!

I attempted to use the `install_modules` make target on Arch Linux, and then rebooted and the module was not added/did not work.

Using the `nvidia-open-git` PKGBUILD from the AUR and pointing it at my repo worked fine.

Perhaps I am missing something, if so, it would be nice to have more documentation in this area if so.

aritger, 2022-05-18:
Do you mean kbuild's `modules_install` target?  Does running `depmod -a` after building that target alter the behavior?  Could you provide more detail on what exactly does not work after a reboot?  What does `modprobe --verbose nvidia` print after a reboot?

Joshua-Ashton, 2022-05-27:
Oh, looks like dmesg answered my question:

```
[    3.513141] NVRM: Open nvidia.ko is only ready for use on Data Center GPUs.
[    3.513144] NVRM: To force use of Open nvidia.ko on other GPUs, see the
[    3.513145] NVRM: 'OpenRmEnableUnsupportedGpus' kernel module parameter described
[    3.513145] NVRM: in the README.
[    3.824916] NVRM: GPU 0000:04:00.0: RmInitAdapter failed! (0x63:0x0:1905)
[    3.827920] NVRM: GPU 0000:04:00.0: rm_init_adapter failed, device minor number 0
[    3.828704] [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000400] Failed to allocate NvKmsKapiDevice
[    3.829593] [drm:nv_drm_probe_devices [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000400] Failed to register device
```

Looks like the Arch package and other distros default packaging this driver default this on, which is why I was not aware of it.

This seems like an annoying non-feature, and even more-so because it doesn't print out as an error to the tty, only dmesg due to the way it's logged, unlike the other misc messages like `Failed to allocate NvKmsKapiDevice`.

Despite this error stating that this is mentioned in the README, grepping here shows that it is not actually mentioned in the README of this repo. Perhaps it is in some auto generated module README that lists all the params (that I don't know where it is). I think it would be much more useful to be mentioned in this repo's readme as well.

I'd much rather see this check removed and have made a PR here  to do so.

https://github.com/NVIDIA/open-gpu-kernel-modules/pull/266

It provides no benefit to end users or developers other than getting in the way and isn't even implemented in a way that would appear different to a broken driver to a user which is not a good look (spewing random unrelated error messages to TTY).

Multiple distros are already patching the repo and defaulting it to off, so it isn't going to stop end users from accidentally using this over the proprietary NVIDIA kernel module anyway.

mtijanic, 2022-05-27:
Hi, sorry, could you please elaborate on:

> This seems like an annoying non-feature, and even more-so because it doesn't print out as an error to the tty, only dmesg due to the way it's logged, unlike the other misc messages like `Failed to allocate NvKmsKapiDevice`.

Both of these just go to `printk()`, how is one ending up on tty and not the other? Is this a tty specific thing where it parses kernel messages of a given format?

aritger, 2022-05-27:
The README in that error message refers to the driver README; sorry that isn't clearer:
http://us.download.nvidia.com/XFree86/Linux-x86_64/515.43.04/README/kernel_open.html

@Joshua-Ashton, I completely share your feelings towards OpenRmEnableUnsupportedGpus, but this behavior was mandated from elsewhere within the company.  Hopefully, this is short-lived and over the next several driver releases, we'll get closer to feature parity, and at that point will be able to remove the option.

For better or worse, we're forced to keep this behavior for now.  I'm going to close this issue.  Sorry.

closed 2022-05-27 by aritger