NVIDIA/open-gpu-kernel-modules #246: RmInitAdapter failed!

robang74, 2022-05-19:
### NVIDIA Open GPU Kernel Modules Version

515.43.04

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Ubuntu 20.04.4 LTS

### Kernel Release

Linux siemens 5.4.0-109-generic #123-Ubuntu SMP Fri Apr 8 09:10:54 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

Unable to determine the device handle for gpu 0000:01:00.0: Not Found

### Describe the bug

The nvidia-smi fails if used in combination with the Open-Source driver but not with the proprietary. I suspect a problem with the loading of the gsp.bin. Thanks, R

### To Reproduce

```
wget https://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/NVIDIA-Linux-x86_64-515.43.04-no-compat32.run

bash ./NVIDIA-Linux-x86_64-515.43.04-no-compat32.run --no-kernel-module -q -s -Z 

git clone https://github.com/NVIDIA/open-gpu-kernel-modules

cd open-gpu-kernel-modules

BUILD_EXCLUSIVE_KERNEL=$(uname -r) IGNORE_PREEMPT_RT_PRESENCE=1 make modules -j$(nproc)
BUILD_EXCLUSIVE_KERNEL=$(uname -r) IGNORE_PREEMPT_RT_PRESENCE=1 make modules_install -j$(nproc)

depmod -a; update-initramfs -u; reboot

lsmod | grep nvidia

nvidia_drm             61440  0
nvidia_modeset       1069056  1 nvidia_drm
nvidia               4558848  1 nvidia_modeset
drm_kms_helper        278528  2 nvidia_drm,i915
drm                   622592  5 drm_kms_helper,nvidia,nvidia_drm,i915

cat /proc/driver/nvidia/version

NVRM version: NVIDIA UNIX Open Kernel Module for x86_64  515.43.04  Release Build  (root@siemens)  Tue 17 May 2022 03:46:22 PM CEST
GCC version:  gcc version 10.2.1 20210110 (Debian 10.2.1-6)

nvidia-smi --help | grep -e "^NVIDIA"

NVIDIA System Management Interface -- v515.43.04

nvidia-smi

Unable to determine the device handle for GPU 0000:01:00.0: Not Found

nvidia-debugdump --list

Found 1 NVIDIA devices
Error: nvmlDeviceGetHandleByIndex(): Not Found
FAILED to get details on GPU (0x0): Not Found
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8729227/nvidia-bug-report.log.gz)

### More Info

root@siemens:/nvidia# cat /etc/modprobe.d/nvidia-opensource-options.conf
options nvidia NVreg_EnableGpuFirmwareLogs=7
options nvidia NVreg_EnableGpuFirmware=0
options nvidia NVreg_EnablePCIeGen3=1

robang74, 2022-05-19:
With or without device driver options, still not work.

Meganium97, 2022-06-02:
I believe it is. On my system, I have a GTX 1650 mobile with max-q design (I looked through the list, it should be there because the 1660 mobiles are there but I guess it just hasn't been included yet) and I get the same error of `Unable to determine the device handle for GPU 0000:01:00.0: Not Found`.

PAR2020, 2022-06-08:
Sorry for the delayed response, @robang74. Could you try the latest drop 515.48.07?
RTX A6000 is supported, DevIDs also added for clarity in the README.md file.
Thank you.

PAR2020, 2022-06-23:
Ping @robang74...

PAR2020, 2022-07-12:
Ping @robang74...?

robang74, 2022-07-14:
> Sorry for the delayed response, @robang74. Could you try the latest drop 515.48.07? RTX A6000 is supported, DevIDs also added for clarity in the README.md file. Thank you.

@PAR2020 : I have tried the last version and it works. Thank you.

closed 2022-07-14 by robang74