NVIDIA/open-gpu-kernel-modules #251: General VR issues.

Bitwolfies, 2022-05-21:
### NVIDIA Open GPU Kernel Modules Version

515.43.04-7

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Arch Linux

### Kernel Release

5.17.9

### Hardware: GPU

GTX 3080 

### Describe the bug

When using the open driver, SteamVR will crash Xorg 4/5 times, this issue is not happening when using the standard closed DKMS version, however, the 1/5 times it does work, it is the smoothest VR i've ever used on Linux, resolving several jittery issues when using closed.

Along with this, neither driver works over Xwayland, Kwin added DRM leasing support, which works fine with AMD cards, but not Nvidia ones.

### To Reproduce

Just try and use SteamVR, and wait for the crash.

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8746107/nvidia-bug-report.log.gz)

SpookySkeletons, 2022-05-24:
Could you elaborate on the smoothness? VR on both nvidia proprietary and amdgpu is HORRIBLY broken as far as their motion vector smoothing goes, is it functional? Could you post a little high framerate footage of it working on a heavy scene?

Bitwolfies, 2022-05-24:
> Could you elaborate on the smoothness? VR on both nvidia proprietary and amdgpu is HORRIBLY broken as far as their motion vector smoothing goes, is it functional? Could you post a little high framerate footage of it working on a heavy scene?

Id be lucky to even get back into a session, but from what was there, the bouncing of the steamvr grids and overlays, along with the steamvr menus awful stutter? Gone.

rwinkhart, 2022-05-25:
> Id be lucky to even get back into a session, but from what was there, the bouncing of the steamvr grids and overlays, along with the steamvr menus awful stutter? Gone.

The first issue you mentioned (with the grid/overlays) is actually due to a poor implementation of asynchronous re-projection. My guess is that SteamVR is disabling asynchronous re-projection on the open driver by default, either due to it not being implemented in the driver, or (more likely) due to the open driver not yet being whitelisted by Valve for enabling asynchronous re-projection.

You can fix this problem on the current closed drivers! 
Close Steam and add:
```
{
   "steamvr" : {
      "enableLinuxVulkanAsync" : false
   }
}
```
...into ~/.steam/steam/config/steamvr.vrsettings. The outer set of braces represent the preexisting ones in the configuration file, so just paste the inside to the bottom of your config file just before the last closing brace.

Once you open Steam back up and launch SteamVR, the menus/grids/overlays should no longer bounce about.

Vixea, 2023-11-26:
Update on this reprojection issue to see if Nvidia would be interested in working on a solution. Basically it's because of the compositor's inability to actually guarantee that it's pipeline will get time to complete on the GPU. See https://gitlab.freedesktop.org/mesa/mesa/-/issues/10101

Vixea, 2024-09-16:
Ping of status?

Wintch, 2026-09-13:
Confirming this also affects Sway/wlroots, not just KWin -- same symptom exactly: the
compositor announces `wp_drm_lease_device_v1` but offers zero connectors on it. Tested on
595.71.05-open (the open kernel modules, not the closed flavor) with an HP Reverb G2. GNOME/
mutter is the only compositor I could get to actually offer the connector on this same
hardware/driver -- it works there cleanly (confirmed both with a custom DRM/wayland-info probe
and with a real OpenXR compositor leasing and rendering to it). So this looks like it's specific
to KWin's and wlroots' own DRM-lease-offering code paths against the NVIDIA driver, not
something wrong with the driver's lease-granting in general