NVIDIA/open-gpu-kernel-modules #255: build failure on Fedora 36

YusufKhan-gamedev, 2022-05-24:
### NVIDIA Open GPU Kernel Modules Version

ce3d74ff6b49f7ec0e5e0aa44417f668b0f7189b

### Operating System and Version

Fedora release 36 (Thirty Six)

### Kernel Release

Linux fedora 5.17.8-300.fc36.x86_64 #1 SMP PREEMPT Mon May 16 01:00:37 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Build Command

make modules -j8
sudo make modules_install -j`nproc`

### Terminal output/Build Log

When building:
```
make[1]: Leaving directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/src/nvidia'
cd kernel-open/nvidia/ && ln -sf ../../src/nvidia/_out/Linux_x86_64/nv-kernel.o nv-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make[2]: Entering directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make[2]: *** /lib/modules/5.17.8-300.fc36.x86_64/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make: *** [Makefile:50: modules] Error 2

```
When installing:
```
make -C kernel-open modules_install
make[1]: Entering directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make[2]: Entering directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make[2]: *** /lib/modules/5.17.8-300.fc36.x86_64/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make[1]: *** [Makefile:82: modules_install] Error 2
make[1]: Leaving directory '/home/yusufkhan/.local/src/open-gpu-kernel-modules/kernel-open'
make: *** [Makefile:58: modules_install] Error 2
```

YusufKhan-gamedev, 2022-05-24:
```
yusufkhan@fedora ~/.l/s/open-gpu-kernel-modules (main) [2]> ls /lib/modules/5.17.8-300.fc36.x86_64/build
/lib/modules/5.17.8-300.fc36.x86_64/build@

```

aritger, 2022-05-25:
If the failure is
```
make[2]: *** /lib/modules/5.17.8-300.fc36.x86_64/build: No such file or directory.  Stop.
```
There isn't much open-gpu-kernel-modules can do.  I'd guess that /lib/modules/5.17.8-300.fc36.x86_64/build is a symlink, and the symlink target doesn't exist.

I think the following Fedora package provides the kernel headers for this kernel?
https://rpmfind.net/linux/RPM/fedora/updates/36/x86_64/Packages/k/kernel-devel-5.17.8-300.fc36.x86_64.html

closed 2022-05-25 by YusufKhan-gamedev