NVIDIA/open-gpu-kernel-modules #259: Unable to determine the device handle for GPU 0000:D8:00.0: Not Found

PetershengWUHU, 2022-05-26:
### NVIDIA Open GPU Kernel Modules Version

515.43.04

### Operating System and Version

CentOS Linux release 7.6.1810

### Kernel Release

Linux localhost.localdomain 3.10.0-957.12.1.el7.x86_64 #1 SMP Mon Apr 29 14:59:59 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

### Build Command

The first way:
make modules -j`nproc`
make modules_install -j`nproc`
sh ./NVIDIA-Linux-[...].run --no-kernel-modules
conclusions:
[root@localhost open-gpu-kernel-modules-main]#  modinfo nvidia | grep license
license:        Dual MIT/GPL
[root@localhost open-gpu-kernel-modules-main]# cat /proc/driver/nvidia/version
NVRM version: NVIDIA UNIX Open Kernel Module for x86_64  515.43.04  Release Build  (roo
GCC version:  gcc version 4.8.5 20150623 (Red Hat 4.8.5-36) (GCC)
However:
Unable to determine the device handle for GPU 0000:D8:00.0: Not Found

The second way:
sh ./NVIDIA-Linux-[...].run 
[image]

### Terminal output/Build Log

[nvidia-bug-report.log2.zip](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8775858/nvidia-bug-report.log2.zip)

### More Info

[image]

amrit1711, 2022-05-26:
@PetershengWUHU 

To enable use of the open kernel modules on GeForce and Workstation GPUs, set the "NVreg_OpenRmEnableUnsupportedGpus" nvidia.ko kernel module parameter to 1. E.g.,

    modprobe nvidia NVreg_OpenRmEnableUnsupportedGpus=1
or, in an /etc/modprobe.d/ configuration file:

    options nvidia NVreg_OpenRmEnableUnsupportedGpus=1

Please test the same and let us know if you still face any issues.

PetershengWUHU, 2022-05-26:
modprobe nvidia NVreg_OpenRmEnableUnsupportedGpus=1
this answer slove my problem.thank you!

closed 2022-05-26 by PetershengWUHU