NVIDIA/open-gpu-kernel-modules #263: drmOpen("nvidia", NULL) returns -1 #257

YusufKhan-gamedev, 2022-05-26:
### NVIDIA Open GPU Kernel Modules Version

ce3d74ff6b49f7ec0e5e0aa44417f668b0f7189b

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Description:	Fedora release 36 (Thirty Six)

### Kernel Release

Linux fedora 5.17.9-300.fc36.x86_64 #1 SMP PREEMPT Wed May 18 15:08:23 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

Its a RTX 2060 from GIGABYTE, I am not going to install the proprietary tool that is suggested

### Describe the bug

GPU file descriptors return -1 or 3

### To Reproduce

Try to open /dev/dri/cardx or use the drmOpen() function.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8781507/nvidia-bug-report.log.gz)

### More Info

[    4.751867] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    4.751868] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    4.751868] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    4.751869] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    4.751869] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    4.751870] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    4.751870] ACPI: [Firmware Bug]: No valid BIOS _PSS frequency found for processor 7
[    4.751871] ACPI: [Firmware Bug]: BIOS needs update for CPU frequency support
[    5.385291] nvidia-gpu 0000:01:00.3: i2c timeout error e0000000
[    5.385294] ucsi_ccg 0-0008: i2c_transfer failed -110
[    5.385295] ucsi_ccg 0-0008: ucsi_ccg_init failed - -110
[    5.385298] ucsi_ccg: probe of 0-0008 failed with error -110
[    5.398888] kauditd_printk_skb: 136 callbacks suppressed
[    5.398889] audit: type=1130 audit(1653589722.262:145): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-udev-settle comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.444839] audit: type=1130 audit(1653589722.308:146): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-fsck@dev-disk-by\x2duuid-cd5cf0c9\x2db7ce\x2d41da\x2dbcf1\x2dae0ccb7c629a comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.459774] audit: type=1130 audit(1653589722.323:147): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-fsck@dev-disk-by\x2duuid-5B81\x2d8B7D comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.463346] EXT4-fs (sda2): mounted filesystem with ordered data mode. Quota mode: none.
[    5.487808] audit: type=1130 audit(1653589722.351:148): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=dracut-shutdown comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.511918] audit: type=1130 audit(1653589722.375:149): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=plymouth-read-write comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.519858] audit: type=1130 audit(1653589722.383:150): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=import-state comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.567885] audit: type=1130 audit(1653589722.431:151): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-tmpfiles-setup comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.570480] audit: type=1334 audit(1653589722.433:152): prog-id=60 op=LOAD
[    5.570670] audit: type=1334 audit(1653589722.434:153): prog-id=61 op=LOAD
[    5.570726] audit: type=1334 audit(1653589722.434:154): prog-id=62 op=LOAD
[    5.602867] RPC: Registered named UNIX socket transport module.
[    5.602869] RPC: Registered udp transport module.
[    5.602870] RPC: Registered tcp transport module.
[    5.602870] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    5.771489] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.771491] Bluetooth: BNEP filters: protocol multicast
[    5.771494] Bluetooth: BNEP socket layer initialized
[    5.905379] NET: Registered PF_QIPCRTR protocol family
[... 35 lines cut ...]
[ 1319.406135] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #202!!!
[ 1319.407115] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #202!!!
[ 1320.797081] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #202!!!
[ 1320.798091] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #202!!!
[ 1326.121176] intel_powerclamp: Stop forced idle injection
[ 1343.137619] intel_powerclamp: Start idle injection to reduce power
[ 1355.185450] intel_powerclamp: Stop forced idle injection
[ 1372.202022] intel_powerclamp: Start idle injection to reduce power
[ 1384.226791] intel_powerclamp: Stop forced idle injection
[ 1401.243395] intel_powerclamp: Start idle injection to reduce power
[ 1411.275389] intel_powerclamp: Stop forced idle injection
[ 1637.846061] nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for x86_64  515.43.04  Release Build  (yusufkhan@)  Tue May 24 06:08:29 PM EDT 2022
[ 1637.846069] NVRM _threadNodeInitTime: Bad threadStateDatabase.timeout.flags: 0x0!
[ 1637.846071] NVRM rmapiAllocWithSecInfo: client:0x0 parent:0x0 object:0x0 class:0x0
[ 1637.846074] NVRM _threadNodeInitTime: Bad threadStateDatabase.timeout.flags: 0x0!
[ 1637.846083] NVRM rmapiAllocWithSecInfo: allocation complete
[ 1637.847149] nvidia_drm: unknown parameter 'NVreg_RmMsg' ignored
[ 1637.847478] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[ 1637.847686] NVRM _threadNodeInitTime: Bad threadStateDatabase.timeout.flags: 0x0!
[ 1637.847688] NVRM _threadNodeInitTime: Bad threadStateDatabase.timeout.flags: 0x0!
[ 1637.847759] nvidia 0000:01:00.0: Direct firmware load for nvidia/515.43.04/gsp.bin failed with error -2
[ 1637.847771] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x61:0x0:1610)
[ 1637.847787] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[ 1637.847824] [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to allocate NvKmsKapiDevice
[ 1637.847891] [drm:nv_drm_probe_devices [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to register device

aritger, 2022-05-26:
Well, nothing is going to work as long as you have this error I see in your logs:

```
May 26 14:28:41 fedora kernel: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x61:0x0:1610)
May 26 14:28:41 fedora kernel: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```

So, let's start with that.  The preceding lines look concerning:

```
[ 1637.847686] NVRM _threadNodeInitTime: Bad threadStateDatabase.timeout.flags: 0x0!
[ 1637.847688] NVRM _threadNodeInitTime: Bad threadStateDatabase.timeout.flags: 0x0!
[ 1637.847759] nvidia 0000:01:00.0: Direct firmware load for nvidia/515.43.04/gsp.bin failed with error -2
[ 1637.847771] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x61:0x0:1610)
[ 1637.847787] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```

The "Bad threadStateDatabase.timeout.flags" messages are not expected, but worse is the firmware load error.  It looks like it is failing to find nvidia/515.43.04/gsp.bin.  Does /lib/firmware/nvidia/515.43.04/gsp.bin exist on your system?

YusufKhan-gamedev, 2022-05-26:
> /lib/firmware/nvidia/515.43.04/gsp.bin

No
```
[yusufkhan@fedora ~]$ ls /lib/firmware/nvidia/
ga102  ga103  ga104  ga106  ga107  gk20a  gm200  gm204  gm206  gm20b  gp100  gp102  gp104  gp106  gp107  gp108  gp10b  gv100  tegra124  tegra186  tegra194  tegra210  tu102  tu104  tu106  tu10x  tu116  tu117
```

aritger, 2022-05-26:
One way is described in https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/README.md

```
Note that the kernel modules built here must be used with gsp.bin firmware and user-space NVIDIA GPU driver components
from a corresponding 515.43.04 driver release. This can be achieved by installing the NVIDIA GPU driver from the .run file using
the --no-kernel-modules option. E.g.,

sh ./NVIDIA-Linux-[...].run --no-kernel-modules
```

YusufKhan-gamedev, 2022-05-26:
Is there a way to get the firmware without the userland?

aritger, 2022-05-27:
You can extract just the gsp.bin if you prefer.  Something like this:
```
wget http://download.nvidia.com/XFree86/Linux-x86_64/515.43.04/NVIDIA-Linux-x86_64-515.43.04.run
sh ./NVIDIA-Linux-x86_64-515.43.04.run -x
mkdir -p /lib/firmware/nvidia/515.43.04/
cp NVIDIA-Linux-x86_64-515.43.04/firmware/gsp.bin /lib/firmware/nvidia/515.43.04/gsp.bin
```

YusufKhan-gamedev, 2022-05-27:
Problem persists

```
yusufkhan@fedora ~/.l/s/drm (nvidia-next)> ./a.out
3
0⏎      
```        
I added a printf for the handle and made sure that card0 was a valid file

YusufKhan-gamedev, 2022-05-27:
Solution: O_CREAT

closed 2022-05-27 by YusufKhan-gamedev