NVIDIA/open-gpu-kernel-modules #265: NULL pointer deference in GrabOwnership+0x4/0x40

YusufKhan-gamedev, 2022-05-27:
### NVIDIA Open GPU Kernel Modules Version

ce3d74ff6b49f7ec0e5e0aa44417f668b0f7189b

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Description:	Fedora release 36 (Thirty Six)

### Kernel Release

Linux fedora 5.17.9-300.fc36.x86_64 #1 SMP PREEMPT Wed May 18 15:08:23 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

Its a RTX 2060 from GIGABYTE, I am not going to install the proprietary tool that is suggested

### Describe the bug

    5.048788] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    5.048788] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    5.048789] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    5.048789] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    5.048790] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    5.048790] ACPI: [Firmware Bug]: Invalid BIOS _PSS frequency found for processor 7: 0x80000000 MHz
[    5.048791] ACPI: [Firmware Bug]: No valid BIOS _PSS frequency found for processor 7
[    5.048791] ACPI: [Firmware Bug]: BIOS needs update for CPU frequency support
[    5.696161] nvidia-gpu 0000:01:00.3: i2c timeout error e0000000
[    5.696165] ucsi_ccg 0-0008: i2c_transfer failed -110
[    5.696166] ucsi_ccg 0-0008: ucsi_ccg_init failed - -110
[    5.696168] ucsi_ccg: probe of 0-0008 failed with error -110
[    5.711771] kauditd_printk_skb: 136 callbacks suppressed
[    5.711772] audit: type=1130 audit(1653611711.576:145): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-udev-settle comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.751815] audit: type=1130 audit(1653611711.616:146): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-fsck@dev-disk-by\x2duuid-cd5cf0c9\x2db7ce\x2d41da\x2dbcf1\x2dae0ccb7c629a comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.763793] audit: type=1130 audit(1653611711.628:147): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-fsck@dev-disk-by\x2duuid-5B81\x2d8B7D comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.767791] EXT4-fs (sda2): mounted filesystem with ordered data mode. Quota mode: none.
[    5.797817] audit: type=1130 audit(1653611711.662:148): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=dracut-shutdown comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.819709] audit: type=1130 audit(1653611711.684:149): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=plymouth-read-write comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.826745] audit: type=1130 audit(1653611711.691:150): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=import-state comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.875769] audit: type=1130 audit(1653611711.740:151): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-tmpfiles-setup comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[    5.878325] audit: type=1334 audit(1653611711.742:152): prog-id=60 op=LOAD
[    5.878401] audit: type=1334 audit(1653611711.742:153): prog-id=61 op=LOAD
[    5.878447] audit: type=1334 audit(1653611711.743:154): prog-id=62 op=LOAD
[    5.911293] RPC: Registered named UNIX socket transport module.
[    5.911296] RPC: Registered udp transport module.
[    5.911296] RPC: Registered tcp transport module.
[    5.911296] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    6.038002] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.038004] Bluetooth: BNEP filters: protocol multicast
[    6.038007] Bluetooth: BNEP socket layer initialized
[    6.223234] NET: Registered PF_QIPCRTR protocol family
[    6.837424] iwlwifi 0000:00:14.3: Conflict between TLV & NVM regarding enabling LAR (TLV = enabled NVM =disabled)
[    7.024052] iwlwifi 0000:00:14.3: Conflict between TLV & NVM regarding enabling LAR (TLV = enabled NVM =disabled)
[    9.509038] e1000e 0000:00:1f.6 eno2: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[    9.509088] IPv6: ADDRCONF(NETDEV_CHANGE): eno2: link becomes ready
[   10.035938] thermal cooling_device11: Setting cooling device state is deprecated
[   11.744620] rfkill: input handler disabled
[   12.420396] Bluetooth: RFCOMM TTY layer initialized
[   12.420400] Bluetooth: RFCOMM socket layer initialized
[   12.420422] Bluetooth: RFCOMM ver 1.11
[   18.664787] rfkill: input handler enabled
[   50.945360] logitech-hidpp-device 0003:046D:1025.0007: HID++ 1.0 device connected.
[  463.244682] nvidia-modeset: Unloading
[  463.262190] NVOC: __nvoc_objDelete: Child class OBJIOVASPACE not freed from parent class OBJVMM.Allocator 00000000ba323f72 released with memory allocations
[  463.262212] [NvPort] *************************************************
[  463.262213] NvPort memory tracking information for allocator 00000000ba323f72:
[  463.262213] ACTIVE: 1 allocations, 644 bytes allocated (616 useful, 28 meta)
[... 71 lines cut ...]
[  463.506747] FS:  00007fb052144bc0(0000) GS:ffff9ce98dd40000(0000) knlGS:0000000000000000
[  463.506748] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  463.506749] CR2: 0000000000000040 CR3: 000000010bb08001 CR4: 00000000003706e0
[  463.506750] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  463.506750] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  463.722840] show_signal: 7 callbacks suppressed
[  463.722841] traps: xss-lock[1807] trap int3 ip:7f1767595df1 sp:7ffc84704890 error:0
[  463.722845] fbcon: Taking over console
[  463.722850]  in libglib-2.0.so.0.7200.1[7f1767559000+91000]
[  463.724632] Console: switching to colour frame buffer device 128x48
[  464.801771] rfkill: input handler disabled
[  471.179449] rfkill: input handler enabled

### To Reproduce

Reload nvidia drivers

### Bug Incidence

Once

### nvidia-bug-report.log.gz

I believe the dmesg would be enough, it includes a core dump but here it is:
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8783095/nvidia-bug-report.log.gz)

aritger, 2022-05-27:
This looks like bad error handling in response to this failure:

```
[ 463.505116] NVRM kgspInitRm_IMPL: missing NVDEC0 engine, cannot initialize GSP-RM
[ 463.505120] NVRM RmInitAdapter: Cannot initialize GSP firmware RM
[ 463.505392] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x63:0x56:1689)
[ 463.506360] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```

The NVDEC0 problem is https://github.com/NVIDIA/open-gpu-kernel-modules/issues/116 which will be fixed in our next release.

PAR2020, 2022-06-01:
NVBug 3667921

qWici, 2022-12-14:
Same error. GTX 3070. Ubuntu 22.04.1. Temporary downgrade driver to 510. Wait for fix in next releases)

aritger, 2023-11-08:
@YusufKhan-gamedev , @qWici : if you still see this, I'd be curious to know
(a) What driver version, and
(b) The kernel log leading up to the failure.  Since the `missing NVDEC0 engine, cannot initialize GSP-RM` bug is fixed, if GSP is still failing to initialize for you, I'd like to see the reason.

We should still fix the error handling path that causes the NULL dereference in GrabOwnership().  If you still see that, could you test this trivial error check earlier in that path?

```
diff --git a/kernel-open/nvidia-drm/nvidia-drm-drv.c b/kernel-open/nvidia-drm/nvidia-drm-drv.c
index e0ddb6cb279b..a64ea7f6b75d 100644
--- a/kernel-open/nvidia-drm/nvidia-drm-drv.c
+++ b/kernel-open/nvidia-drm/nvidia-drm-drv.c
@@ -620,6 +620,10 @@ static int __nv_drm_master_set(struct drm_device *dev,
 {
     struct nv_drm_device *nv_dev = to_nv_device(dev);
 
+    if (!nv_dev || !nv_dev->pDevice) {
+        return -EINVAL;
+    }
+
     /*
      * If this device is driving a framebuffer, then nvidia-drm already has
      * modeset ownership. Otherwise, grab ownership now.
```

qWici, 2023-11-10:
@aritger nope, I already have another computer)

YusufKhan-gamedev, 2024-01-06:
@aritger The current 565.blah release doesnt have this issue at first glance, but that might be due to this particular codepath not being triggered at all as /dev/nvidia* doesnt exist.