NVIDIA/open-gpu-kernel-modules #271: make modules_install failed

Bubble-mt, 2022-05-28:
### NVIDIA Open GPU Kernel Modules Version

515.43.04

### Operating System and Version

openEuler release 22.03 LTS

### Kernel Release

Linux taishan200-2280-2s64p-128g--a134 5.10.0-60.18.0.50.oe2203.aarch64 #1 SMP Wed Mar 30 02:43:08 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

### Build Command

**1. Set signing key**
cd /usr/lib/modules/$(uname -r)/build/certs

sudo tee x509.genkey >/dev/null << 'EOF'
[ req ]
default_bits = 4096
distinguished_name = req_distinguished_name
prompt = no
string_mask = utf8only
x509_extensions = myexts

[ req_distinguished_name ]
CN = Modules

[ myexts ]
basicConstraints=critical,CA:FALSE
keyUsage=digitalSignature
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid
EOF

openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform PEM -out signing_key.pem -keyout signing_key.priv

**2. Change compiler flags**
cd open-gpu-kernel-modules-main
sed -i -e s/-march=armv8-a/-march=armv8-a+lse/g */*/Makefile */Kbuild

**3. make install driver**
make modules -j`nproc`
make modules_install -j`nproc`

### Terminal output/Build Log

# make modules_install
make -C kernel-open modules_install
make[1]: Entering directory '/root/open-gpu-kernel-modules-main/kernel-open'
make[2]: Entering directory '/usr/src/kernels/5.10.0-60.31.0.60.oe2203.aarch64'
  INSTALL /root/open-gpu-kernel-modules-main/kernel-open/nvidia-drm.ko
  INSTALL /root/open-gpu-kernel-modules-main/kernel-open/nvidia-modeset.ko
  INSTALL /root/open-gpu-kernel-modules-main/kernel-open/nvidia-peermem.ko
  INSTALL /root/open-gpu-kernel-modules-main/kernel-open/nvidia-uvm.ko
  INSTALL /root/open-gpu-kernel-modules-main/kernel-open/nvidia.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.10.0-60.31.0.60.oe2203.aarch64
make[2]: Leaving directory '/usr/src/kernels/5.10.0-60.31.0.60.oe2203.aarch64'
make[1]: Leaving directory '/root/open-gpu-kernel-modules-main/kernel-open'

# ll /usr/src/kernels/5.10.0-60.31.0.60.oe2203.aarch64/certs/signing_key.pem
-rw-r--r-- 1 root root 1.8K 2022-05-28 17:37 /usr/src/kernels/5.10.0-60.31.0.60.oe2203.aarch64/certs/signing_key.pem

# cd /lib/modules/5.10.0-60.31.0.60.oe2203.aarch64/kernel/drivers/video/
# ls
nvidia-drm.ko  nvidia.ko  nvidia-modeset.ko  nvidia-peermem.ko  nvidia-uvm.ko

### More Info

My env has **/usr/lib/modules/$(uname -r)/build/certs/signing_key.pem** file, but when I execute make install driver, it would to be install failed. 
In /lib/modules/$(uname -r)/kernel/drivers/video directory, there are ko files which can be insmod mannually.

YusufKhan-gamedev, 2022-05-28:
I doubt this is a build faliure...can you run `tree | grep nvi` on your /lib/modules directory?

luizoti, 2022-05-28:
Output of `tree | grep nvi`:

```   ├── nvidiadetector.py
│       └── nvidiadetector.cpython-38.pyc
│   │   ├── _appengine_environ.py
│   │   │   ├── _appengine_environ.cpython-38.pyc
```
Same error here:

```make -C kernel-open modules_install
make[1]: Entering directory '/tmp/open-gpu-kernel-modules/kernel-open'
make[2]: Entering directory '/usr/src/linux-headers-5.13.0-44-generic'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
  INSTALL /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-drm.ko
  INSTALL /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-modeset.ko
  INSTALL /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-peermem.ko
  INSTALL /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-uvm.ko
  INSTALL /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia.ko
  SIGN    /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-peermem.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  SIGN    /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-modeset.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  SIGN    /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-drm.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  SIGN    /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  SIGN    /lib/modules/5.13.0-44-generic/kernel/drivers/video/nvidia-uvm.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  /lib/modules/5.13.0-44-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.13.0-44-generic'
make[1]: Leaving directory '/tmp/open-gpu-kernel-modules/kernel-open'
```

> I doubt this is a build faliure...can you run `tree | grep nvi` on your /lib/modules directory?

YusufKhan-gamedev, 2022-05-29:
Sorry, I guess I was too vague, `cd /lib/modules && tree | grep nvi` anyway, thesis is that these arent errors, most likely they are being installed but they arent being signed which is a problem with your authority but nearly harmless.

Bubble-mt, 2022-05-30:
Build was succeed. There were ko files, but  I installed them by make install failed.
# cd /lib/modules && tree | grep nvi
    │           ├── nvidia-drm.ko
    │           ├── nvidia.ko
    │           ├── nvidia-modeset.ko
    │           ├── nvidia-peermem.ko
    │           └── nvidia-uvm.ko

Bubble-mt, 2022-05-30:
> Sorry, I guess I was too vague, `cd /lib/modules && tree | grep nvi` anyway, thesis is that these arent errors, most likely they are being installed but they arent being signed which is a problem with your authority but nearly harmless.
> 
> For OPs case, I have no idea why his modules arent being installed

Build was succeed. There were ko files, but  I installed them by make install failed.
`# cd /lib/modules && tree | grep nvi`
    │           ├── nvidia-drm.ko
    │           ├── nvidia.ko
    │           ├── nvidia-modeset.ko
    │           ├── nvidia-peermem.ko
    │           └── nvidia-uvm.ko

YusufKhan-gamedev, 2022-05-31:
Your make install is not showing any errors other than signing errors(result in kernel taint which is totally fine) AFAIK you simply have to reboot and disable the nouveau modeseting driver in grub and enable nvidia modesetting and youll be fine

Bubble-mt, 2022-06-01:
I have tried to disable nouveau driver, reboot, restart to build and make install .But it still would show signing errors when make install, how could I deal the error?

luizoti, 2022-06-06:
> I have tried to disable nouveau driver, reboot, restart to build and make install .But it still would show signing errors when make install, how could I deal the error?

Try this [https://superuser.com/a/1659287](url)

Bubble-mt, 2022-07-14:
When I use the newest code, I execute make modules_install command succeed, there is no error or warning anymore.
I will close this issue, thanks all!
[image]

closed 2022-07-14 by Bubble-mt