NVIDIA/open-gpu-kernel-modules #272: Version difference with the proprietary driver causes csgo to crash

Sumandora, 2022-05-28:
### NVIDIA Open GPU Kernel Modules Version

515.43.04-7 installed via https://archlinux.org/packages/extra/x86_64/nvidia-open/

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Arch Linux / KDE Plasma

### Kernel Release

Linux arch 5.17.9-arch1-1 #1 SMP PREEMPT Wed, 18 May 2022 17:30:11 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 2060

### Describe the bug

After a certain time of csgo running, it crashes.
It crashing has the side effect of freezing the entire Xorg server.

### To Reproduce

I'm not able to force this crash, but I never got through the freeze time. So testing it is very easy.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

I'm sadly unable to provide this, due to having switched to the proprietary driver for now.
But I'm able to provide the following:
The csgo crash:
```
[  283.236321] csgo_linux64[1476]: segfault at 0 ip 00007faae5c9396b sp 00007faaca50a140 error 6 in client_client.so[7faae4e00000+1dfa000]
[  283.236343] Code: c7 83 c8 1f 01 00 00 00 00 00 44 8b a3 a8 0d 01 00 4c 8d ab c0 0c 01 00 e8 a2 78 ab ff 8b 70 20 48 8b 08 49 63 d4 48 c1 e2 06 <89> 34 11 48 8b 08 48 c7 44 11 08 00 00 00 00 44 89 60 20 48 8d 83
```
A bunch of errors before csgo finally crashing:
```
[  274.920300] NVRM: Xid (PCI:0000:29:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[  274.920302] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[  274.920302] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[  274.920304] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d0003f; hObject=0xbeee502d; paramsStatus=0x00000000; status=0x00000065
[  274.920305] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_client.c:805
[  274.920308] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:155
[  274.920312] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:995
[  274.920319] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32dccf22f4de01 >= 32dccf22c202bc
[  274.920320] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[  274.920321] NVRM: Xid (PCI:0000:29:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[  274.920323] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[  274.920324] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[  274.920325] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d0003f; hObject=0xbeee3901; paramsStatus=0x00000000; status=0x00000065
[  274.920326] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_client.c:805
[  274.920384] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:155
[  274.920388] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:995
[  274.920396] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32dccf22f4de01 >= 32dccf22c202bc
[  274.920397] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[  274.920398] NVRM: Xid (PCI:0000:29:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[  274.920400] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[  274.920401] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[  274.920402] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d0003f; hObject=0xbeee0100; paramsStatus=0x00000000; status=0x00000065
[  278.942591] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[  278.942595] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[  278.942597] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[  278.942598] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[  278.942599] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[  278.942601] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[  282.922540] NVRM _kgspRpcDrainOneEvent: Failed to process received event 4100: status=0x1f
[  282.922546] NVRM _issueRpcAndWait: rpcRecvPoll failed with status 0x0000001f for fn 4100!
[  282.922547] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[  282.922565] nvidia-modeset: ERROR: GPU:0: Failed to query display engine channel state: 0x0000c57e:0:0:0x0000001f
[  282.928478] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[  282.959818] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0x4c
```

### More Info

I looked at the disassembly of the code region which crashed the game.
There were 3 MOV instructions. The third being the one which crashed the game.
It looked like a class or array which is being written to, but the method does not have any strings nearby so I'm unsure about what is actually happening.

The only "fix" I know of is to switch to the proprietary driver.

LethalManBoob, 2022-05-29:
I dont know if this warrents its own thread but the open source dkms drivers for nvidia cause opengl games to softlock my whole pc. the whole pc runs at like one frame every 1 minute and i need to hard shut down. normal drivers have no such issue. 
rtx3070 archlinux
ryzen 3800x

ill give more detailed info if needed but im not installing it again and risk issues with my hardware due to the restarts

tuantran1632001, 2022-05-29:
> cause opengl games to softlock my whole pc

Seem to be the same problem, I also have similar error message as  @Sumandora while playing Factorio.

Sumandora, 2022-05-29:
> I dont know if this warrents its own thread but the open source dkms drivers for nvidia cause opengl games to softlock my whole pc. the whole pc runs at like one frame every 1 minute and i need to [...]

Restarting is not necessary. Press Ctrl+Alt+Backspace. This hotkey "zaps" the entire xorg server. Practically restarting it. Try to use dmesg after that and check if you got similar errors. If so it is likely that you have the same error

amrit1711, 2022-05-30:
@Sumandora 
Thanks for reporting issue, I have filed a bug 3664227 internally for tracking purpose. Shall try to reproduce issue locally first and if required any further information, I will get back to you.

rnd-ash, 2022-05-31:
I've noticed the same with other source engine games such as Black-Mesa and TF2. 
```
[21155.483662] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32efcc9f0bedc9 >= 32efcc9ed91284
[21155.483666] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[21155.483672] NVRM: Xid (PCI:0000:01:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[21155.483676] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[21155.483677] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[21155.483681] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d000cd; hObject=0xbeee4901; paramsStatus=0x00000000; status=0x00000065
[21155.483684] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_client.c:805
[21155.483703] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:155
[21155.483709] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:995
[21155.483722] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32efcc9f0bedc9 >= 32efcc9ed91284
[21155.483722] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[21155.483723] NVRM: Xid (PCI:0000:01:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[21155.483725] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[21155.483726] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[21155.483728] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d000cd; hObject=0xbeeea0b5; paramsStatus=0x00000000; status=0x00000065
[21155.483728] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_client.c:805
[21155.483732] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:155
[21155.483735] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:995
[21155.483847] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32efcc9f0bedc9 >= 32efcc9ed91284
[21155.483849] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[21155.483852] NVRM: Xid (PCI:0000:01:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[21155.483854] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[21155.483855] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[21155.483857] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d00008; hObject=0xb00f000e; paramsStatus=0x00000000; status=0x00000065
[21155.483864] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32efcc9f0bedc9 >= 32efcc9ed91284
[21155.483865] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[21155.483866] NVRM: Xid (PCI:0000:01:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[21155.483868] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[21155.483868] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[21155.483870] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d000cd; hObject=0xbeee9097; paramsStatus=0x00000000; status=0x00000065
[21155.483871] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_client.c:805
[21155.483881] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:155
[21155.483884] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:995
[21155.483891] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32efcc9f0bedc9 >= 32efcc9ed91284
[21155.483892] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[21155.483893] NVRM: Xid (PCI:0000:01:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[21155.483894] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[21155.483894] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[21155.483896] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d000cd; hObject=0xbeee0404; paramsStatus=0x00000000; status=0x00000065
[21155.483905] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32efcc9f0bedc9 >= 32efcc9ed91284
[21155.483905] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[21155.483906] NVRM: Xid (PCI:0000:01:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[21155.483907] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[21155.483908] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[21155.483910] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d000cd; hObject=0xbeee502d; paramsStatus=0x00000000; status=0x00000065
[21155.483910] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_client.c:805
[21155.483914] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:155
[21155.483916] NVRM nvAssertFailedNoLog: Assertion failed: status == NV_OK @ rs_server.c:995
[21155.483921] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32efcc9f0bedc9 >= 32efcc9ed91284
[21155.483922] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[21155.483922] NVRM: Xid (PCI:0000:01:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[... cut ...]
[21163.594503] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[21163.594600] NVRM _kgspProcessRpcEvent: Unexpected RPC function 0xa
[21163.611528] NVRM serverFreeResourceTree: hObject 0xbeee0400 not found for client 0xc1d000cd
[21163.611537] NVRM serverFreeResourceTree: hObject 0xbeee0401 not found for client 0xc1d000cd
[21163.611547] NVRM serverFreeResourceTree: hObject 0xbeee0402 not found for client 0xc1d000cd
[21163.611556] NVRM serverFreeResourceTree: hObject 0xbeee0403 not found for client 0xc1d000cd
```

mtijanic, 2022-06-02:
For some reason, the RPC to clean up some resources does not finish in time. This probably mean that GSP got stuck in an infinite loop or otherwise crashed, but it could be that it just needs to do way too much to free it.

Could you maybe reproduce with [`NVreg_RmMsg=":"`](https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/197) to capture more debug logs?

Also, do you perhaps know if in the minute leading up to the crash there was a graphics-heavy process shutdown? Or maybe a particular change int he game, such as a level reload? Something that could trigger a cleanup of a large resource tree. One way we have seen this happen in the past is if a GPU-superheavy process gets a SIGKILL and the driver attempts to reap all the allocations.

LethalManBoob, 2022-06-02:
In my case i can say that it was completly random and it was every thing from counter strike source to cruelty squad.

Sumandora, 2022-06-03:
> For some reason, the RPC to clean up some resources does not finish in time. This probably mean that GSP got stuck in an infinite loop or otherwise crashed, but it could be that it just needs to do [...]

I can't test until I get home, but from my experience crashes happen as soon as I got transferred from spectator to actual player, also got a crash as soon as getting shot. So that could possibly be it. Over a few rather weird attempts of getting more information about the referenced asm in dmesg. I found out that I apparently crashed because the freeze time effect or something like that was loaded. But I am not sure if.I Analyse that correctly because my way of analyzing was to take a source leak of csgo and trying to find the bytes but they were not there. Maybe compiler or version (leak is from 2018) differences. So tldr: It could very well be that.

Posting full crashlog when home

amrit1711, 2022-06-03:
@Sumandora 
I also tried reproducing issue on below configuration setup but did not observed any crash after playing CSGO for an hour.
Arch Linux  +  5.18.1-arch1-1  + NVIDIA GeForce RTX 2080  +  Driver 515.48.07 with --no-kernel-modules  +  DELL U2412M

Can you please confirm repro frequency and share nvidia bug report.
Also let me know if you have any specific game or display settings.

Sumandora, 2022-06-03:
> > For some reason, the RPC to clean up some resources does not finish in time. This probably mean that GSP got stuck in an infinite loop or otherwise crashed, but it could be that it just needs to do way too much to free it.
> > Could you maybe reproduce with [`NVreg_RmMsg=":"`](https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/197) to capture more debug logs?
> > Also, do you perhaps know if in the minute leading up to the crash there was a graphics-heavy process shutdown? Or maybe a particular change int he game, such as a level reload? Something that could trigger a cleanup of a large resource tree. One way we have seen this happen in the past is if a GPU-superheavy process gets a SIGKILL and the driver attempts to reap all the allocations.
> 
> I can't test until I get home, but from my experience crashes happen as soon as I got transferred from spectator to actual player, also got a crash as soon as getting shot. So that could possibly be it. Over a few rather weird attempts of getting more information about the referenced asm in dmesg. I found out that I apparently crashed because the freeze time effect or something like that was loaded. But I am not sure if I analysed that correctly because my way of analyzing was to take a source leak of csgo and trying to find the bytes but they were not there. Maybe compiler or version (leak is from 2018) differences. But I tried to find a connection between the version regardless. So tldr: It could very well be that.
> 
> Posting full crashlog when home

I used the logging argument.
I also increased my Kernel Log size to 64M, because my first attempt failed.

Full crash log: https://gist.github.com/Sumandora/9d86eb639d56116c2ee32207b49a0a44
I have included 3000 lines +/- to the actual segfault.

> @Sumandora I also tried reproducing issue on below configuration setup but did not observed any crash after playing CSGO for an hour. Arch Linux  +  5.18.1-arch1-1  + NVIDIA GeForce RTX 2080  + [...]

I'm going to include a bit more system information:
I already provided that in the csgo-issue-tracker-issue: https://gist.github.com/Sumandora/3d5d858d62d08addbd36a7ea293adc74
I use betterfs, KDE Plasma, Ryzen 5 2600 and a RTX 2060, 16 GB of Memory
I got 2 Monitors, both of them are 4k.
And my Arch Linux, like the game, is installed on a SSD

Some of these things might have a impact...
But yes the bug is still reproducible and is not fixed.

mtijanic, 2022-06-03:
Thanks! From just before the crash:
```
[  140.879819] NVRM RmFreeUnusedClients: freeing abandoned client 0xc1d0001a
[  140.879821] NVRM rmapiFreeClientListWithSecInfo: Nv01FreeClientList: numClients: 1
```
means that some process has closed its `/dev/nvidia*` fd (either via `close()` or just by dying and OS reaping the fd) without going through the proper teardown process; and for some reason this breaks GSP.

>  I have included 3000 lines +/- to the actual segfault.

Could you maybe upload the full log file? It's very verbose and the 6k does not provide enough context to see what client `0xc1d0001a` is, or what it has allocated that is now problematic to free/reap. With the full log we can get an audit trail for the client in question.

Sumandora, 2022-06-03:
> Thanks! From just before the crash:
>
> ```
> [  140.879819] NVRM RmFreeUnusedClients: freeing abandoned client 0xc1d0001a
> [  140.879821] NVRM rmapiFreeClientListWithSecInfo: Nv01FreeClientList: [...]

[dmesg.tar.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8833160/dmesg.tar.gz)
I had to increase my kernel log length even more, now its 1 Gigabyte, because 64 Megabytes also wasnt enough to capture the log from beginning.
However here is the full-full log you have been requesting.
The file is about 17 MB compressed and 170 MB uncompressed.

mtijanic, 2022-06-03:
Thank you, this is super helpful! Seems like this time it was different, and it's not a reap of an unused client, but a more controlled free, but still same problem:
```
[  161.064792] NVRM rmapiFreeWithSecInfo: Nv01Free: client:0xc1d0005e object:0xbeee0100
[  161.067693] NVRM rmapiUnmapFromCpuWithSecInfo: Nv04UnmapMemory: client:0xc1d00018 device:0xbeef0004 memory:0xcaf0004b pLinearAddr:144929000 flags:0x0
[  163.075327] NVRM threadStateYieldCpuIfNecessary: Yielding
[  165.069620] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: currentTime: 32dcb567d964d7 >= 32dcb567a68992
[  165.069624] NVRM _threadNodeCheckTimeout: _threadNodeCheckTimeout: Timeout was set to: 4000 msecs!
[  165.069629] NVRM: GPU at PCI:0000:29:00: GPU-f37e16b7-7bf3-5af6-0ca7-1d588cffdd1c
[  165.069631] NVRM: Xid (PCI:0000:29:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
[  165.069634] NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
[  165.069635] NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:200
[  165.069638] NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d0005e; hObject=0xbeeea0b5; paramsStatus=0x00000000; status=0x00000065
```
It is trying to release a DMA channel, which is probably still in use, and for some reason GSP is unable to stop it within the allotted 4s timeslice.

I'll pass this on into the internal bug to the engineers more familiar with these code paths.

PAR2020, 2022-06-08:
@Sumandora, can we get a nvidia-bug-report.log on this one with the open source driver? More data is needed from it to close with our devs. @mtijanic, would it be helpful to get all rm messages also for Nikita?

Thanks guys!

Sumandora, 2022-06-08:
> @Sumandora, can we get a nvidia-bug-report.log on this one with the open source driver? More data is needed from it to close with our devs. @mtijanic, would it be helpful to get all rm messages also for Nikita?
> 
> Thanks guys!
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8865246/nvidia-bug-report.log.gz)

Does this help? ^^

DonielMoins, 2022-06-22:
I am having the same issue here as well, where can I find the log files to help out?

dylif, 2022-06-22:
I am having the same issue here:
Kernel: 5.17.7-arch1-1
Driver: nvidia-open 515.43.04

Seems like something in the kernel module is locking up whenever I run an OpenGL game (i.e. Minecraft), as there is a lot of reports of timeouts. Strangely running a Vulkan game (GTA 5 through Proton and DXVK) works perfectly. 

This does not happen with the proprietary kernel modules; both OpenGL and Vulkan games work perfectly.

Please note I have tried adding the `nvidia-drm.modeset=1` kernel parameter which changed nothing.

Attached is a snippet of a log that I think is of interest.
[nvidia.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8693995/nvidia.log)

Hopefully this helps

T-D3V, 2022-06-28:
I have this issue too when trying to launch minecraft right now: NVRM: Xid (PCI:0000:09:00): 119, pid='<unknown>', name=<unknown>, Timeout waiting for RPC from GSP! Expected function FREE (0x0 0x0).
I'm using current linux-zen kernel and nvidia-open-dkms

frznfngrs, 2022-06-30:
I am also experiencing this error with RTX 3090 with ML/AI workloads

PAR2020, 2022-06-30:
Hi all - we have a fix and getting it integrated. Will let you know which driver release it's in.

Sumandora, 2022-06-30:
> Hi all - we have a fix and getting it integrated. Will let you know which driver release it's in.

Sounds awesome. I'm hyped.

frznfngrs, 2022-08-07:
> 
Any update on this?

ngolovliov-nv, 2022-08-08:
Does the latest driver release, 515.65, address the issue for you?

andrewathalye, 2022-08-09:
The latest driver release appears to resolve crashing issues for me. I have not yet tested all apps that formerly crashed, but several no longer due.

Sumandora, 2022-08-16:
> Does the latest driver release, 515.65, address the issue for you?

I just tested it, it works.
No more crashes.
So I guess this can be closed.
Thanks for patching the problems, now we can go crazy, gaming on the kernel-open drivers :+1:

closed 2022-08-16 by Sumandora

875d, 2023-01-26:
I'm still experiencing this on 525.85.05

yfaimisaka, 2023-02-21:
I'm `nvidia-open-dkms 525.89.02-3`   `kernel 6.1.12-zen1-1-zen` have the same issue like this
```
NVRM: Xid (PCI:0000:01:00): 119, pid=2061, name=alacritty, Timeout waiting for RPC from GSP! Expected function 103 (GSP_RM_ALLOC) (0x80 0x38).
NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 103!
NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:205
NVRM rpcRmApiAlloc_GSP: GspRmAlloc failed: hClient=0xc1d00013; hParent=0xc1d00013; hObject=0xfade0001; hClass=0x00000080; paramsSize=0x00000038; paramsStatus=0x00000000; status=0x00000065
NVRM serverFreeResourceTree: hObject 0xfade0002 not found for client 0xc1d00013
NVRM: Xid (PCI:0000:01:00): 119, pid=2061, name=alacritty, Timeout waiting for RPC from GSP! Expected function 10 (FREE) (0xfade0001 0x0).
NVRM _issueRpcAndWait: rpcRecvPoll timedout for fn 10!
NVRM nvAssertFailedNoLog: Assertion failed: 0 @ rpc.c:205
NVRM rpcRmApiFree_GSP: GspRmFree failed: hClient=0xc1d00013; hObject=0xfade0001; paramsStatus=0x00000000; status=0x00000065
```