NVIDIA/open-gpu-kernel-modules #279: [BUG] Lutris does not open and does not recognise GPU

TheRealOne78, 2022-06-02:
### NVIDIA Open GPU Kernel Modules Version

515.48.07 - Gentoo kernel-open build

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Gentoo Base System release 2.8

### Kernel Release

5.18.1-gentoo-r1 - configured and compiled by myself

### Hardware: GPU

NVIDIA GeForce RTX 2060 (UUID: GPU-466269ac-80e7-d5cf-f317-6157a8421d30)

### Describe the bug

After emerging the gentoo driver for nvidia with the `kernel-open` flag and restarted, I couldn't open lutris anymore.
From the dumped info, it seems like lutris can't recognise this GPU

Lutris log:
```
2022-06-02 19:26:26,589: No GPU available on this system!
2022-06-02 19:26:27,273: Starting Lutris 0.5.10.1
2022-06-02 19:26:27,399: Writing full game list from MAME to /home/therealone/.cache/lutris/
2022-06-02 19:26:27,402: Couldn't get any output for mame -listxml
2022-06-02 19:26:27,402: MAME did not write anything to /home/therealone/.cache/lutris/mame/
2022-06-02 19:26:27,404: Using NVIDIA drivers for for Open
2022-06-02 19:26:27,404: GPU: NVIDIA GeForce RTX 2060
2022-06-02 19:26:27,404: No GPU available on this system!
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/lutris/gui/application.py", line 409, in do_command
    run_all_checks()
  File "/usr/lib/python3.9/site-packages/lutris/startup.py", line 156, in run_all_checks
    check_driver()
  File "/usr/lib/python3.9/site-packages/lutris/startup.py", line 81, in check_driver
    if drivers.is_outdated():
  File "/usr/lib/python3.9/site-packages/lutris/util/graphics/drivers.py", line 112, in is_o
    major_version = int(driver_version.split(".")[0])
ValueError: invalid literal for int() with base 10: 'for'
2022-06-02 19:26:27,405: Shutting down Lutris
```

### To Reproduce

1. Emerge version 515.48.07 of x11-drivers/nvidia-drivers with the `kernel-open` flag, restart and login
2. Open lutris
3. Notice it doesn't open so open a terminal
4. Run `lutris` and get this dump error info:
```
2022-06-02 19:26:26,589: No GPU available on this system!
2022-06-02 19:26:27,273: Starting Lutris 0.5.10.1
2022-06-02 19:26:27,399: Writing full game list from MAME to /home/therealone/.cache/lutris/
2022-06-02 19:26:27,402: Couldn't get any output for mame -listxml
2022-06-02 19:26:27,402: MAME did not write anything to /home/therealone/.cache/lutris/mame/
2022-06-02 19:26:27,404: Using NVIDIA drivers for for Open
2022-06-02 19:26:27,404: GPU: NVIDIA GeForce RTX 2060
2022-06-02 19:26:27,404: No GPU available on this system!
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/lutris/gui/application.py", line 409, in do_command
    run_all_checks()
  File "/usr/lib/python3.9/site-packages/lutris/startup.py", line 156, in run_all_checks
    check_driver()
  File "/usr/lib/python3.9/site-packages/lutris/startup.py", line 81, in check_driver
    if drivers.is_outdated():
  File "/usr/lib/python3.9/site-packages/lutris/util/graphics/drivers.py", line 112, in is_o
    major_version = int(driver_version.split(".")[0])
ValueError: invalid literal for int() with base 10: 'for'
2022-06-02 19:26:27,405: Shutting down Lutris
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8825860/nvidia-bug-report.log.gz)

Santiblocks, 2022-06-04:
As a workaround, you can patch startup.py on /usr/lib/python3.9/site-packages/lutris/startup.py to not do 'run_all_checks()'

maximilianovermeyer, 2022-06-07:
This is just a bug in lutris:
https://github.com/lutris/lutris/issues/4290

TheRealOne78, 2022-06-07:
So basically the proprietary version would be `NVIDIA UNIX Kernel Module for x86_64 515.43.04` and the open source version would be `NVIDIA UNIX Open Kernel Module for x86_64 515.43.04`?

If yes, what would happen if instead of `NVIDIA UNIX Open Kernel Module for x86_64 515.43.04` it would be `NVIDIA UNIX Kernel Module for x86_64 515.43.04-open` or add another number character to the version number that would differentiate the propritary kernel from the open kernel? For example add a 0 at the begining `0515.43.04` or something like this? Or add at the end of the version a .0 in case some programs requires a version or later: `515.43.04.0`

Honestly, changing completly the version scheme sounds like a bad idea, but in the same time the version scheme was already changed by nvidia and it is a real issue, that in my opinion, it could be partially solved by thinking how generally developers write their code.

aritger, 2022-06-08:
Sorry that the version format here tripped the lutris version parser.  It sounds like this is resolved in top of tree of lutris.  I'm going to go ahead and close this Issue.  Please reopen it if the fix in lutris is not sufficient.

closed 2022-06-08 by aritger