NVIDIA/open-gpu-kernel-modules #281: improper resolution passed to monitor

jumpers775, 2022-06-03:
### NVIDIA Open GPU Kernel Modules Version

515.48.07

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

5.18.1-arch1-1

### Hardware: GPU

NVIDIA GeForce RTX 2060 (UUID: GPU-27024b8f-ce0b-2546-7f78-676a22fa9c42)

### Describe the bug

When I set the resolution to 75 hz my monitor says input not supported. this is because my monitor is only capable of displaying exactly what is shown in its edid. I am using an acer sb220q bi. this issue first occured in the proprietary drivers with the 510 line. not entirely sure why this is, but im more than happy to provide an additional information.

### To Reproduce

connect an acer sb220q bi (or likely other acer monitors that lock the resolutions to whats in the edid), and set the resolution to 75 hz.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8835324/nvidia-bug-report.log.gz)

aritger, 2022-06-08:
Thanks for the report.  Could you please capture the edid.bin file for your monitor and attach it here?  Using nvidia-settings, go to the display page, and click on "Acquire EDID..."

jumpers775, 2022-06-08:
well, i usually run wayland, but it seems im getting `fish: Job 1, 'nvidia-settings -V' terminated by signal SIGSEGV (Address boundary error)` when running nvidia-settings? anyway, i have also attached the [edid](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8865477/edid.zip), which it wouldnt let me upload without compressing.

ThisNekoGuy, 2022-08-03:
@jumpers775 I don't think `nvidia-settings` supports wayland yet; other important functionality such as changing the fan speed and overclocking isn't available under wayland yet either, so that isn't a surprise

On a side note, though not directly related...
Being able to have literally any settings under Wayland would be great; assuming we can even get Wayland to work right first... that hasn't been great, personally, for things like Window stability, [proper HiDPI resolution support](https://bugs.kde.org/show_bug.cgi?id=447217), and [correct framerates](https://bugs.kde.org/show_bug.cgi?id=442575) but I haven't tested these for some months

ValZapod, 2022-08-03:
EDID decoded using https://git.linuxtv.org/edid-decode.git/ (it has quite strange 1920x1080   **74.972503** Hz  16:9     83.894 kHz    174.500000 MHz (476 mm x 268 mm))

```
00 ff ff ff ff ff ff 00 04 72 ab 06 ab 32 00 02
14 1e 01 03 80 30 1b 78 ee 5a 65 a7 56 52 9c 27
0f 50 54 b3 0c 00 71 4f 81 80 81 c0 81 00 95 00
b3 00 d1 c0 01 01 2a 44 80 a0 70 38 27 40 30 20
35 00 dc 0c 11 00 00 1a 02 3a 80 18 71 38 2d 40
58 2c 45 00 dc 0c 11 00 00 1e 00 00 00 fd 00 30
4c 1f 54 12 00 0a 20 20 20 20 20 20 00 00 00 fc
00 53 42 32 32 30 51 0a 20 20 20 20 20 20 01 f0

02 03 1f b1 4b 90 01 02 03 04 05 11 12 13 14 1f
65 03 0c 00 10 00 68 1a 00 00 01 01 30 4b e6 02
3a 80 18 71 38 2d 40 58 2c 45 00 dc 0b 11 00 00
1e 01 1d 00 72 51 d0 1e 20 6e 28 55 00 dc 0b 11
00 00 1e 8c 0a d0 8a 20 e0 2d 10 10 3e 96 00 dc
0b 11 00 00 18 2a 44 80 a0 70 38 27 40 30 20 35
00 dc 0b 11 00 00 1a 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 61

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: ACR
    Model: 1707
    Serial Number: 33567403
    Made in: week 20 of 2020
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 48 cm x 27 cm
    Gamma: 2.20
    DPMS levels: Standby Suspend Off
    RGB color display
    Default (sRGB) color space is primary color space
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6533, 0.3369
    Green: 0.3222, 0.6113
    Blue : 0.1533, 0.0605
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    Apple   :   640x480    66.666667 Hz   4:3     35.000 kHz     30.240000 MHz
    DMT 0x08:   800x600    56.250000 Hz   4:3     35.156 kHz     36.000000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x11:  1024x768    70.069359 Hz   4:3     56.476 kHz     75.000000 MHz
  Standard Timings:
    DMT 0x15:  1152x864    75.000000 Hz   4:3     67.500 kHz    108.000000 MHz
    DMT 0x23:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz
    DMT 0x55:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz
    DMT 0x1c:  1280x800    59.810326 Hz  16:10    49.702 kHz     83.500000 MHz
    DMT 0x2f:  1440x900    59.887445 Hz  16:10    55.935 kHz    106.500000 MHz
    DMT 0x3a:  1680x1050   59.954250 Hz  16:10    65.290 kHz    146.250000 MHz
    DMT 0x52:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz
  Detailed Timing Descriptors:
    DTD 1:  1920x1080   74.972503 Hz  16:9     83.894 kHz    174.500000 MHz (476 mm x 268 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   5 Vback   31 Vpol N
    DTD 2:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz (476 mm x 268 mm)
                 Hfront   88 Hsync  44 Hback  148 Hpol P
                 Vfront    4 Vsync   5 Vback   36 Vpol P
    Display Range Limits:
      Monitor ranges (GTF): 48-76 Hz V, 31-84 kHz H, max dotclock 180 MHz
    Display Product Name: 'SB220Q'
  Extension blocks: 1
[... 34 lines cut ...]
                 Hfront  110 Hsync  40 Hback  220 Hpol P
                 Vfront    5 Vsync   5 Vback   20 Vpol P
    DTD 5:   720x480    59.940060 Hz   3:2     31.469 kHz     27.000000 MHz (476 mm x 267 mm)
                 Hfront   16 Hsync  62 Hback   60 Hpol N
                 Vfront    9 Vsync   6 Vback   30 Vpol N
    DTD 6:  1920x1080   74.972503 Hz  16:9     83.894 kHz    174.500000 MHz (476 mm x 267 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   5 Vback   31 Vpol N
Checksum: 0x61  Unused space in Extension Block: 24 bytes

Warnings:

EDID:
  CTA-861: Multiple native progressive timings are defined.

Failures:

Block 0, Base EDID:
  Basic Display Parameters & Features: sRGB is signaled, but the chromaticities do not match.
Block 1, CTA-861 Extension Block:
  Missing VCDB, needed for Set Selectable RGB Quantization to avoid interop issues.
EDID:
  Base EDID: Some timings are out of range of the Monitor Ranges:
    Horizontal Freq: 28.125 - 83.894 kHz (Monitor: 31.000 - 84.000 kHz)
```

ThisNekoGuy, 2022-08-17:
@aritger On a similar note, because it sounds like `jumpers775` meant "framerate" rather than "resolution(?)," I also experience an issue with running Wayland where the framerate Hz is capped to half of my display's maximum refresh rate where this isn't an issue on X11/Xorg

On X11 I can run `3840x2160 @ 60 Hz` (the refresh rate of my display) but on Wayland `3840x2160` is capped at 30Hz (half of that of my display).
I mentioned this to the KDE devs over [here](https://bugs.kde.org/show_bug.cgi?id=457855), but I was told it was a driver bug.