NVIDIA/open-gpu-kernel-modules #285: Possible to build OS-independent kernel separately?

despair86, 2022-06-06:
### NVIDIA Open GPU Kernel Modules Version

965db

### Operating System and Version

N/A

### Kernel Release

SunOS sun-pc 5.11 i86pc i386 i86pc

### Build Command

N/A

### Terminal output/Build Log

N/A

### More Info

I have extracted the kernel interface code from the same corresponding release tag (515.48.07), placed it in https://github.com/despair86/open-gpu-kernel-modules for now, is it possible to build the platform independent kernel separately at this time?

despair86, 2022-06-06:
i'll come back to this in a bit

despair86, 2022-06-06:
this is now a functional issue at least, i was able to build the driver in two parts

closed 2022-06-06 by despair86

mtijanic, 2022-06-06:
Hey there. The driver in this repo shares a lot of code with the proprietary driver but they are _fundamentally_ different architectures, and it is not possible to make this code behave as the proprietary driver. It will not work without gsp.bin loaded to the GSP core.

If you wanted to port this code to sunos, the first step of importing the kernel-interface from the proprietary driver was good. However, now you need to implement all the OS-specific logic for fetching gsp.bin, which is currently only implemented in the linux kernel interface layer; implementing [these 3 functions](https://github.com/despair86/open-gpu-kernel-modules/blob/main/sunos-kmod/nvidia_subr.c#L1319-L1340) to return the gsp.bin payload (check `nv_get_firmware()` in `nv.c` for linux impl) _should_ be enough to make the driver attempt to initialize GSP.