NVIDIA/open-gpu-kernel-modules #290: NVreg EnableBrightnessControl causes xorg application crashes

tchofy, 2022-06-06:
### NVIDIA Open GPU Kernel Modules Version

515.48.07 965db985520c0c2aefa56dc2e81f9bfa67e91375

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Garuda Linux

### Kernel Release

Linux odyssey 5.18.1-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Mon, 30 May 2022 17:53:16 +0000 x86_64 GNU/Linux, pre compiled from arch repos

### Hardware: GPU

GPU 0: NVIDIA GeForce GTX 1650 (UUID: GPU-b60558fd-42d3-683a-441d-7dabf7fa7261)

### Describe the bug

Graphical applications under X.org don't work if the `nvidia` module is loaded with the `NVreg_RegistryDwords=EnableBrightnessControl=1` parameter, which exposes brightness settings through /sys/class/backlight/nvidia_0 on the proprietary driver for laptops. A wayland session will work, so the problem seems specific to xorg, but the nvidia_0 backlight interface is still missing.
If the parameter is removed and the modules are reloaded, it becomes possible to start an X session again, but I cannot control brightness through the newly exposed /sys/class/backlight/acpi_video0 interface, and the screen is stuck on max brightness.

I got the same results by using `NVIDIA[...].run -m=kernel-open` and the `nvidia-open-dkms` package from arch repos, and also tried the default kernel (`5.18.1.arch1-1`).

Looking through the source code, I couldn't find any references to `EnableBrightnessControl`, or by doing
 `grep -r EnableBrightnessControl /usr/src/nvidia-515.48.07/`, While on the proprietary drivers `nvidia-dkms 515.48.07-2` and under it could be found on the file /usr/src/nvidia-515.48.07/nvidia/nv-kernel.o_binary, so it seems to me that this feature isn't implemented in the open source modules.

journalctl -p3 -xb
[jctl.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8844692/jctl.txt)

### To Reproduce

- Add `nvidia.NVreg_RegistryDwords=EnableBrightnessControl=1` to kernel parameters
- Start an x session

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8844714/nvidia-bug-report.log.gz)

aritger, 2022-06-08:
Thanks for the report.  Could you check whether the same problem reproduces if you use the proprietary kernel modules, but force them to use GSP offload?  Set the nvidia.ko kernel module parameter NVreg_EnableGpuFirmware=1:

http://us.download.nvidia.com/XFree86/Linux-x86_64/515.48.07/README/gsp.html

The open kernel modules necessarily use GSP offload, but the proprietary kernel modules (for GeForce) don't yet use GSP offload by default.  This experiment will help us identify if the problem is in GSP offload (which seems likely), or with the open kernel modules themselves.

Thanks!

tchofy, 2022-06-08:
Ah, that seems to be the case. After rebooting with that parameter using `nvidia-dkms 515.48.07-2` I got stuck on the plymouth boot screen, but if I could Esc out of it fast enough before it loaded my DM I could drop into a tty.

I couldn't start a wayland session either, and attempting to generate a log using nvidia-bug-report.sh extracted from the .run would cause it to crash, even with the --safe-mode and --extra-system-data flags, and the resulting log didn't get past the first few checks on /sys/bus/pci/devices/0000:01:00.1/power/*.

I did manage to grab the following journalctl log manually:
[GSPoffload_jctl.txt](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8858270/GSPoffload_jctl.txt)

aritger, 2022-06-08:
Thanks for performing that experiment.  I've filed bug NVIDIA-internal bug 3676312 for this issue.  There are several different problems, which I think explain why the failure signature differs between the test with the open kernel modules versus the closed kernel modules.  But, they all have the same root cause.

I don't have a good suggestion for a work around for now... I think it will require a new release from NVIDIA.

aritger, 2023-05-30:
@tchofy: I believe this issue was addressed several releases ago.  Could you please retest with 535.43.02?

I'm going to optimistically close, but please reopen if you still see this with recent releases.