NVIDIA/open-gpu-kernel-modules #295: Higher temps when using opensource nvidia drivers

DaVikingMan, 2022-06-07:
### NVIDIA Open GPU Kernel Modules Version

515.48.07-3

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Arch Linux

### Kernel Release

5.18.2-arch1-1

### Hardware: GPU

GTX 1660ti

### Describe the bug

When using the opensource drivers, the system reports considerably higher gpu temps than the closed source ones. The closed source driver idles around 40C while the opensource one idles around 46C and this increases when doing higher-demanding tasks.

### To Reproduce

1. Install nvidia-open drivers
2. Run these drivers
3. Check output of nvidia-smi(or any other temperature application).

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8854072/nvidia-bug-report.log.gz)

KyodaiKen, 2022-06-09:
The blob driver also has a similar issue on Linux. For my RTX 3090 STRIX OC with Driver version 510 installed from the package downloaded from nvidia.com keeps the memory at full base clock, which causes the card to report 110+ W idle power draw. I also see this in the amount of heat coming out of my radiators.

I have a 1440p and a 4K screen both with 144Hz connected via DisplayPort. The 1440p one has FreeSync enabled and the other one G-Sync. The idle power draw is about 43 Watts. I did the same on Linux, but there it's pulling over 110 Watts.

lankylonky22, 2022-06-19:
Plz fix it asap, it literally renders Nvidia-open unusable