NVIDIA/open-gpu-kernel-modules #300: symbol lookup error: /usr/lib/libnvidia-vulkan-producer.so: undefined symbol: wlEglInitializeSurfaceExport

thesword53, 2022-06-11:
### NVIDIA Open GPU Kernel Modules Version

nvidia 515.48.07

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

5.18.1-arch1-1

### Hardware: GPU

GeForce RTX 2080 SUPER

### Describe the bug

Swapchain creation with `VK_KHR_wayland_surface` results to program termination with error `symbol lookup error: /usr/lib/libnvidia-vulkan-producer.so: undefined symbol: wlEglInitializeSurfaceExport`

### To Reproduce

Run any Vulkan application in native Wayland (No Xwayland)

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8883756/nvidia-bug-report.log.gz)

jp7677, 2022-06-11:
See https://github.com/NVIDIA/egl-wayland/commit/fee26e9cbd0890bab890f4534d7e92e78d8d0d37#commitcomment-75822177

aritger, 2022-06-13:
Yes, thanks to the pointer to the regressing egl-wayland commit.  Since the bug isn't in the open kernel modules, I'm going to go ahead and close this Issue.

closed 2022-06-13 by aritger