NVIDIA/open-gpu-kernel-modules #302: [BUG] Some electron apps don't start unless root

TheRealOne78, 2022-06-12:
### NVIDIA Open GPU Kernel Modules Version

515.48.07

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Gentoo Base System release 2.8

### Kernel Release

5.18.1-gentoo-r1 custom

### Hardware: GPU

NVIDIA GeForce RTX 2060 (UUID: GPU-466269ac-80e7-d5cf-f317-6157a8421d30)

### Describe the bug

Some electron-based programs like diospiroverde/WazzApp or balena-io/etcher don't start unless starting as root (with su or sudo/doas).
The following error appears:
Wazzapp:
```
[14021:0612/144001.874060:FATAL:gpu_data_manager_impl_private.cc(445)] GPU process isn't usable. Goodbye.
/bin/whatsapp: line 2: 14021 Trace/breakpoint trap   /opt/wazzapp/wazzapp
```

Balena-etcher:
```
[14608:0612/144133.601584:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenaBhOt7B/balena-etcher-electron: line 10: 14608 Trace/breakpoint trap   "${script_dir}"/balena-etcher-electron.bin "$@"
```

### To Reproduce

1. Start the executables

2.1 In wazzapp, it will spawn a gtk window then kill, and output this:
```
❯ whatsapp
/bin/sh: line 1: update-alternatives: command not found
(node:15147) electron: The default of contextIsolation is deprecated and will be changing from false to true in a future release of Electron.  See https://github.com/electron/electron/issues/23506 for more information
[15147:0612/144254.448252:FATAL:gpu_data_manager_impl_private.cc(445)] GPU process isn't usable. Goodbye.
/bin/whatsapp: line 2: 15147 Trace/breakpoint trap   /opt/wazzapp/wazzapp
```
2.2 Balena ethcer won't spawn anything and will output this:
```
❯ ./balenaEtcher-1.5.115-x64.AppImage
{"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1321:34)\n    at TLSSocket.emit (events.js:223:5)\n    at TLSSocket._finishInit (_tls_wrap.js:794:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:608:12)","config":{"url":"https://balena.io/etcher/static/config.json","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.5.115"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"json","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1},"code":"CERT_HAS_EXPIRED"}
[15547:0612/144351.896416:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
/tmp/.mount_balena6hG7pm/balena-etcher-electron: line 10: 15547 Trace/breakpoint trap   "${script_dir}"/balena-etcher-electron.bin "$@"
``` 

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/8885746/nvidia-bug-report.log.gz)

### More Info

Running balena as root will just give this harmfull warning message:
```
{"message":"certificate has expired","stack":"Error: certificate has expired\n    at TLSSocket.onConnectSecure (_tls_wrap.js:1321:34)\n    at TLSSocket.emit (events.js:223:5)\n    at TLSSocket._finishInit (_tls_wrap.js:794:8)\n    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:608:12)","config":{"url":"https://balena.io/etcher/static/config.json","method":"get","headers":{"Accept":"application/json, text/plain, */*","User-Agent":"axios/1.5.115"},"transformRequest":[null],"transformResponse":[null],"timeout":0,"responseType":"json","xsrfCookieName":"XSRF-TOKEN","xsrfHeaderName":"X-XSRF-TOKEN","maxContentLength":-1,"maxBodyLength":-1},"code":"CERT_HAS_EXPIRED"}
```

Running wazzapp as root without `--no-sandbox` won't work.

TheRealOne78, 2022-06-12:
Discord works fine as expected even if it is electron-based.
I will put in this issue chat more electron-based programs that don't work

tchofy, 2022-06-12:
Something I can add to this, it happens on the proprietary drivers too, and the issue seems to come and go randomly for me, even after going through a clean reboot.

Another workaround to launch it without root, is by using either the `--disable-gpu-sandbox` or `--no-sandbox` flag.
This seems to happen very often with apps using electron12 (Discord uses 13) or older, so it seems more like an electron issue for me.

TheRealOne78, 2022-06-12:
I haven't tested this in the proprietary version of this version but I am pretty sure it doesn't happen to the older nvidia versions (before open sourcing). Users that did not test this with the new driver. There aren't bugs from users experiencing this with the old version.

I also tested balena-etcher way before nvidia open-kernel in the same system, worked as expected.

TheRealOne78, 2022-06-12:
There is also this issue https://github.com/balena-io/etcher/issues/3770

aaronp24, 2022-06-12:
Is there a way to tell what, exactly, is tripping the sandbox protection? It's likely that Electron just needs to update its sandboxing rules.

TheRealOne78, 2022-06-12:
> Is there a way to tell what, exactly, is tripping the sandbox protection? It's likely that Electron just needs to update its sandboxing rules.

@tchofy just mentioned that Electron already solved this in the latest releases of Electron. However programs that still use old version of electron like version 12, would not work, and would need updating.

Are you suggesting that older versions of the nvidia driver accepted a bug in Electron12 and older because of an unintended bug in the older nvidia driver, and a handful ammount of programs needs updating to Electron13 or newer, 
or that handfull amount of programs needs updating to electron13 because of a new bug inside the open source nvidia driver?

aaronp24, 2022-06-12:
This is a often problem with sandboxing software: they generate a list of allowed commands by tracing what an application does, and then allow only those commands. When the system software such as the NVIDIA driver or Mesa evolve to require new features, the sandbox needs to be updated to allow them.

It sounds like the right fix here is for applications to update to Electron 13, or to somehow update Electron 12 to have the same sandboxing configuration as Electron 13. I'm not particularly familiar with Electron so hopefully Electron 12 can be updated in one place rather than having to update a separate copy embedded in every application...

TheRealOne78, 2022-06-13:
Well it looks like Electron12 and below are not suported anymore, so the only way would be to update every single existent electron12 and below to something much newer.
https://github.com/electron/electron/issues/34522

closed 2022-06-13 by TheRealOne78