NVIDIA/open-gpu-kernel-modules #311: Install from src error

wukong1992, 2022-06-20:
### NVIDIA Open GPU Kernel Modules Version

tag: 515.48.07

### Operating System and Version

CentOS Linux release 7.3.1611 (Core) 

### Kernel Release

4.9.2-3.0.0.std7b.el7.5.x86_64 #1 SMP Thu Apr 26 17:33:02 CST 2018 x86_64 x86_64 x86_64 GNU/Linux

### Build Command

make modules_install -j`nproc` NV_VERBOSE=1

### Terminal output/Build Log

test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video
make -f /usr/src/kernels/4.9.2-3.0.0.std7b.el7.5.x86_64/scripts/Makefile.modinst
  mkdir -p /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; cp /root/jsm/open-gpu-kernel-modules/kernel-open/nvidia-drm.ko /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-drm.ko ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-drm.ko || true && true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-drm.ko
  mkdir -p /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; cp /root/jsm/open-gpu-kernel-modules/kernel-open/nvidia-modeset.ko /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-modeset.ko ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-modeset.ko || true && true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-modeset.ko
  mkdir -p /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; cp /root/jsm/open-gpu-kernel-modules/kernel-open/nvidia-peermem.ko /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-peermem.ko ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-peermem.ko || true && true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-peermem.ko
  mkdir -p /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; cp /root/jsm/open-gpu-kernel-modules/kernel-open/nvidia-uvm.ko /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-uvm.ko ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-uvm.ko || true && true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia-uvm.ko
  mkdir -p /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; cp /root/jsm/open-gpu-kernel-modules/kernel-open/nvidia.ko /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia.ko ; true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia.ko || true && true /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia.ko
  /bin/sh /usr/src/kernels/4.9.2-3.0.0.std7b.el7.5.x86_64/scripts/depmod.sh /sbin/depmod 4.9.2-3.0.0.std7b.el7.5.x86_64 ""

### More Info

I reboot, and install driver: 'sh NVIDIA-Linux-x86_64-515.48.07.run --no-kernel-modules' but the output of 'nvidia-smi': NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.

wukong1992, 2022-06-20:
I find the files ' include/generated/autoconf.h' and 'include/config/auto.conf' all exist.

wukong1992, 2022-06-21:
`ilename:       /lib/modules/4.9.2-3.0.0.std7b.el7.5.x86_64/kernel/drivers/video/nvidia.ko
firmware:       nvidia/515.48.07/gsp.bin
alias:          char-major-195-*
version:        515.48.07
supported:      external
license:        Dual MIT/GPL
srcversion:     FAF867C627FC8DEDD45D08A
alias:          pci:v000010DEd*sv*sd*bc06sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
depends:        drm
vermagic:       4.9.2-3.0.0.std7b.el7.5.x86_64 SMP mod_unload 
parm:           NvSwitchRegDwords:NvSwitch regkey (charp)
parm:           NvSwitchBlacklist:NvSwitchBlacklist=uuid[,uuid...] (charp)
parm:           NVreg_ResmanDebugLevel:int
parm:           NVreg_RmLogonRC:int
parm:           NVreg_ModifyDeviceFiles:int
parm:           NVreg_DeviceFileUID:int
parm:           NVreg_DeviceFileGID:int
parm:           NVreg_DeviceFileMode:int
parm:           NVreg_InitializeSystemMemoryAllocations:int
parm:           NVreg_UsePageAttributeTable:int
parm:           NVreg_EnablePCIeGen3:int
parm:           NVreg_EnableMSI:int
parm:           NVreg_TCEBypassMode:int
parm:           NVreg_EnableStreamMemOPs:int
parm:           NVreg_RestrictProfilingToAdminUsers:int
parm:           NVreg_PreserveVideoMemoryAllocations:int
parm:           NVreg_EnableS0ixPowerManagement:int
parm:           NVreg_S0ixPowerManagementVideoMemoryThreshold:int
parm:           NVreg_DynamicPowerManagement:int
parm:           NVreg_DynamicPowerManagementVideoMemoryThreshold:int
parm:           NVreg_EnableGpuFirmware:int
parm:           NVreg_EnableGpuFirmwareLogs:int
parm:           NVreg_OpenRmEnableUnsupportedGpus:int
parm:           NVreg_EnableUserNUMAManagement:int
parm:           NVreg_MemoryPoolSize:int
parm:           NVreg_KMallocHeapMaxSize:int
parm:           NVreg_VMallocHeapMaxSize:int
parm:           NVreg_IgnoreMMIOCheck:int
parm:           NVreg_NvLinkDisable:int
parm:           NVreg_EnablePCIERelaxedOrderingMode:int
parm:           NVreg_RegisterPCIDriver:int
parm:           NVreg_EnableDbgBreakpoint:int
parm:           NVreg_RegistryDwords:charp
parm:           NVreg_RegistryDwordsPerDevice:charp
parm:           NVreg_RmMsg:charp
parm:           NVreg_GpuBlacklist:charp
parm:           NVreg_TemporaryFilePath:charp
parm:           NVreg_ExcludedGpus:charp
parm:           NVreg_DmaRemapPeerMmio:int
parm:           rm_firmware_active:charp
`

aritger, 2022-06-29:
@wukong1992 : can you please reproduce the problem again, run `nvidia-bug-report.sh` and attach the resulting log file here?  In particular, I'm interested in your dmesg output.  That may tell us why nvidia-smi isn't able to talk to nvidia.ko.

aritger, 2023-05-30:
@wukong1992 : can you please reproduce the problem again with the latest release (535.43.02), run nvidia-bug-report.sh and attach the resulting log file here? In particular, I'm interested in your dmesg output. That may tell us why nvidia-smi isn't able to talk to nvidia.ko.