NVIDIA/open-gpu-kernel-modules #318: Flickering in sway with 515.57 nvidia driver

nvidiaLinuxUser, 2022-07-05:
### NVIDIA Open GPU Kernel Modules Version

515.57-3

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

5.18.9-arch1-1

### Hardware: GPU

GPU 0: NVIDIA GeForce GTX 1660 SUPER (UUID: GPU-fed1a529-23f1-0db4-87bf-93b5a2c6ffc6)

### Describe the bug

sway flickers.

### To Reproduce

Install sway with the latest NVIDIA open source drivers and run `cat /dev/urandom` in the terminal to see the artifacts

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9043719/nvidia-bug-report.log.gz)

### More Info

After updating the NVIDIA open source driver to version 515.57, I started seeing flickering. I even reinstalled Arch Linux but that didn’t help. GSP firmware is loaded. This is strange, because on 515.48.07 sway worked perfectly, there were no flickers.

https://forums.developer.nvidia.com/t/flickering-in-sway-with-515-57-nvidia-driver/219587

niv, 2022-07-06:
Thank you for the bug report. Tracking internally in bug 3707157.

uamind, 2022-07-11:
Me too,my arch upgrade 515.57 vulkan isn't working

nvidiaLinuxUser, 2022-07-17:
I can also confirm that when I run games through Proton (Xwayland) I see these flickers in cutscenes.

ronymuniz, 2022-07-20:
Usage [Linux: OpenSuse Tumbleweed]
After updating the NVIDIA open source driver to version 515.57, I immediately had a frozen black screen issue during boot (see illustrative image in the link: https://i0.wp.com/windows11tools.com/wp-content/uploads/2022/01/image-17.png?resize=640%2C254&ssl=1). I even reinstalled Open Suse Tumbleweed, but that didn't help. This is strange, because on 515.48.07 everything worked perfectly.

nvidiaLinuxUser, 2022-08-02:
515.65.01, nothing has changed...

rsmunix, 2022-08-04:
driver 515.65.01 I haven't tested it yet.  Did it really not work for you?

😟

nvidiaLinuxUser, 2022-08-04:
@rsmunix Unfortunately yes, sway is still flickering

nvidiaLinuxUser, 2022-09-05:
The flicker seems to disappear when you run the git version of sway with WLR_RENDERER=vulkan

critkitten, 2022-09-15:
> The flicker seems to disappear when you run the git version of sway with WLR_RENDERER=vulkan. Please fix it. UPDATE: Cursor sometimes freezes with WLR_RENDERER=vulkan UPDATE2: grim is not working:
> 
> ```
> failed to copy output HDMI-A-1
> ```

the grim thing is wlroots/vulkan issue https://gitlab.freedesktop.org/wlroots/wlroots/-/issues/3290
same with amdgpu if sway use vulkan renderer.

nvidiaLinuxUser, 2022-09-23:
When will wlroots work properly with NVIDIA?
?

nvidiaLinuxUser, 2022-10-09:
https://www.reddit.com/r/swaywm/comments/xyy14s/how_to_prevent_flickering_on_nvidia_gpus/

nvidiaLinuxUser, 2022-11-12:
525.53, the problem has not been fixed.

Crystal4276, 2022-11-20:
> 525.53, the problem still remains.

Hello, @niv 
I was wondering if this bug 3707157 is very far at the bottom of NVIDIA priorities, or if there are some investigations already on-going  ?

MahouShoujoMivutilde, 2022-12-12:
I am on https://github.com/hyprwm/Hyprland/commit/e8b99ae13a0525f9fbd6899fe39fe723879f86b9 (wlroots based compositor), and I had random flickering when running without the [above mentioned wlroots patch](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/318#issuecomment-1280820195), but since `525.60.11` of **proprietary** driver I don't need the patch anymore - _it works flawlessly even without it._

Can't test on opensource driver (old gpu), but since the readme says

> This code base is shared with NVIDIA's proprietary drivers

I thought it might be relevant.

Crystal4276, 2022-12-13:
I am on `525.60.11` with an RTX 2080, the mouse cursor is still disappearing upon opening an app.

nvidiaLinuxUser, 2022-12-13:
@MahouShoujoMivutilde, the flickering has indeed disappeared in wlroots git version, but the cursor is still invisible. Thanks for the report.

evofromko, 2023-01-12:
Same problems with Qtile + Wayland + nVidia 3070Ti. Flickering and no mouse cursor.  ( nVidia 525.78.01-2 on Arch Linux )

Fxzzi, 2023-02-01:
Same issues described here

GrabbenD, 2023-03-23:
Nvidia needs to add support for missing GBM protocols to resolve performance issues and adress missing features like 10bit support with wlroots based compositors such as Sway, Hyprland and so on.

I've been using NVIDIA for ages but I'm starting to consider if AMD could be a better choice to get proper hardware acceleration in Wayland since I'm tired of using Windows and Linux is just perfect for work

oifj34f34f, 2023-03-30:
sway-git & nvidia 530.41.03 — nothing has changed...

Crystal4276, 2023-10-16:
> This bug is still not fixed, after almost 1 year. Hopefully nouveau will get GSP firmware support this year...

Yep, at this point i'm also considering Nouveau...

oifj34f34f, 2023-12-09:
I just switched to using Nouveau driver. Hopefully its performance will improve with the latest news.

Sivecano, 2024-01-07:
stuff's still not working. at this point I'm literally just telling people to stay away from nvidia GPUs....

Crystal4276, 2024-01-10:
> I just switched to using Nouveau driver. Hopefully its performance will improve with the latest news.
> 
> UPD: https://www.phoronix.com/news/Nouveau-NVK-One-Win :-)

Could you test it ? 
Does it fix the issue with wlroots ?

Crystal4276, 2024-01-10:
> The problem with nouveau is the lack of hardware acceleration on new cards. I tried it last week, but couldn't retain due to high cpu loads.

you need to install kernel 6.7 when available.

oifj34f34f, 2024-01-10:
> you need to install kernel 6.7 when available.

And add to `nouveau.config=NvGspRm=1` to /etc/default/grub

oifj34f34f, 2024-01-10:
> @Crystal4276 I think it's mainline now. @oifj34f34f You tried it ? How's the performance ?

Well, I haven't tested any games except GZDoom so I can't say anything. However, performance has improved a bit (e.g. in the browser)

oifj34f34f, 2024-01-10:
> What about video playback in browser (also in mpv) ?

AFAIK, NVDEC still only works with proprietary drivers

oifj34f34f, 2024-01-10:
> > > What about video playback in browser (also in mpv) ?
> > 
> > 
> > AFAIK, NVDEC still only works with proprietary drivers
> 
> I mean't [vulkan playback btw](https://github.com/mpv-player/mpv/issues/11739).

Ah, haven't tried it yet