NVIDIA/open-gpu-kernel-modules #322: Cannot build successfully

wenhaopang, 2022-07-09:
### NVIDIA Open GPU Kernel Modules Version

515.49.06

### Operating System and Version

Ubuntu 18.04.6 LTS

### Kernel Release

5.4.0-91-generic    x86_64 GNU/Linux

### Build Command

I followed the installations on the readme file

`sudo sh ./NVIDIA-Linux-x86_64-515.49.06.run --no-kernel-modules`
`make modules`
and
`sudo make modules_install`

### Terminal output/Build Log

This is the output after running `sudo make modules_install`

```
make -C kernel-open modules_install
make[1]: Entering directory '/home/.../open-gpu-kernel-modules-515.49.06/kernel-open'
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-91-generic'
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-drm.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-modeset.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-peermem.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-uvm.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia.ko
  DEPMOD  5.4.0-91-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-91-generic'
make[1]: Leaving directory '/home/.../open-gpu-kernel-modules-515.49.06/kernel-open'
```

### More Info

my question is after running all the commands，driver was not installed successfully
when I running `nvidia-smi`
output：
`NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.`

YusufKhan-gamedev, 2022-07-09:
Try running `lsmod | grep nvidia` ?

wenhaopang, 2022-07-10:
> Try running `lsmod | grep nvidia` ?
> 
> Maybe its a nvidia-smi issue. Also I notice that the version you give is not supported by the modules...

No output after running `lsmod | grep nvidia`

Which version. NVIDIA-Linux-x86_64-515.49.06.run？
And I tried to install _open-gpu-kernel-modules-515.48.07_ and _NVIDIA-Linux-x86_64-515.48.07.run_ using steps above，and then running `nvidia-smi` , but no output.

wenhaopang, 2022-07-10:
> Try running `lsmod | grep nvidia` ?
> 
> Maybe its a nvidia-smi issue. Also I notice that the version you give is not supported by the modules...

Thanks for your answer.

No output after running `lsmod | grep nvidia`

Which version?  NVIDIA-Linux-x86_64-515.49.06.run？
And I tried to install _open-gpu-kernel-modules-515.48.07_ and _NVIDIA-Linux-x86_64-515.48.07.run_ using the steps above，and then running nvidia-smi , but no output.

YusufKhan-gamedev, 2022-07-10:
`make -C kernel-open modules_install`

This doesnt look like whats supposed to be done :( ditch `-C kernel-open` and follow the instructions on the main page. Also, if you ran `-C kernel-open` when building, thats not how its supposed to be done, your supposed to build everything, `src/`, and `kernel-open`, you cant pick and chose or else it wont work at all.

pendragons-code, 2022-07-11:
To be fair tho he is running an older version of ubuntu

niv, 2022-07-11:
* Did the module even try to load?
* If not, can you try to modprobe/insmod it?
* Anything in `dmesg`?

wenhaopang, 2022-07-12:
> `make -C kernel-open modules_install`
>
> This doesnt look like whats supposed to be done :( ditch `-C kernel-open` and follow the instructions on the main page. Also, if you ran `-C kernel-open` when [...]

When I run `sudo make modules_install`， `make -C kernel-open modules_install` appears.

wenhaopang, 2022-07-12:
> To be fair tho he is running an older version of ubuntu

I have tried running an older open-gpu-kernel-modules version，` open-gpu-kernel-modules-515.48.07`. 
Maybe I should try Ubuntu20.04

wenhaopang, 2022-07-12:
> To be fair tho he is running an older version of ubuntu

I have tried running an older open-gpu-kernel-modules version， _open-gpu-kernel-modules-515.48.07._
Maybe I should try Ubuntu20.04

wenhaopang, 2022-07-12:
> To be fair tho he is running an older version of ubuntu

I have tried running an older open-gpu-kernel-modules version， open-gpu-kernel-modules-515.48.07.
Maybe I should try Ubuntu20.04.

wenhaopang, 2022-07-12:
> @wenhaopang Please share nvidia bug report for triage purpose.

After I run `sudo make modules_install` , there are no other bugs except the following results
```
make -C kernel-open modules_install
make[1]: Entering directory '/home/.../open-gpu-kernel-modules-515.49.06/kernel-open'
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-91-generic'
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-drm.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-modeset.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-peermem.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-uvm.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia.ko
  DEPMOD  5.4.0-91-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-91-generic'
make[1]: Leaving directory '/home/.../open-gpu-kernel-modules-515.49.06/kernel-open'
```

wenhaopang, 2022-07-12:
> @wenhaopang Please share nvidia bug report for triage purpose.

After I run `sudo make modules_install` , there are no other bugs except the following results.

```
make -C kernel-open modules_install
make[1]: Entering directory '/home/.../open-gpu-kernel-modules-515.49.06/kernel-open'
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-91-generic'
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-drm.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-modeset.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-peermem.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia-uvm.ko
  INSTALL /home/.../open-gpu-kernel-modules-515.49.06/kernel-open/nvidia.ko
  DEPMOD  5.4.0-91-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-91-generic'
make[1]: Leaving directory '/home/.../open-gpu-kernel-modules-515.49.06/kernel-open'
```

YusufKhan-gamedev, 2022-07-12:
None of those are bugs, it *should* be installed...`sudo dmesg` would be nice, so would `tree /lib/modules | grep nv`

wenhaopang, 2022-07-16:
> None of those are bugs, it _should_ be installed...`sudo dmesg` would be nice, so would `tree /lib/modules | grep nv`

Sorry for not replying in time. 
How can I use `sudo dmesg` or `tree /lib/modules | grep nv` to solve the current problem.

YusufKhan-gamedev, 2022-07-16:
To see if anything is wrong with your system, or weird.

wenhaopang, 2022-07-19:
> To see if anything is wrong with your system, or weird.
> 
> Edit: and to see if you have it installed, but not working. Also when running the tree command, if it says `No such file or directory` you may have to use the `/lib/modules` directory that corresponds with your kernel.

After I run `sudo dmesg`, there is a long list of outputs. Here is part of the outputs:
```
[ 1157.036170] PKCS#7 signature not signed with a trusted key
[ 1157.043851] nvidia-nvlink: Nvlink Core is being initialized, major device number 237
[ 1157.046099] NVRM: The NVIDIA GPU 0000:af:00.0 (PCI ID: 10de:1db5)
               NVRM: installed in this system is not supported by open
               NVRM: nvidia.ko because it does not include the required GPU
               NVRM: System Processor (GSP).
               NVRM: Please see the 'Open Linux Kernel Modules' and 'GSP
               NVRM: Firmware' sections in the driver README, available on
               NVRM: the Linux graphics driver download page at
               NVRM: www.nvidia.com.
[ 1157.046221] nvidia: probe of 0000:af:00.0 failed with error -1
[ 1157.046241] NVRM: The NVIDIA probe routine failed for 1 device(s).
[ 1157.046242] NVRM: None of the NVIDIA devices were initialized.
[ 1157.046453] nvidia-nvlink: Unregistered Nvlink Core, major device number 237
```

And after I run tree command `tree /lib/modules | grep nvidia`. The following is the output.

│   │   │   │   │   ├── nvidia
│   │   │   │   │   ├── nvidia
│   │   │   │   │   │   └── nvidiafb.ko
│   │   │   │   ├── nvidia-drm.ko
│   │   │   │   ├── nvidia.ko
│   │   │   │   ├── nvidia-modeset.ko
│   │   │   │   ├── nvidia-peermem.ko
│   │   │   │   ├── nvidia-uvm.ko
│   │   │   │   ├── nvidia-uvm.ko.bak
    │   │   │   │   ├── nvidia
    │   │   │   │   ├── nvidia
    │   │   │   │   │   └── nvidiafb.ko

closed 2022-07-19 by aritger

wenhaopang, 2022-07-21:
Thanks all.