NVIDIA/open-gpu-kernel-modules #326: RmInitAdapter failed!

BahmanFarshbaf, 2022-07-20:
### NVIDIA Open GPU Kernel Modules Version

515.49.06

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Ubuntu 22.04 LTS

### Kernel Release

5.15.0-41-generic

### Hardware: GPU

RTX 3080 12GB 220A, nvidia-smi doesnt work!!

### Describe the bug

The nvidia-smi fails when used in combination with the Open-Source driver and proprietary driver!

### To Reproduce

wget https://download.nvidia.com/XFree86/Linux-x86_64/515.57/NVIDIA-Linux-x86_64-515.57.run

bash ./NVIDIA-Linux-x86_64-515.57.run --no-kernel-module -q -s -Z 

git clone https://github.com/NVIDIA/open-gpu-kernel-modules

cd open-gpu-kernel-modules

BUILD_EXCLUSIVE_KERNEL=$(uname -r) IGNORE_PREEMPT_RT_PRESENCE=1 make modules -j$(nproc)
BUILD_EXCLUSIVE_KERNEL=$(uname -r) IGNORE_PREEMPT_RT_PRESENCE=1 make modules_install -j$(nproc)

depmod -a; update-initramfs -u; reboot

lsmod | grep nvidia

nvidia_drm             69632  0
nvidia_modeset       1064960  1 nvidia_drm
nvidia               4501504  1 nvidia_modeset
drm_kms_helper        307200  1 nvidia_drm
drm                   606208  4 drm_kms_helper,nvidia,nvidia_drm

cat /proc/driver/nvidia/version

NVRM version:  NVIDIA UNIX Open Kernel Module for x86_64  515.57  Release Build Mi 20. Jul 11:53:35 CEST 2022
GCC version:  gcc version 11.2.0 (Ubuntu 11.2.0-19ubuntu1)

nvidia-smi --help | grep -e "^NVIDIA"

NVIDIA System Management Interface -- v515.57

nvidia-smi

Unable to determine the device handle for GPU 0000:06:00.0: Not Found

nvidia-debugdump --list

Found 1 NVIDIA devices
Error: nvmlDeviceGetHandleByIndex(): Not Found
FAILED to get details on GPU (0x0): Not Found

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9148884/nvidia-bug-report.log)

YusufKhan-gamedev, 2022-07-20:
`NVreg_OpenRmEnableUnsupportedGpus=1 `?

BahmanFarshbaf, 2022-07-21:
Fixed by updating bios!

closed 2022-07-21 by BahmanFarshbaf

ai-nikolai, 2022-12-04:
@BahmanFarshbaf thanks a lot for reporting this bug.

Could you please elaborate on what the necessary bios update was needed to fix the problem.

As a very similar is encountered here with:
```
NVIDIA Open GPU Kernel Modules Version
525.60.11

Operating System and Version
Ubuntu 22.04 LTS

Kernel Release
5.15.0-41-generic

Hardware: GPU
RTX 3080 12GB 220A, nvidia-smi doesnt work!!
```

ai-nikolai, 2022-12-04:
**Update from above:**
```
NVIDIA UNIX x86_64 Kernel Module  
525.60.11

Operating System and Version
Ubuntu 22.04 LTS

Kernel Release
5.15.0-41-generic

Hardware: GPU
RTX 3090 24GB 220A, nvidia-smi WORKS!
```
The way to fix it was to switch the drivers to the proprietary ones. (Ubuntu > Settings > About > Software Updates > Additional Drivers -> Switch To: `Using NVIDA driver metapackage from nivida-driver-525 (proprietary)`

YusufKhan-gamedev, 2022-12-04:
@ai-nikolai Create an issue

atommaki, 2023-05-01:
I ran into the same issue, Ubuntu 22.04, driver `530`, the binary driver works, the open gives me this error. My BIOS is up to date. Please reopen this issue.

```
NVIDIA UNIX x86_64 Kernel Module  
530.41.03

Operating System and Version
Ubuntu 22.04 LTS

Kernel Release
5.15.0-71-lowlatency

Hardware: GPU
GTX 1660 Super.
```