NVIDIA/open-gpu-kernel-modules #327: Build fails on Ubuntu 20.04 with kernel 5.18.12

flaviusburca, 2022-07-20:
### NVIDIA Open GPU Kernel Modules Version

branch 515.57

### Operating System and Version

Ubuntu 20.04.4 LTS

### Kernel Release

Linux flavius-pc 5.18.12 #1 SMP PREEMPT_DYNAMIC Tue Jul 19 21:01:48 EEST 2022 x86_64 x86_64 x86_64 GNU/Linux

### Build Command

make modules -j8 CC=gcc TARGET_ARCH=x86_64 NV_VERBOSE=1

### Terminal output/Build Log

  sh ./scripts/gen_ksymdeps.sh /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.o >> /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.nvidia-peermem.o.cmd
   { echo ; echo '/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.o: $(wildcard ./tools/objtool/objtool)' ; } >> /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.nvidia-peermem.o.cmd
  if objdump -h /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.o | grep -q __ksymtab; then gcc -E -D__GENKSYMS__ -Wp,-MMD,/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.nvidia-peermem.o.d  -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector-strong -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -I/home/flavius/open-gpu-kernel-modules/kernel-open/common/inc -I/home/flavius/open-gpu-kernel-modules/kernel-open -Wall -MD -Wno-cast-qual -Wno-error -Wno-format-extra-args -D__KERNEL__ -DMODULE -DNVRM -DNV_VERSION_STRING=\"515.57\" -Wno-unused-function -Wuninitialized -fno-strict-aliasing -ffreestanding -mno-red-zone -mcmodel=kernel -DNV_UVM_ENABLE -Werror=undef -DNV_SPECTRE_V2=0 -DNV_KERNEL_INTERFACE_LAYER -I/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem -UDEBUG -U_DEBUG -DNDEBUG -DNV_BUILD_MODULE_INSTANCES=0  -fsanitize=bounds -fsanitize=shift -fsanitize=bool -fsanitize=enum  -DMODULE  -DKBUILD_BASENAME='"nvidia_peermem"' -DKBUILD_MODNAME='"nvidia_peermem"' -D__KBUILD_MODNAME=kmod_nvidia_peermem /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.c | scripts/genksyms/genksyms    -r /dev/null > /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.tmp_nvidia-peermem.ver; ld  -m elf_x86_64 -r -o /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.tmp_nvidia-peermem.o /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.o -T /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.tmp_nvidia-peermem.ver; mv -f /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.tmp_nvidia-peermem.o /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.o; rm -f /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/.tmp_nvidia-peermem.ver; fi
  ld  -m elf_x86_64    -r -o /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem.o /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.o
  { echo  /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem/nvidia-peermem.o; nm /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem.o | awk '$1 == "U" { printf("%s%s", x++ ? " " : "", $2) }'; echo; } > /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem.mod
  {   echo /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia.ko;   echo /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-uvm.ko;   echo /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-modeset.ko;   echo /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-drm.ko;   echo /home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-peermem.ko; :; } | awk '!x[$0]++' - > /home/flavius/open-gpu-kernel-modules/kernel-open/modules.order
sh ./scripts/modules-check.sh /home/flavius/open-gpu-kernel-modules/kernel-open/modules.order
make -f ./scripts/Makefile.modpost
  sed 's/\.ko$/\.o/' /home/flavius/open-gpu-kernel-modules/kernel-open/modules.order | scripts/mod/modpost -m -a  -o /home/flavius/open-gpu-kernel-modules/kernel-open/Module.symvers -e -i Module.symvers   -T -
ERROR: modpost: "drm_gem_mmap_obj" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-drm.ko] undefined!
ERROR: modpost: "nvKmsKapiGetFunctionsTable" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-drm.ko] undefined!
ERROR: modpost: "nvidia_unregister_module" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-modeset.ko] undefined!
ERROR: modpost: "nvidia_get_rm_ops" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-modeset.ko] undefined!
ERROR: modpost: "nvidia_register_module" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-modeset.ko] undefined!
ERROR: modpost: "__x86_indirect_thunk_rbp" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-modeset.ko] undefined!
ERROR: modpost: "nvUvmInterfaceSessionCreate" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-uvm.ko] undefined!
ERROR: modpost: "nvUvmInterfacePmaAllocPages" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-uvm.ko] undefined!
ERROR: modpost: "nvUvmInterfaceDupAllocation" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-uvm.ko] undefined!
ERROR: modpost: "nvUvmInterfacePmaUnregisterEvictionCallbacks" [/home/flavius/open-gpu-kernel-modules/kernel-open/nvidia-uvm.ko] undefined!
WARNING: modpost: suppressed 64 unresolved symbol warnings because there were too many)
make[3]: *** [scripts/Makefile.modpost:134: /home/flavius/open-gpu-kernel-modules/kernel-open/Module.symvers] Error 1
make[3]: *** Deleting file '/home/flavius/open-gpu-kernel-modules/kernel-open/Module.symvers'
make[2]: *** [Makefile:1753: modules] Error 2
make[2]: Leaving directory '/usr/src/linux-5.18.12'

BahmanFarshbaf, 2022-07-20:
IGNORE_PREEMPT_RT_PRESENCE=1 ?

flaviusburca, 2022-07-21:
> IGNORE_PREEMPT_RT_PRESENCE=1 ?

the kernel does not have the PREEMPT_RT patch set