NVIDIA/open-gpu-kernel-modules #331: BUG: unable to handle page fault for address: ffff9fb6070ca000 when using compressed firmware

NickCao, 2022-07-23:
### NVIDIA Open GPU Kernel Modules Version

515.57

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

NixOS 22.11

### Kernel Release

5.18.12

### Hardware: GPU

NVIDIA GeForce GTX 1660 Ti

### Describe the bug

Kernel oops when using nvidia driver with compressed gsp firmware.

### To Reproduce

- `xz /lib/firmware/nvidia/<version>/gsp.bin`
- reboot

### Bug Incidence

Always

### nvidia-bug-report.log.gz

not applicable

### More Info

dmesg from open source driver: https://pb.nichi.co/damp-ocean-spice
dmesg from proprietary driver: https://pb.nichi.co/tower-agree-fine

mtijanic, 2022-07-25:
Hi there! Can you please verify that you have `CONFIG_FW_LOADER_COMPRESS` enabled? Try running:
```
grep CONFIG_FW_LOADER /boot/config-`uname -r`
```
Can you also try this with one of the other firmwares and see if they're loaded successfully?

The decompression is handled entirely by the kernel and the reason this is crashing is because `request_firmware()` is returning compressed data for whatever reason.

NickCao, 2022-07-25:
I can confirm `CONFIG_FW_LOADER_COMPRESS=y` is set, also NixOS compresses all firmwares by default, and `i915` loads firmwares just fine.

aritger, 2022-07-25:
It may also be good to print `fw->size` (just to make sure there isn't some mismatch... say decompressed data but compressed size).

NickCao, 2022-07-25:
```
[    3.776910] NVRM-TEST: Got firmware header: 7f 45 4c 46
```
decompression is fine, let me check the length.

NickCao, 2022-07-25:
```
NVRM-TEST: Got firmware size: 2025128
```
(formatted with %x)

mtijanic, 2022-07-25:
Yes, that's the correct size. I'm very confused about this. Think we'll have to try and repro internally..

mtijanic, 2022-07-25:
Sorry for the stupid question, but you _can_ boot successfully with the uncompressed `gsp.bin`? :smile:

NickCao, 2022-07-25:
yes, the uncompressed `gsp.bin` works fine:
(for non NixOS users: the patch below just rules nvidia firmwares from being compressed)
```diff
diff --git a/nixos/modules/services/hardware/udev.nix b/nixos/modules/services/hardware/udev.nix
index 2e9deebbb74..8732aac422f 100644
--- a/nixos/modules/services/hardware/udev.nix
+++ b/nixos/modules/services/hardware/udev.nix
@@ -171,10 +171,13 @@ let
       mv etc/udev/hwdb.bin $out
     '';

-  compressFirmware = if config.boot.kernelPackages.kernelAtLeast "5.3" then
-    pkgs.compressFirmwareXz
-  else
-    id;
+  compressFirmware = x:
+    if
+      config.boot.kernelPackages.kernelAtLeast "5.3" && !(lib.strings.hasPrefix "nvidia" x.name)
+    then
+      pkgs.compressFirmwareXz x
+    else
+      id x;

   # Udev has a 512-character limit for ENV{PATH}, so create a symlink
   # tree to work around this.
```

ValZapod, 2022-08-03:
> uncompressed `gsp.bin` works fine:

Why do you even compress the file? I am confused :) This is still extra cycles even if it somehow accelerates stuff like packers do (UPX packer)...

NickCao, 2022-08-03:
Compressing firmwares greatly reduces their sizes, on a typical NixOS system, it save us 500M.

colemickens, 2022-08-27:
@NickCao thanks for the patch - happy to have it, not so happy to need it; hopefully that changes soon!

NickCao, 2022-11-12:
This seems to no longer reproduce at the release of 525.53, not sure if it is related to the GSP firmware split.

ValZapod, 2022-11-12:
>GSP firmware split.

What is that?

NickCao, 2022-11-12:
> > GSP firmware split.
> 
> What is that?

GSP firmware is now distributed as multiple firmware files: this release has gsp_tu10x.bin and gsp_ad10x.bin replacing gsp.bin from previous releases.
See: https://github.com/NVIDIA/open-gpu-kernel-modules/releases/tag/525.53

ValZapod, 2022-11-12:
Got it.

mtijanic, 2022-11-12:
Thanks for the update @NickCao. I think this might be an upstream bug that only manifests with firmware above a certain size. Would be best to verify and file a bug upstream.

ValZapod, 2022-11-12:
>I think this might be an upstream bug that only manifests with firmware above a certain size.

Did you cut up the firmware because of that?

mtijanic, 2022-11-12:
> Did you cut up the firmware because of that?

No, but it's a nice collateral benefit :)
There's low level architectural differences between the two bin files.