NVIDIA/open-gpu-kernel-modules #333: High GPU power consumption when using NVDEC

thesword53, 2022-07-24:
### NVIDIA Open GPU Kernel Modules Version

515.57

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

5.18.14-arch1-1

### Hardware: GPU

GeForce RTX 2080 SUPER

### Describe the bug

GPU consumes on average 61W when I use NVDEC and only 35W with VDPAU.
I also noticed the GPU clock stays at 1650MHz with NVDEC. With VDPAU, the GPU clock is around 950MHz.

### To Reproduce

`mpv --hwdec=nvdec video.mp4` for NVDEC
`mpv --hwdec=vdpau-copy video.mp4` for VDPAU

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9176112/nvidia-bug-report.log.gz)

ValZapod, 2022-08-03:
You mean this? https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/261

Same happen on windows or not?

thesword53, 2022-08-03:
> You mean this? https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/261

Yes. GPU is stuck at P2 state when I use any CUDA application including NVDEC and NVENC (VDPAU and DXVA2/D3D11VA are not affected).
Even `nvidia-vaapi-driver` (https://github.com/elFarto/nvidia-vaapi-driver) is affected.

> Same happen on windows or not?

Yes, it also happens on Windows.

ValZapod, 2022-08-04:
> Yes, it also happens on Windows.

Not true. Just checked mpv.com --vf=format=gamma=pq:primaries=bt.2020:colorlevels=full --target-colorspace-hint=yes  -hwdec=nvdec --gpu-api=vulkan  -vo=gpu-next Z:\Raya.and.the.Last.Dragon.2021.2160p.DSNP.WEB-DL.DDP.5.1.Atmos.DoVi.by.DVT.mp4

After initial P2 in seconds it get to P5 and if you open some other app to P3 (in nvidia-smi.exe)

> | 41%   44C    P5    37W / 260W |   2403MiB / 11264MiB |     14%      Default |

So this is HEVC with Dolby Vision colorspace that is decoded using Vulkan shader with at least luma reshaping (doubt chroma reshaping happens, that is only from Apple TV+).

thesword53, 2022-08-04:
> Not true. Just checked mpv.com --vf=format=gamma=pq:primaries=bt.2020:colorlevels=full --target-colorspace-hint=yes -hwdec=nvdec --gpu-api=vulkan -vo=gpu-next Z:\Raya.and.the.Last.Dragon.2021.2160p.DSNP.WEB-DL.DDP.5.1.Atmos.DoVi.by.DVT.mp4

`-hwdec=nvdec` is not working on Windows: you are using software decoder. Try with `-hwdec=nvdec-copy` instead.
You can also try this command `ffmpeg -hwaccel cuda -i video.mp4 -an -c:v rawvideo -f null -` to test NVDEC.
And check on task manager the "Video Decode" graph to see if GPU is really used to decode video.

ValZapod, 2022-08-04:
> You can also try this command

You cannot, this decodes frames not as needed, but as fast as possible! Here P2/P0 is more than expected!

>-hwdec=nvdec is not working on Windows: you are using software decoder. 

**With d3d11 context, with vulkan it works, gpu-next added support for it some time ago.** Anyway, software decoder is always used with nvdec, what are talking about? Only hevc_cuvid does not use directly h26x decoder of ffmpeg, but even it uses it for probing, because it needs to first convert it to annex B format to send to decoder layer. Anyway, cuvid is only useful for ffplay and iterlaced hevc, and maybe some samples of H.264 that are not supported by h264 like with many reorderings as encoded by reference encoder. To say that NVDEC is not using software decoder is strange, since nvdec is just a HWACCEL for software decoder, so of course software decoder is in use.

>And check on task manager the "Video Decode" graph to see if GPU is really used to decode video.

I did. It works.

thesword53, 2022-08-05:
> To say that NVDEC is not using software decoder is strange, since nvdec is just a HWACCEL for software decoder, so of course software decoder is in use.

I neved said NVDEC is using software decoder. I said you are using software decoder: it meams you are not using NVDEC at all.
When I put `-hwdec=nvdec`, `mpv` doesn't print `Using hardware decoding (nvdec).`. Maybe because I use the latest stable version that doesn't support the `gpu-next` option.

ValZapod, 2022-08-05:
> latest stable version that doesn't support the `gpu-next` option.

-vo gpu is too old and too buggy. Just use normal source of mpv for windows. https://github.com/zhongfly/mpv-winbuild This is what developers use. Also you definition of stable is not correct. We in ffmpeg define lastest stable as HEAD of master, and in mpv I am not sure, but I think it is one of last tagged releases.

ValZapod, 2022-08-05:
> I neved said NVDEC is using software decoder.

But it does. You do not understand the basics of how it works.

thesword53, 2022-08-13:
> But it does. You do not understand the basics of how it works.

You don't undersand what I said.

> After initial P2 in seconds it gets to P5 and if you open some other app to P3 (in nvidia-smi.exe)

As you said this issue seems to be fixed on Windows but GPU doesn't go to P0 state: Only P2 state only.
But on Linux with both proprietary and open source driver GPU still stays at P2 state when I use NVENC or NVDEC.

ValZapod, 2022-08-19:
> As you said this issue seems to be fixed on Windows

No, it does go to P0 when you decode as fast as possible, i.e. what you proposed (slightly wrong on windows, pipe "-" is not supposed to be used, instead use NT kernel's abstraction NUL (this is like /dev/null or like .. and . in linux folders)): ffmpeg -hwaccel cuda -i video.mp4 -an -c:v rawvideo -f null NUL 

In fact you can get perfect 100% utilisation if you decode with -c:v h264_cuvid on a file with too much slices (usually Blu-ray has 4 slices, while that file I tested has 41).

BTW, please confirm you got vulkan + nvdec chip working in mpv. I think mpv is planning to do d3d11 and nvdec, too, but not for now.

ValZapod, 2022-09-07:
> mpv.com --vf=format=gamma=pq:primaries=bt.2020:colorlevels=full --target-colorspace-hint=yes -hwdec=nvdec --gpu-api=vulkan -vo=gpu-next Z:\Raya.and.the.Last.Dragon.2021.2160p.DSNP.WEB-DL.DDP.5.1.Atmos.DoVi.by.DVT.mp4

Did you check this commandline using this https://github.com/zhongfly/mpv-winbuild?

cronyakatsuki, 2022-12-02:
> Did you check this commandline using this https://github.com/zhongfly/mpv-winbuild?
Arch linux user here with an amd/nvidia optimus laptop. Using nvdec on a hybrid setting when running amd igpu and offloading decoding to the nvidia gpu.

That command you provided make the gpu power usage go from 1w p8 to 12w p0 with 7% gpu usage and make the colors are bad, while the nvdec-copy with gpu, not gpu-next make it go to 7w p0 with 2-3% gpu usage.

The gpu is an 1650 mobile, mpv version, and nvidia driver version is 525.60.11
```
mpv 0.35.0 Copyright © 2000-2022 mpv/MPlayer/mplayer2 projects
 built on UNKNOWN
FFmpeg library versions:
   libavutil       57.28.100
   libavcodec      59.37.100
   libavformat     59.27.100
   libswscale      6.7.100
   libavfilter     8.44.100
   libswresample   4.7.100
FFmpeg version: n5.1.2
```

ValZapod, 2022-12-03:
>make the colors are bad

Gpu produces YCbCr. RGB Colors have nothing to do with a GPU. I mean, there is Directx 11 for YCbCr to RGB too...

voidpointertonull, 2022-12-16:
This issue got quite derailed while the problem (at least the description of the symptom) appears to be quite simple.

I've been looking into power efficiency problems lately which is how I ended up here too, and generally it seems like that as observed by others too, just having an active compute context forces the GPU into the silly compute-forced P2 state which is an issue with the proprietary driver too. Forcing P2 state has a long history, it's been around for a long time to prevent the "gaming" GPUs being fully utilized by compute workloads, and while I'm not sure if the power consumption problem existed before, I suspect that it was always a problem, it just became significantly more apparent as power efficiency started getting ignored more and more, and as VRAM became significantly hungrier for power in the past few generations.

I believe that when a compute workload is detected, the GPU sticks too hard to the P2 state not just preventing higher clocks, but also apparently using an incredibly inefficient logic for lowering clocks.
On the higher-end Ampere GPUs just having a compute context with some memory allocated gets power consumption to 100+ W. There seems to be a decay though as if there's nothing done, then after several seconds the clock slowly goes down, but given that this resembles 2+ decades old software controlled power management, it's just as efficient. This means that a modern desktop system can have its idle power consumption tripled when doing very light processing on the GPU, making it better to just avoid the GPU completely, and do the light work on the CPU instead.

Sharing my findings as a conclusion with no hope of improvement any soon as it seems to be a very long lived issue not getting enough attention, while on the other hand there are frequent regressions which often go unresolved for a long time. Posts discussing these kind of problems just tend to be abandoned with the ones on the Nvidia forum often just going with no response at all while the problem is getting worse with the ever increasing static power consumption of the new GPUs.
Also, this P2 issue is just one of the many power management problems. Occasionally there's an improvement with a new driver, but then by that time multiple regressions appear. One step forward, multiple steps back.

thesword53, 2022-12-17:
And on Linux at least there is no way to disable P2 forced state. I am even wondering why Nvidia added that.

thesword53, 2023-02-05:
Since Discord added AV1 video acceleration every Nvidia GPU is forced to P2 state when the app is open:
https://www.reddit.com/r/nvidia/comments/10pgs70/discord_drops_memory_clock_by_200_mhz_on_nvidia/

Why did you not remove that from your drivers since there are topics from 2017/2018 complaining about that:
https://forums.developer.nvidia.com/t/whats-the-situation-with-vdpau-vaapi-nvdec/61031
https://forums.developer.nvidia.com/t/nvdec-forces-gpu-into-p2-cuda-state-much-higher-power-consumption-than-with-vdpau/55466

On Windows at least GPU can be on lower state and P2 state can be disabled but not on Linux so everybody using NVDEC on this project (https://github.com/elFarto/nvidia-vaapi-driver) are wasting energy and have a nerfed GPU.

ValZapod, 2023-02-05:
>acceleration every Nvidia GPU is forced to P2 state when the app is open:

That is a bug in Discord, that was fixed with nvidia control panel update today.

thesword53, 2023-02-06:
> That is a bug in Discord
That is not a Discord bug, that is an issue from Nvidia drivers: Discord uses NVENC for AV1 video encoding which trigger P2 state. The fix is just a profil for Discord that disables the forced P2 state for that app.

thesword53, 2023-03-01:
530.30.02 beta driver released and the issue is still there.

Glog78, 2023-03-28:
The difference for me are  50W  which is huge for the same task.

vdpau ->
[image]

nvenc ->
[image]

would be cool if someone has a solution cause

amrit1711, 2023-04-18:
We have filed a bug 200504689 internally for tracking purpose and we do have local repro for investigation.
Since this bug is not limited to Open GPU kernel modules, I will close this thread internally and request you to follow up on below linux developer forum threads.
https://forums.developer.nvidia.com/t/nvdec-forces-gpu-into-p2-cuda-state-much-higher-power-consumption-than-with-vdpau/55466
https://forums.developer.nvidia.com/t/whats-the-situation-with-vdpau-vaapi-nvdec/61031

closed 2023-04-18 by amrit1711

Arecsu, 2025-02-04:
Have there been any updates so far? Although people (myself included) kept reporting issues and reproducible examples at those forum links, the problem is still there. There have not been updates from Nvidia developers in the forums neither 😞. I tend to watch videos in a usual basis and it's hard to see the high power consumption they produce on Linux compared to Windows. Would be awesome if this gets fixed!

Relevant discussion related to this issue as well: https://github.com/elFarto/nvidia-vaapi-driver/issues/279

Glog78, 2025-02-09:
Arecsu -> my current workaround is to use mpv with vulkan backend and 570.86.16 << this basically on the vulkan side fixes all the issues i had with hw decoding while using reasonable power ...   
Yes 570.86.16 was required for me cause av1 on older drivers didn't work with mpv/yt-dlp (so the changes to the av1 decoder are very welcomed here)

Arecsu, 2025-02-09:
> Arecsu -> my current workaround is to use mpv with vulkan backend and 570.86.16 << this basically on the vulkan side fixes all the issues i had with hw decoding while using reasonable power ... Yes [...]

Good workaround. Although this solves video playback for local files, it won't work for browser usage unfortunately :/

Arecsu, 2025-05-27:
Alright, seems that the problem has been mostly fixed lately, at least when using Firefox + https://github.com/elFarto/nvidia-vaapi-driver

Using CachyOS, Firefox 138, Nvidia Open Kernel Module, driver 570.153.02. Everything runs between 28w and 35w in my RTX 3080. AV1, VP9, H264, 1080p-4k. CPU usage stays mostly the same so the videos seem to be running in the GPU.

When running `MPV [video]`, everything seems to be fine as well. However, with `MPV --hwdec=nvdec [video]` it is still running at ~110w for some reason.

pallaswept, 2025-11-06:
Nice of nvidia staff to stop by and keep us in the loop, thank you.

Arecsu, 2025-11-06:
To understand correctly: do we need to set CUDA_DISABLE_PERF_BOOST to 1 in order to fix this high power issue? I would guess perf boost is enabled by default. If that's the case, I find the default rather interesting. Is it because of additional negative side effects disabling it?

pallaswept, 2025-11-06:
Yes. When the application executes with this environment variable set it will disable the "perf boost" (it's not a boost when the card should be at P0, don't know why they called it this)

The P2 power state is enforced when CUDA is executing on the GPU, to provide a consistent and reliable behaviour of the card (so like, NASA don't get a math error when they're planning manned spaceflight or something). It's mostly about restricting memory clocks to avoid errors there.

The interface to NVDEC is via CUDA, so that's why it happens when using NVDEC. Since we're not sending humans to Mars or something, we can safely disable it, but the default is there for a good reason.

pallaswept, 2025-11-07:
I made this to show to a friend and thought it might be interesting here. Three tests, each 2 minutes on 100% fans to cool down, then 4 minutes playing a 4k video in youtube:

[image]

I grabbed shots for encoding in OBS and also simultaneous decode/encode, and metrics from standard tools, results reflect these.

pallaswept, 2025-12-03:
Well, dang I guess I gave nvidia too much credit, but at least the weird name makes sense. This only does half the job. It prevents the GPU from being boosted *TO* P2, but it still leaves the GPU *CAPPED* at P2.

So we still can't play games and stream them at the same time.

Nvidia whyyyyyyyyy

mtijanic, 2025-12-10:
FYI, nvbug 5719918 to add app profiles for this.  I'll try to fast track it into an upcoming release.

mtijanic, 2025-12-18:
> I'll try to fast track it into an upcoming release.

Well, the track isn't quite as fast as I'd hoped, but this should show up in the first 595.xx release.

However, you don't have to wait for it! You can put this in your `~/.nv/nvidia-application-profiles-rc` and it should work with all supported driver versions (so pre-Turing too)
```
{
    "profiles": [
        {
            "name": "CudaNoStablePerfLimit",
            "settings": ["0x166c5e", 0 ]
        }
    ],
    "rules": [
        { "pattern" : { "feature" : "procname", "matches" : "obs" },     "profile" : "CudaNoStablePerfLimit" },
        { "pattern" : { "feature" : "procname", "matches" : "Discord" }, "profile" : "CudaNoStablePerfLimit" }
    ]
}
```

mtijanic, 2025-12-18:
> Since when did app profiles work on Wayland? Fantastic.

Since probably forever, it's just the GUI that doesn't. Most app profiles in the graphics realm also have an environment variable that you can set instead, and that's probably easier. This is a CUDA knob though, and only available through the app profile.

Yes, I'm aware of how unergonomic it is that this is an app-profile knob and the other one is an environment variable. I would like to unify these so all knobs can be set through either mechanism, but that requires (cross-component/team) driver changes, and so it'll take some time.

pallaswept, 2025-12-18:
That's awesome. Your future enhancements sound really nice but I'm sure everyone will be really happy with what we've got here. I don't think it's too unergonomic. Thanks again mate you're our hero.

> Having `CudaNoStablePerfLimit` and no `CUDA_DISABLE_PERF_BOOST` on a video decoding app will just make the GPU run hotter for no benefit.

Thanks for the extra advice. I found that the design is very flexible but also safe. If I have several apps bypassing the boost or the limit, if any one app does demand the limit, it will be enforced. For my desktop the setup seemed quite simple:

- Global `export CUDA_DISABLE_PERF_BOOST=1` (in kwin_env.sh so it's visible to any GUI apps started by KDE) (this allows video chat or media playback without fan noise)
- Profiles as you demonstrated above, only for apps that might reduce P0 like in a game, to P2 (obs, vesktop, firefox, haruna, etc) (this makes games high performance even while streaming or recording)
- Launch CUDA loads like AI training etc, in containers, where `CUDA_DISABLE_PERF_BOOST` remains unset. P2 will be engaged when this is done, regardless of any of the above in use (makes sure real compute loads are reliable)

I was still playing with it when I got this notification.  Since I forgot to test everything last time 🤕 Here is a full set of test apps all at once, with vesktop and OBS and firefox and heaven all playing nicely together:

```
> nvidia-smi
Thu Dec 18 22:37:52 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 580.105.08             Driver Version: 580.105.08     CUDA Version: 13.0     |
+-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3090        On  |   00000000:07:00.0  On |                  N/A |
| 71%   72C    P0            347W /  350W |    6156MiB /  24576MiB |    100%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A            2642      G   /usr/bin/kwin_wayland                   202MiB |
|    0   N/A  N/A            2705      G   /usr/bin/Xwayland                         6MiB |
|    0   N/A  N/A            2940    C+G   /usr/bin/coolercontrol                  220MiB |
|    0   N/A  N/A            6800      G   ./heaven_x64                              4MiB |
|    0   N/A  N/A            9808      G   ./heaven_x64                              4MiB |
|    0   N/A  N/A           20475      G   ./heaven_x64                              4MiB |
|    0   N/A  N/A           26689      G   ...ount_VesktopoYivB/vesktop.bin         81MiB |
|    0   N/A  N/A           27185      G   /usr/lib64/firefox/firefox             1254MiB |
|    0   N/A  N/A           27308    C+G   ...r/lib64/firefox/browser 4 rdd        448MiB |
|    0   N/A  N/A           27863      G   ...asma-browser-integration-host          4MiB |
|    0   N/A  N/A           28189    C+G   /usr/bin/obs                            806MiB |
|    0   N/A  N/A           29613      G   ./heaven_x64                           1620MiB |
+-----------------------------------------------------------------------------------------+
```
And then I can just run something with the P2 BOOST still cleared eg
```
podman run --rm --device nvidia.com/gpu=all nvcr.io/nvidia/k8s/cuda-sample:nbody nbody -benchmark -numbodies=1214400
```
And the card comes back down to P2:

```
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3090        On  |   00000000:07:00.0  On |                  N/A |
| 76%   71C    P2            342W /  350W |    5257MiB /  24576MiB |    100%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A            2642      G   /usr/bin/kwin_wayland                   430MiB |
|    0   N/A  N/A            2705      G   /usr/bin/Xwayland                         6MiB |
[... cut ...]
|    0   N/A  N/A           27185      G   /usr/lib64/firefox/firefox              637MiB |
|    0   N/A  N/A           27308    C+G   ...r/lib64/firefox/browser 4 rdd        504MiB |
|    0   N/A  N/A           45996      G   ./heaven_x64                              4MiB |
|    0   N/A  N/A           54046      G   ./heaven_x64                           1620MiB |
|    0   N/A  N/A           54384      C   /cuda-samples/sample                    402MiB |
+-----------------------------------------------------------------------------------------+

```
It's perfect!