NVIDIA/open-gpu-kernel-modules #339: Kernel Panic on Boot on  LENOVO Legion 7 AMD (82N6)

FlorianFranzen, 2022-07-29:
### NVIDIA Open GPU Kernel Modules Version

515.57

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

NixOS Unstable 22.11.20220724.e494a90 (Raccoon)

### Kernel Release

5.18.13

### Hardware: GPU

NVIDIA Corporation GA104M [GeForce RTX 3070 Mobile / Max-Q] 

### Describe the bug

When I boot my system fails to load all kernel modules and does complete boot.

Here are the relevant entries to my dmesg log:
```
Jul 29 10:42:50 hostname kernel: nvidia: loading out-of-tree module taints kernel.
Jul 29 10:42:50 hostname kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 239
Jul 29 10:42:50 hostname kernel: NVRM cpuidInfoAMD: Unrecognized AMD processor in cpuidInfoAMD
Jul 29 10:42:50 hostname kernel: nvidia 0000:01:00.0: enabling device (0000 -> 0003)
Jul 29 10:42:50 hostname kernel: nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
Jul 29 10:42:50 hostname kernel: snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
Jul 29 10:42:50 hostname kernel: snd_hda_intel 0000:01:00.1: Disabling MSI
Jul 29 10:42:50 hostname kernel: snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
Jul 29 10:42:50 hostname kernel: snd_hda_intel 0000:05:00.6: enabling device (0000 -> 0002)
Jul 29 10:42:50 hostname kernel: input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input11
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0: bound i2c-CLSA0100:00-cs35l41-hda.0 (ops cs35l41_hda_comp_ops [snd_hda_scodec_cs35l41])
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0: bound i2c-CLSA0100:00-cs35l41-hda.1 (ops cs35l41_hda_comp_ops [snd_hda_scodec_cs35l41])
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0: autoconfig for ALC287: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0:    mono: mono_out=0x0
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0:    inputs:
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0:      Mic=0x19
Jul 29 10:42:50 hostname kernel: snd_hda_codec_realtek hdaudioC1D0:      Internal Mic=0x12
Jul 29 10:42:50 hostname kernel: NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  515.57  Release Build  (nixbld@)  Mon Jul 25 16:10:05 UTC 2022
Jul 29 10:42:50 hostname kernel: input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input12
Jul 29 10:42:50 hostname kernel: input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input13
Jul 29 10:42:50 hostname kernel: input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input14
Jul 29 10:42:50 hostname kernel: input: HDA NVidia HDMI/DP,pcm=10 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input15
Jul 29 10:42:50 hostname kernel: input: HDA NVidia HDMI/DP,pcm=11 as /devices/pci0000:00/0000:00:01.1/0000:01:00.1/sound/card0/input16
Jul 29 10:42:50 hostname kernel: nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for x86_64  515.57  Release Build  (nixbld@)  Mon Jul 25 16:08:24 UTC 2022
Jul 29 10:42:50 hostname kernel: [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
Jul 29 10:42:51 hostname kernel: BUG: unable to handle page fault for address: ffffbf2c8efc3000
Jul 29 10:42:51 hostname kernel: #PF: supervisor write access in kernel mode
Jul 29 10:42:51 hostname kernel: #PF: error_code(0x0003) - permissions violation
Jul 29 10:42:51 hostname kernel: PGD 100000067 P4D 100000067 PUD 100197067 PMD 135143067 PTE 8000000135016161
Jul 29 10:42:51 hostname kernel: Oops: 0003 [#1] PREEMPT SMP NOPTI
Jul 29 10:42:51 hostname kernel: CPU: 14 PID: 985 Comm: systemd-udevd Tainted: G           O      5.18.13 #1-NixOS
Jul 29 10:42:51 hostname kernel: Hardware name: LENOVO 82N6/LNVNB161216, BIOS GKCN53WW 02/25/2022
Jul 29 10:42:51 hostname kernel: RIP: 0010:memset_erms+0x9/0x10
Jul 29 10:42:51 hostname kernel: Code: c1 e9 03 40 0f b6 f6 48 b8 01 01 01 01 01 01 01 01 48 0f af c6 f3 48 ab 89 d1 f3 aa 4c 89 c8 c3 90 49 89 f9 40 88 f0 48 89 d1 <f3> aa 4c 89 c8 c3 90 49 89 fa 40 0f b6 ce 48 b8 01 01 01 01 01 01
Jul 29 10:42:51 hostname kernel: RSP: 0018:ffffbf2c8168b7f8 EFLAGS: 00010286
Jul 29 10:42:51 hostname kernel: RAX: 0000000000000000 RBX: ffffbf2c9067ada8 RCX: 0000000000001000
Jul 29 10:42:51 hostname kernel: RDX: 0000000000001000 RSI: 0000000000000000 RDI: ffffbf2c8efc3000
Jul 29 10:42:51 hostname kernel: input: Integrated Camera: Integrated C as /devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-3/1-3:1.0/input/input23
Jul 29 10:42:51 hostname kernel: RBP: ffffbf2c8168b8e0 R08: 0000000000000067 R09: ffffbf2c8efc3000
Jul 29 10:42:51 hostname kernel: R10: ffffbf2c9067b0e8 R11: 000000000000000d R12: ffffbf2c8e656000
Jul 29 10:42:51 hostname kernel: R13: 000000000000000d R14: ffff98fa7db44020 R15: ffff98fa7fc10020
Jul 29 10:42:51 hostname kernel: FS:  00007f0597f38d40(0000) GS:ffff990052180000(0000) knlGS:0000000000000000
[... 40 lines cut ...]
Jul 29 10:42:53 hostname kernel: RIP: 0010:memset_erms+0x9/0x10
Jul 29 10:42:53 hostname kernel: Code: c1 e9 03 40 0f b6 f6 48 b8 01 01 01 01 01 01 01 01 48 0f af c6 f3 48 ab 89 d1 f3 aa 4c 89 c8 c3 90 49 89 f9 40 88 f0 48 89 d1 <f3> aa 4c 89 c8 c3 90 49 89 fa 40 0f b6 ce 48 b8 01 01 01 01 01 01
Jul 29 10:42:53 hostname kernel: RSP: 0018:ffffbf2c8168b7f8 EFLAGS: 00010286
Jul 29 10:42:53 hostname kernel: RAX: 0000000000000000 RBX: ffffbf2c9067ada8 RCX: 0000000000001000
Jul 29 10:42:53 hostname kernel: RDX: 0000000000001000 RSI: 0000000000000000 RDI: ffffbf2c8efc3000
Jul 29 10:42:53 hostname kernel: RBP: ffffbf2c8168b8e0 R08: 0000000000000067 R09: ffffbf2c8efc3000
Jul 29 10:42:53 hostname kernel: R10: ffffbf2c9067b0e8 R11: 000000000000000d R12: ffffbf2c8e656000
Jul 29 10:42:53 hostname kernel: R13: 000000000000000d R14: ffff98fa7db44020 R15: ffff98fa7fc10020
Jul 29 10:42:53 hostname kernel: FS:  00007f0597f38d40(0000) GS:ffff990052180000(0000) knlGS:0000000000000000
Jul 29 10:42:53 hostname kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 29 10:42:53 hostname kernel: CR2: ffffbf2c8efc3000 CR3: 000000014d490000 CR4: 0000000000750ee0
Jul 29 10:42:53 hostname kernel: PKRU: 55555554
```

### To Reproduce

Try to run the open source nvidia driver via `hardware.nvidia.open = true;` on a Lenovo Legion 7 AMD from 2021.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Since I can not boot the system, I can not provide that to you, sorry.

mtijanic, 2022-07-29:
Looks to be a duplicate of #331 . Could you please check if your /lib/firmware/nvidia/515.57/gsp.bin is comressed with xz, and if so, uncompress it?

FlorianFranzen, 2022-07-29:
@mtijanic Thank you for the feedback, it indeed looks like it is a duplicate of #331. After I removed the compression, it boots successfully.

closed 2022-07-29 by FlorianFranzen