NVIDIA/open-gpu-kernel-modules #340: Nvidia DMA is not working with AMD memory encryption (SME)

andrei-dubovik, 2022-07-31:
### NVIDIA Open GPU Kernel Modules Version

515.57

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

5.17.9-arch1-1

### Hardware: GPU

NVIDIA GeForce GTX 1660 SUPER

### Describe the bug

I have a Ryzen 3660, which supports SME. I've tried activating it with the `mem_encrypt=on` kernel boot option. The system loads fine and dmesg reports that SME is active:

```
[    0.122077] AMD Memory Encryption Features active: SME
```

I also have IOMMU enabled in the UEFI, and dmesg reports that it's working. However, if I try to start the Xorg server, it fails with the following error:

```
[    21.357] (II) NVIDIA: Reserving 24576.00 MB of virtual memory for indirect memory
[    21.357] (II) NVIDIA:     access.
[    24.364] (EE) NVIDIA(GPU-0): Failed to initialize DMA.
[    24.365] (EE) NVIDIA(0): Failed to allocate push buffer
```

I've tried this with Linux version 5.17.9-arch1-1 or 5.18.14-arch1-1, and with Nvidia 515.57. For Nvidia I've tried both closed sourced and open sources drivers, as well as non-dkms and dkms versions. I've also tried recompiling the open source drivers. I always get the same error.

I've found online this earlier post on the gentoo bugzilla: https://bugs.gentoo.org/652408. As far as I read from that post, the driver should work with memory encryption enabled. I've also glanced at the source code, and there is a small code section with `#ifdef` on `CONFIG_AMD_MEM_ENCRYPT`, which suggests the code at least intends to work with AMD SME. That code also checks `NV_PGPROT_DECRYPTED_PRESENT`, which seems undefined by default. I've tried defining it and recompiling the code, but I still get the same error.

### To Reproduce

Enable SME on a Ryzon processor using the `mem_encrypt=on` kernel boot option, start the Xorg sever.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

(Not including the output, because it collects *everything* about my computer and I do not feel comfortable providing all that identifying information. I'm happy to provide a specific part, if required. Though I do not expect the issue to be specific to my computer, seems like a general issue of whether the recent driver works with the recent kernel when AMD SME is enabled.)

Avamander, 2022-08-29:
Can confirm on Ubuntu 22.04 with driver version 515.65.01.

simcop2387, 2022-09-05:
I've recently run into this with debian 12 (bookworm) and 515.65.01 also.

MabaKalox, 2022-12-04:
Is it even supposed to work? Direct memory access suppose no CPU action in this process, but only the CPU knows encryption keys.

andrei-dubovik, 2022-12-05:
@MabaKalox I do not know exactly but I guess the driver can mark video memory pages as unencrypted. With SME, it's up to the OS to mark pages for encryption, so a driver can change that for specific pages. Also, as I've mentioned in the report, there is an `#ifdef` on `CONFIG_AMD_MEM_ENCRYPT` in the code, so at least something is supposed to happen, it seems.