NVIDIA/open-gpu-kernel-modules #349: Brightness control not working in open drivers

149segolte, 2022-08-14:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Does this happen with the proprietary driver (of the same version) as well?

No

### Operating System and Version

Arch Linux

### Kernel Release

5.19.1-arch2-1

### Hardware: GPU

NVIDIA GeForce RTX 2070 with Max-Q Design

### Describe the bug

I tried switching to Nvidia open drivers in Arch Linux. This involved replacing the 'nvidia' package with the 'nvidia-open' package. After reboot, everything seems to be working fine but the brightness is fixed at max.

I checked /sys/class/backlight/, with proprietary drivers I had a 'nvidia_0' directory, now it has been replaced by the 'acpi_video0'. I have tried all the tools I could find including 'light', 'brightnessctl', 'wluma' (for Wayland), 'xbacklight' (for Xorg) etc. None seem to work. Manually modifying the brightness file in the 'acpi_video0' directory does not work. Even with sudo the file does not retain the changes.

### To Reproduce

- Remove 'nvidia' package
```bash
pacman -Rns nvidia
```
- Install 'nvidia-open' package
```bash
pacman -S nvidia-open
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9332928/nvidia-bug-report.log.gz)

aritger, 2022-08-15:
Thanks for the report.  I think this has the same root cause as https://github.com/NVIDIA/open-gpu-kernel-modules/issues/290: the plumbing between nvidia.ko and gsp.bin is not currently wired up for brightness controls.  We're working on that for a future release.

As an experiment, can you test with the proprietary kernel modules, but force them to use GSP offload? Set the nvidia.ko kernel module parameter NVreg_EnableGpuFirmware=1:

http://us.download.nvidia.com/XFree86/Linux-x86_64/515.48.07/README/gsp.html

The open kernel modules necessarily use GSP offload, but the proprietary kernel modules (for GeForce) don't yet use GSP offload by default. This experiment will verify that the problem is in GSP offload.

149segolte, 2022-08-16:
Yes, I can confirm that setting that parameter makes the proprietary drivers too also lose the brightness control. I guess that resolves this issue being part of #290. I will close this for now.
Something that I would like to know is, can the GSP offload be disabled in the open source drivers?

closed 2022-08-16 by 149segolte

aritger, 2022-08-16:
Thanks for conducting that test.

And, no, the open source kernel modules cannot operate without GSP offload.

We'll try to get brightness + GSP offload fixed soon.