NVIDIA/open-gpu-kernel-modules #351: 6.0 rc1 Kernel, kernel-dkms does fail

ptr1337, 2022-08-15:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Operating System and Version

CachyOS / Archlinux based

### Kernel Release

6.0rc1

### Build Command

Building kernel-dkms for 6.0rc1 kernel

### Terminal output/Build Log

```
  LD [M]  /var/lib/dkms/nvidia/515.65.01/build/nvidia.o
ld -r -o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-interface.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-pci.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-dmabuf.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-acpi.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-cray.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-dma.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-i2c.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-mmap.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-p2p.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-pat.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-procfs.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-procfs-utils.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-usermap.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-vm.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-vtophys.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/os-interface.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/os-mlock.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/os-pci.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/os-registry.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/os-usermap.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-modeset-interface.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-pci-table.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-kthread-q.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-memdbg.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-ibmnpu.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-report-err.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-rsync.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-msi.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-caps.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv-frontend.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nv_uvm_interface.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nvlink_linux.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/nvlink_caps.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/linux_nvswitch.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/procfs_nvswitch.o /var/lib/dkms/nvidia/515.65.01/build/nvidia/i2c_nvswitch.o
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘cursor_plane_req_config_update’:
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c:88:32: warning: unused variable ‘nv_drm_plane_state’ [-Wunused-variable]
   88 |     struct nv_drm_plane_state *nv_drm_plane_state =
      |                                ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c:87:27: warning: unused variable ‘nv_dev’ [-Wunused-variable]
   87 |     struct nv_drm_device *nv_dev = to_nv_device(plane->dev);
      |                           ^~~~~~
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘plane_req_config_update’:
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c:189:9: warning: unused variable ‘ret’ [-Wunused-variable]
  189 |     int ret = 0;
      |         ^~~
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘nv_drm_plane_atomic_set_property’:
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c:504:32: warning: unused variable ‘nv_drm_plane_state’ [-Wunused-variable]
  504 |     struct nv_drm_plane_state *nv_drm_plane_state =
      |                                ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c: In function ‘nv_drm_enumerate_crtcs_and_planes’:
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-crtc.c:1148:13: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 1148 |             struct drm_plane *overlay_plane =
      |             ^~~~~~
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-modeset.c: In function ‘__will_generate_flip_event’:
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-modeset.c:98:10: warning: unused variable ‘overlay_event’ [-Wunused-variable]
   98 |     bool overlay_event = false;
      |          ^~~~~~~~~~~~~
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-modeset.c:97:10: warning: unused variable ‘primary_event’ [-Wunused-variable]
   97 |     bool primary_event = false;
      |          ^~~~~~~~~~~~~
/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm/nvidia-drm-modeset.c:96:23: warning: unused variable ‘primary_plane’ [-Wunused-variable]
   96 |     struct drm_plane *primary_plane = crtc->primary;
      |                       ^~~~~~~~~~~~~
ld -r -o /var/lib/dkms/nvidia/515.65.01/build/nvidia-modeset/nv-modeset-interface.o /var/lib/dkms/nvidia/515.65.01/build/nvidia-modeset/nvidia-modeset-linux.o /var/lib/dkms/nvidia/515.65.01/build/nvidia-modeset/nv-kthread-q.o
  LD [M]  /var/lib/dkms/nvidia/515.65.01/build/nvidia-modeset.o
  LD [M]  /var/lib/dkms/nvidia/515.65.01/build/nvidia-peermem.o
  LD [M]  /var/lib/dkms/nvidia/515.65.01/build/nvidia-drm.o
  LD [M]  /var/lib/dkms/nvidia/515.65.01/build/nvidia-uvm.o
  MODPOST /var/lib/dkms/nvidia/515.65.01/build/Module.symvers
ERROR: modpost: "drm_framebuffer_put" [/var/lib/dkms/nvidia/515.65.01/build/nvidia-drm.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:128: /var/lib/dkms/nvidia/515.65.01/build/Module.symvers] Error 1
make[1]: *** [Makefile:1771: modules] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.0.0-rc1-1-cachyos-rc/build'
make: *** [Makefile:82: modules] Error 2
```

ptr1337, 2022-08-15:
Dont saw the PR before, probably fixes this.
https://github.com/NVIDIA/open-gpu-kernel-modules/pull/350

mtijanic, 2022-08-18:
Already fixed and tracked internally as bug 3694026, should be in the next release. Can use #350 as a workaround in the meantime.

closed 2022-08-18 by ptr1337

sjmuniz, 2022-08-20:
Confirming fix worked.

palexdev, 2023-01-01:
Has this been fixed in the proprietary version?

ptr1337, 2023-01-01:
> Has this been fixed in the proprietary version?

Yes, its fixed