NVIDIA/open-gpu-kernel-modules #352: Mkinitcpio can't load nvidia module

yenfeng-shetiko, 2022-08-16:
### NVIDIA Open GPU Kernel Modules Version

515.65.01-4

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Description:	Arch Linux

### Kernel Release

Linux myarch 5.19.1-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Thu, 11 Aug 2022 15:22:00 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 Ti Laptop GPU (UUID: GPU-7376a1a3-f476-a257-ed5a-1c4f869d71c9)

### Describe the bug

When I add nvidia nvidia_modeset nvidia_uvm nvidia_drm to /etc/mkinitcpio.conf and run 'sudo mkinitcpio -P', I got an error: "==> ERROR: module not found: `nvidia'==> ERROR: module not found: `nvidia_modeset'==> ERROR: module not found: `nvidia_uvm'==> ERROR: module not found: `nvidia_drm'" 

### To Reproduce

sudo vim /etc/mkinitcpio.conf
(add nvidia nvidia_modeset nvidia_uvm nvidia_drm to line 'MODULES=()')
sudo mkinitcpio -P

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9350261/nvidia-bug-report.log.gz)

mtijanic, 2022-08-18:
Hey there! The only issue I see in the logs is
```
 NVRM objClInitPcieChipset: *** Chipset Setup Function Error!
```
While it is weird, because your chipset is [certainly supported](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/src/nvidia/arch/nvalloc/common/inc/nvcst.h#L183-L184), it should not be fatal in itself.

Can you try loading the module with [`RmMsg=":"`](https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/197) and running `nvidia-bug-report.sh` again?

yenfeng-shetiko, 2022-08-19:
[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9383193/nvidia-bug-report.log.gz)

spartus434, 2022-09-21:
Out of curiosity do you have the nvidia-dkms package installed?  If not you may want to install it and let it make the kernel modules for the gpu.  I don't believe the Zen kernel has a specific Nvidia package on Arch so you need to use dkms or install with the Nvidia installer running on the Zen kernel.

yenfeng-shetiko, 2023-01-10:
> Out of curiosity do you have the nvidia-dkms package installed? If not you may want to install it and let it make the kernel modules for the gpu. I don't believe the Zen kernel has a specific Nvidia [...]

Sorry, I did install the wrong package, thanks for pointing it out.

closed 2023-01-10 by yenfeng-shetiko

AKHIL-DyC, 2024-01-20:
> I am also affected by this issue
> 
> ```
> ❯ sudo mkinitcpio -P
> ==> Building image from preset: /etc/mkinitcpio.d/linux-amd.preset: 'default'
>   -> -k /boot/vmlinuz-linux-amd -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-amd.img
> ==> Starting build: 6.0.2-AMD
>   -> Running build hook: [systemd]
>   -> Running build hook: [autodetect]
>   -> Running build hook: [modconf]
>   -> Running build hook: [block]
>   -> Running build hook: [keyboard]
>   -> Running build hook: [sd-encrypt]
>   -> Running build hook: [fsck]
>   -> Running build hook: [filesystems]
> ==> Generating module dependencies
> ==> Creating lzop-compressed initcpio image: /boot/initramfs-linux-amd.img
> ==> Image generation successful
> ==> Building image from preset: /etc/mkinitcpio.d/linux.preset: 'default'
>   -> -k /boot/vmlinuz-linux -c /etc/mkinitcpio.conf -g /boot/initramfs-linux.img
> ==> Starting build: 6.0.2-arch1-1
>   -> Running build hook: [systemd]
>   -> Running build hook: [autodetect]
>   -> Running build hook: [modconf]
>   -> Running build hook: [block]
> ==> WARNING: Possibly missing firmware for module: xhci_pci
>   -> Running build hook: [keyboard]
>   -> Running build hook: [sd-encrypt]
> ==> WARNING: Possibly missing firmware for module: qat_4xxx
>   -> Running build hook: [fsck]
>   -> Running build hook: [filesystems]
> ==> ERROR: module not found: `nvidia'
> ==> ERROR: module not found: `nvidia_uvm'
> ==> Generating module dependencies
> ==> Creating zstd-compressed initcpio image: /boot/initramfs-linux.img
> ==> WARNING: errors were encountered during the build. The image may not be complete.
> ==> Building image from preset: /etc/mkinitcpio.d/linux.preset: 'fallback'
>   -> -k /boot/vmlinuz-linux -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-fallback.img -S autodetect
> ==> Starting build: 6.0.2-arch1-1
>   -> Running build hook: [systemd]
>   -> Running build hook: [modconf]
>   -> Running build hook: [block]
> ==> WARNING: Possibly missing firmware for module: aic94xx
> ==> WARNING: Possibly missing firmware for module: bfa
> ==> WARNING: Possibly missing firmware for module: qed
> ==> WARNING: Possibly missing firmware for module: qla1280
> ==> WARNING: Possibly missing firmware for module: qla2xxx
> ==> WARNING: Possibly missing firmware for module: wd719x
> ==> WARNING: Possibly missing firmware for module: xhci_pci
>   -> Running build hook: [keyboard]
>   -> Running build hook: [sd-encrypt]
> ==> WARNING: Possibly missing firmware for module: qat_4xxx
>   -> Running build hook: [fsck]
>   -> Running build hook: [filesystems]
> ==> ERROR: module not found: `nvidia'
> ==> ERROR: module not found: `nvidia_uvm'
> ==> Generating module dependencies
> ==> Creating zstd-compressed initcpio image: /boot/initramfs-linux-fallback.img
> ==> WARNING: errors were encountered during the build. The image may not be complete.
> ❯ yay -Ss nvidia-dkms
> aur/nvidia-dkms-6 515.65.01-2 (+0 0.00)
>     NVIDIA drivers - module sources
> extra/nvidia-dkms 520.56.06-1 (41.8 MiB 67.2 MiB) (Installed)
>     NVIDIA drivers - module sources
> ```

did you find solution?

vb2007, 2024-01-24:
Yeah may I ask what package exactly? I've installed `linux-headers` and `nvidia` (for laptop gpu), and the install failed with the same issues both when running `sudo pacman -S nvidia-utils` (reinstall) and `sudo mkinitcpio -P`. Maybe I'm using wrong headers?

yenfeng-shetiko, 2024-01-29:
> Yeah may I ask what package exactly? I've installed `linux-headers` and `nvidia` (for laptop gpu), and the install failed with the same issues both when running `sudo pacman -S nvidia-utils` (reinstall) and `sudo mkinitcpio -P`. Maybe I'm using wrong headers?

Sorry, I have now removed these parameters from /etc/mkinitcpio.conf