NVIDIA/open-gpu-kernel-modules #354: mpv/libplacebo VK_ERROR_INITIALIZATION_FAILED with nvidia vulkan and wayland

CoelacanthusHex, 2022-08-19:
### NVIDIA Open GPU Kernel Modules Version

515.65.01-5

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

5.19.2-zen1-1-zen

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 Laptop GPU (UUID: GPU-7ef8ea0d-17bb-7e77-0311-e1e259621525)

### Describe the bug

when use mpv with nvidia, vulkan and wayland, mpv will play audio but don't display video window.
in log, we can see 'vk->CreateSwapchainKHR(...): VK_ERROR_INITIALIZATION_FAILED'

### To Reproduce

`mpv --no-config --gpu-api=vulkan` with Nvidia GPU in Wayland
if Nvidia is not default, please specify Nvidia GPU with `--vulkan-device=`

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9383373/nvidia-bug-report.log.gz)

### More Info

https://github.com/mpv-player/mpv/issues/10552

CoelacanthusHex, 2022-08-19:
Since it only happened on Wayland and https://github.com/NVIDIA/open-gpu-kernel-modules/issues/317 too.
Maybe wayland part of vulkan driver has problems.

suppai, 2022-09-19:
According to https://forums.developer.nvidia.com/t/negativo17-wayland-vkcreateswapchainkhr-returns-vk-error-initialization-failed/215593:

```
[ekurzinger](https://forums.developer.nvidia.com/u/ekurzinger)
[Employee](https://forums.developer.nvidia.com/g/Employee)
[Jun 8](https://forums.developer.nvidia.com/t/negativo17-wayland-vkcreateswapchainkhr-returns-vk-error-initialization-failed/215593/4)

There is an ABI incompatibility between the 515 driver and previous versions of our egl-wayland library which might be the cause of this. Updating to the latest version, 1.1.10, should resolve the issue if that is the case.

Going forward this should never happen again as we’ve eliminated the driver <=> egl-wayland ABI which the Vulkan WSI depended on. However, doing so required one final breaking change, unfortunately.
```

CoelacanthusHex, 2022-09-19:
I still meet this crash although I installed egl-wayland 1.1.11 on Arch Linux

wael444, 2022-12-10:
can reproduce this issue as well

meisme-dev, 2023-02-11:
I can reproduce this if using NVIDIA with my AMD gpu enabled, but if I disable my AMD gpu (from the BIOS settings) I cannot reproduce this.

1-minute-to-midnight, 2024-06-26:
No updates on this issue? I have this problem as well. Can't use mpv in KDE wayland

realyukii, 2025-04-02:
I have a similar problem  but the graphic card I use is amd and I use Xorg for display server:
```
$  mpv Suzume\ but\ is\ it\ okay\ if\ it\'s\ lofi？\ \[T2EkSBnaj-o\].mp4
[playlistmanager] auto showamount: 19
● Video  --vid=1  (vp9 1080x1080 25 fps) [default]
● Audio  --aid=1  (aac 2ch 44100 Hz 128 kbps) [default]
[vo/gpu/libplacebo] EnumeratePhysicalDevices(inst, &num, NULL): VK_ERROR_INITIALIZATION_FAILED (../src/vulkan/context.c:985)
[vo/gpu/libplacebo] Found no suitable device, giving up.
[vo/gpu/libplacebo] Failed initializing vulkan device
AO: [pipewire] 44100Hz stereo 2ch floatp
VO: [gpu] 1080x1080 yuv420p
(Paused) AV: 00:00:43 / 00:04:09 (18%) A-V:  0.000
```

the error recently printed shortly after I perform full upgrade on the system, not sure which updates affect the changes