NVIDIA/open-gpu-kernel-modules #357: PASSTHROUGH GPU in VMM，256MB memory is used but no process run in GPU

SYAAAA, 2022-08-30:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Description:   Alibaba Cloud Linux (Aliyun Linux) release 2.1903 LTS (Hunting Beagle)

### Kernel Release

4.19.91

### Hardware: GPU

GPU 0: Tesla T4 (UUID: GPU-b358a171-e37d-5881-19cc-a6bf09d949cf)

### Describe the bug

GPU is passthrough in VMM. NVIDIA-smi show 256MB is used, but no process is running on GPU.
[image]
[image]

### To Reproduce

No useful method can reproduce the bud, becasue of the VMM is privated.
But we want to know the mechanism of the nvidia-smi showing memory used.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9449447/nvidia-bug-report.log.gz)

mtijanic, 2022-08-30:
Hey there! The GSP microprocessor carves out 256MB of VRAM for its own use. This bit will include any ucode running on any microcontroller (gsp.bin but also things like PMU, etc), the heap for the GSP, the communication infra between GSP and CPU and a few other things. I believe this is what nvidia-smi is reporting.

Even with the proprietary driver, the T4 is using the GSP offload model by default, as shown here: https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/src/nvidia/src/kernel/gpu_mgr/gpu_mgr.c#L735

You can try the proprietary driver with the regkey of `NVreg_EnableGpuFirmware=0` to see if the memory usage goes down.
To verify whether GSP offload is in use, you can run `nvidia-smi -q | grep GSP`. If the version is "N/A", then it is not in use.

SYAAAA, 2022-08-30:
> Hey there! The GSP microprocessor carves out 256MB of VRAM for its own use. This bit will include any ucode running on any microcontroller (gsp.bin but also things like PMU, etc), the heap for the [...]

Thanks your reply !!!!!

Your answer give me a hint which the 256MB memory is used by gpu itself. 
So when I installed the NVIDIA GPU Driver completely, not pre-compiled it and mount the ko* so* bin* in the container such as the NVIDIA Docker，the memory is ok, and no more 256MB is used.

Thanks very much!!!!  The issue cost my three days, now the bug is located.

[image]

closed 2022-08-30 by SYAAAA