NVIDIA/open-gpu-kernel-modules #358: Computer crashes when playing games (screen freeze and reboot)

noahboegli, 2022-09-01:
### NVIDIA Open GPU Kernel Modules Version

515.65.01-9

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

5.19.5-arch1-1

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3070 Ti (UUID: GPU-10dd02b9-41d1-a7bb-1fa1-982e157cd5e5)

### Describe the bug

When playing games (such as X-Plane, Prepar3D or even Minecraft), the display freezes for a short period of time and the computer reboots.

The issue also happens on Windows with the latest driver release.

I am unable to reproduce the crash when stress-testing the card. It only happens when playing games.

There is are no overheating issues (room is 25, CPU 60 and GPU less than 70).

The issue also happened with my GTX 1070, which I replaced because I thought it was dying of age.

There are no logs entries in the sysylog.

### To Reproduce

- For the crash to happen quickly, I generally load Minecraft with Shaders and X-Plane at the same time
- Let the program run for a while
- The crash happens most of the time within one hour, but it can take longer, and sometimes, although rarely, it does not happen

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9473687/nvidia-bug-report.log.gz)

closed 2022-09-03 by noahboegli

noahboegli, 2022-09-08:
@balenamiaa I closed it because it was not linked to the GPU (hardware or software) . I should have added the explanation but I closed it on the phone and forgot to come back to add a note, so here it is: 

It was in fact due to an issue with the PCI-E lane speed of my motherboard (an ASUS ROG STRIX Z490-E GAMING) . The latest firmware version fixed it. 

I've bought a new GPU, a new PSU but all of the issues have been solved with a (free) BIOS flash that took 10 minutes.

It's amazing because I've googled and read literally hundreds of pages on the issue over 2 weeks, only to find one that mentioned the motherboard firmware right after I updated the firmware out of pure despair.

balenamiaa, 2022-09-08:
Thanks, your explanation is very useful!