NVIDIA/open-gpu-kernel-modules #359: High refresh rate (>60 Hz) not effectively applied on multi-monitor setup

Crystal4276, 2022-09-02:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Does this happen with the proprietary driver (of the same version) as well?

I cannot test this

### Operating System and Version

Arch / Cinnamon 5.4.9 DE

### Kernel Release

5.19.6-arch1-1 

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 2080 (UUID: GPU-b0867e4b-eda0-c18b-65ec-6155a2c0cc55)

### Describe the bug

High refresh rate not applied  for high refresh rate monitors connected with additionnal 60 Hz monitors

### To Reproduce

-- use Xorg
-- plug at least two monitors: one with refresh rate >60Hz, the others monitior with a **max** refresh rate of 60Hz
-- Set the high refresh monitor to a value > 60 Hz, for instance 100Hz and other "old" monitors to their max refresh of 60 Hz
-- Measure the effective/real refresh rate on the high refresh rate monitor, (eg. https://www.displayhz.com/) 

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9479410/nvidia-bug-report.log)

### More Info

To note : if the 60 Hz monitors are physically disconnected from the GPU, the high refresh rate monitor will then work properly at its max refresh rate.

ValZapod, 2022-09-07:
That was a big problem in windows too. When they fixed it, 60 fps monitor still was having some strange problems with VRR.

TimSousa1, 2022-11-01:
Same here on linuxmint.

sTiKyt, 2023-01-30:
Still an issue, on both open and proprietary driver
60Hz monitor off:
[image]

60Hz monitor on:
[image]

Any chance of getting this fixed anytime soon, or any workarounds?

This seems like a big problem for anyone using multiple screens while gaming/working on videos/etc.

I don't need my second screen to be more than 60 Hz, i use it for system information and utilities, but i expect my 165 Hz screen to perform at 165 Hz in games

gilvbp, 2023-03-24:
Same problem with new driver release  530.41.03 (not beta)

sTiKyt, 2023-04-08:
> @sTiKyt Same problem with the last driver release 530.41.03 (released yesterday) Related to #471

Well, for now just add `/etc/environment` setting as described above

sTiKyt, 2023-05-31:
> I'm glad you found the __GL_SYNC_DISPLAY_DEVICE environment variable. I believe there is equivalent control within nvidia-settings, too.
> 
> Given multiple monitors, as part of the same desktop, with different refresh rates, is the sync-to-vblank behavior well defined? We can only really sync to one monitor, and then presentation on all other monitors will tear.
> 
> Maybe the driver could default to whichever monitor has the highest refresh rate, but I worry changing default behavior would break existing configurations for other users.
> 
> Is there particular behavior that you think the driver should adopt as the default?

I didn't have any visual issues since setting that variable, but I'm not sure about any corner cases this env var might produce

TimSousa1, 2023-05-31:
Screen tearing seems to be the major issue when setting that variable

sTiKyt, 2023-07-02:
> Screen tearing seems to be the major issue when setting that variable

Didn't have any throughout my experience, may differ for others tho

sTiKyt, 2024-08-11:
> I didn't have any visual issues since setting that variable, but I'm not sure about any corner cases this env var might produce

Just a little update, been using that variable for over a year now, and i'm still not experiencing any issues

sTiKyt, 2024-08-11:
> I'm glad you found the __GL_SYNC_DISPLAY_DEVICE environment variable. I believe there is equivalent control within nvidia-settings, too.
>
> Given multiple monitors, as part of the same desktop, with [...]

Would be really nice if you could clarify where is forementioned control located in nvidia-settings, because i coudn't find anything like that and i think it would be useful to many users finding this issue later

sTiKyt, 2026-01-11:
Just an update after using this for couple years, works great, no issues, pain in the ass to find this issue every time i want to reinstall tho

Binary-Eater, 2026-02-02:
I think @aritger has given a great response to why there may not be a "fix" that can be approached without causing regressions for other users.

> Given multiple monitors, as part of the same desktop, with different refresh rates, is the sync-to-vblank behavior well defined? We can only really sync to one monitor, and then presentation on all [...]

So a "fix" here might be a "regression" for another user. That's partially why I encouraged discussions on the sync-to-vblank behavior in my comment before closing out the issue. I have some internal thoughts on the matter I am testing out and seeing are viable options.