NVIDIA/open-gpu-kernel-modules #360: RTX3070 fails to resume from suspend. Fail to determine display capabilities. Failed to tear down display.

Brixus3, 2022-09-05:
### NVIDIA Open GPU Kernel Modules Version

nvidia-open-dkms 515.65.01-10

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

Linux archlinux 5.19.6-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 31 Aug 2022 22:09:42 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3070 Laptop GPU (UUID: GPU-cc7080b8-a8bb-7670-3c92-ae4535025206)

### Describe the bug

On wake-up from sleep system usually freezes until hard reset then boots normally. Sometimes I’ll get a black screen with a single cursor then freeze or the login screen will come up after a delay but the system runs remarkably slow. Journal shows 
```
archlinux kernel: nvidia-modeset: ERROR: GPU:0: Failed to determine display capabilities
archlinux kernel: nvidia-modeset: ERROR: GPU:0: Failed to tear down Disp
archlinux kernel: nvidia-modeset: ERROR: GPU:0: Failed to determine display capabilities
archlinux kernel: nvidia-modeset: ERROR: GPU:0: Failed to tear down Disp
archlinux kernel: nvidia-modeset: ERROR: GPU:0: Failed detecting connected display devices
archlinux kernel: nvidia-modeset: ERROR: GPU:0: Failed detecting connected display devices
archlinux kernel: nvidia-modeset: ERROR: GPU:0: Failed detecting connected display devices
```
I also posted this error a while back at [https://forums.developer.nvidia.com/t/nvidia-515-48-07-fails-to-detect-laptop-display-on-wake-up-from-sleep-on-arch-linux/217657](url) with no response. 

After wake-up nvidia-smi:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.65.01    Driver Version: 515.65.01    CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  ERR!                On   | 00000000:01:00.0 N/A |                  N/A |
|ERR!  ERR! ERR!     N/A /  N/A |     17MiB /  8192MiB |     N/A      Default |
|                               |                      |                 ERR! |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
```

nvidia-persistence daemon:
```
* nvidia-persistenced.service - NVIDIA Persistence Daemon
     Loaded: loaded (/usr/lib/systemd/system/nvidia-persistenced.service; enabled; preset: disabled)
     Active: active (running) since Sun 2022-09-04 19:17:24 CDT; 38min ago
    Process: 732 ExecStart=/usr/bin/nvidia-persistenced --user nvidia-persistenced (code=exited, status=0>
   Main PID: 734 (nvidia-persiste)
      Tasks: 1 (limit: 76928)
     Memory: 800.0K
        CPU: 5ms
     CGroup: /system.slice/nvidia-persistenced.service
             `-734 /usr/bin/nvidia-persistenced --user nvidia-persistenced

Sep 04 19:17:24 archlinux systemd[1]: Starting NVIDIA Persistence Daemon...
Sep 04 19:17:24 archlinux nvidia-persistenced[734]: Started (734)
Sep 04 19:17:24 archlinux systemd[1]: Started NVIDIA Persistence Daemon.
```
lsmod grep | nvidia:
```
nvidia_drm             73728  12
nvidia_uvm           2801664  4
nvidia_modeset       1359872  8 nvidia_drm
nvidia               5287936  1196 nvidia_uvm,nvidia_modeset
```

I've done quite a bit of googling trying to figure this one out. I can't seem to find anyone else with this issue. I'm hoping someone here has some insight.

### To Reproduce

Arch linux on Oryx Pro Laptop Intel with Nvidia 3070 hybrid. 
Install open-dkms driver 
close lid or enter sleep
try to wakeup

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9486003/nvidia-bug-report.log.gz)

### More Info

Successful resume from suspend.

nejcb1, 2022-09-08:
I also have problems with Geforce RTX 3070 Ti Laptop GPU and 5.19.7-arch1. I use nvidia standard driver, and I have problems with:
- slow start of GUI (waiting 1minute for login screen to show up).
- settings menu wont work when clicking (GNOME)
- Chrome browser - very slow start
- second monitor is not working

If I unistall nvidia driver and use noveau it boots normally. Not quite same issue as yours, but problems with similar hardware configuration.

2tefan, 2022-12-11:
I had the same problem  but the newest proprietary driver seems to have fixed it (`nvidia-dkms 525.60.11-1`).

Unfortunately, the open-source driver is still broken on my machine (`nvidia-open-dkms 525.60.11-3`).

Also, make sure you enable these two SystemD modules:

`nvidia-suspend.service`
`nvidia-hibernate.service`

I hope this also works for you.

### Current version of Nvidia tools
```
nvidia-dkms 525.60.11-1
nvidia-settings 525.60.11-2
nvidia-utils 525.60.11-1
opencl-nvidia 525.60.11-1
```

### My setup

- OS: Arch Linux
- Kernel: `6.0.12-zen1-1-zen`
- Graphics Card: RTX 2080