NVIDIA/open-gpu-kernel-modules #363: Reproducible Xid 31 error when switching between Linux Console and X.org running Google Chrome

birdie-github, 2022-09-05:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora release 36 (Thirty Six)

### Kernel Release

5.19.7

### Hardware: GPU

NVIDIA GeForce GTX 1660 Ti

### Describe the bug

CPU: Ryzen 5800X
MB: X570 chipset
GPU: GTX 1660 Ti

OS: Fedora 36/Linux 5.19.7/X.org/NVIDIA 515.65.01/XFCE 4 **without** compositing

Steps to reproduce:

    Open Google Chrome
    Switch to Linux console (Ctrl + Alt + F2)
    Return to X.org (Alt + F1/F7)

Result:

NVRM: GPU at PCI:0000:08:00: GPU-UID
NVRM: Xid (PCI:0000:08:00): 31, pid=169800, name=chrome, Ch 00000023, intr 00000000. MMU Fault: ENGINE CE0 HUBCLIENT_HSCE0 faulted @ 0x0_0abe0000. Fault is of type FAULT_PTE ACCESS_TYPE_VIRT_READ
NVRM: Xid (PCI:0000:08:00): 31, pid=174421, name=chrome, Ch 00000023, intr 00000000. MMU Fault: ENGINE CE0 HUBCLIENT_HSCE0 faulted @ 0x0_0abe0000. Fault is of type FAULT_PTE ACCESS_TYPE_VIRT_READ

Nothing breaks as a result of this error but I’d love not to get it anyways.

P.S. Mozilla Firefox is also running. Not sure if they are both needed to trigger the bug. Google Chrome must absolutely be running.

### To Reproduce

See above

### Bug Incidence

Always

### nvidia-bug-report.log.gz

N/A

### More Info

This is a bug report against the proprietary driver (since reports at https://forums.developer.nvidia.com/c/gpu-graphics/linux/148 are largely ignored).

amrit1711, 2022-09-06:
@birdie-github 
I have filed a bug 3785853 internally for tracking purpose.
I will close this thread as it is not specific to open GPU kernel modules.
However, I will keep updating on devtalk forum for further updates.

closed 2022-09-06 by amrit1711

Walkability, 2025-05-18:
I know this is super old, but I get the XID: 31 error fairly often on a GTK 1650 ti. Happens when playing halo on steam and the game freezes and I have to quit and restart but the system doesn't crash. I'm on arch linux with the zen kernel using closed source nvidia beta drivers.

pinkycollie, 2025-05-18:
The XID:31 error on your Arch Linux system with the NVIDIA 1650 Ti is a common issue. Since you're using the Zen kernel and beta drivers, driver/kernel incompatibility is a likely culprit.
I recommend trying the LTS kernel first for improved stability. If that doesn't resolve the problem, consider switching to the latest stable NVIDIA driver. Disabling runtime power management is another helpful step. Finally, monitoring your GPU temperature and power usage might reveal underlying hardware issues.
Let me know if you have any questions.