NVIDIA/open-gpu-kernel-modules #365: Monitor goes to standby in VT console

dimich-dmb, 2022-09-06:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Arch Linux

### Kernel Release

Linux dimich 5.19.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 05 Sep 2022 18:09:09 +0000 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce GTX 1660 SUPER (UUID: GPU-08a3b790-0cae-501b-dcc1-d73f40e5af63)

### Describe the bug

Monitor goes to standby (with "No video input", "Enter standby mode" messages) after switching to VT console.
If Xorg server autostart is disabled monitor goes to standby immediately after nvidia modules loaded during boot.
When nvidia modules aren't loaded no problems with console observed.

### To Reproduce

* Boot system with Xorg server autostart.
* Switch to VT console (Ctrl+Alt+F1)

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9494014/nvidia-bug-report.log.gz)

### More Info

With proprietary driver of the same version monitor goes to standby only when Xorg server is not running and nvidia_drm module is loaded with `modeset=1`. When Xorg server is running monitor doesn't go to standby but VT console is broken (not updated after scrolling).

aaronp24, 2022-09-07:
It looks like your system is running in legacy VGA text mode, which isn't currently supported in the open kernel modules. Please try using EFI boot or a vesafb framebuffer console; the open kernel modules rely on nvidia-modeset to restore console modes for VT switches, but nvidia-modeset isn't able to set legacy VGA modes.

For the proprietary driver, there's a known hardware issue in Turing GPUs where VGA text scrolling doesn't work properly while an nvidia-modeset client such as Xorg or nvidia-drm with the `modeset=1` parameter is loaded. Unfortunately, it's not a problem that we can work around. Using a framebuffer console is your best bet if you need these scenarios to work properly.

dimich-dmb, 2022-09-07:
> Please try using EFI boot or a vesafb framebuffer console

I have no UEFI boot in my system. VESA framebuffer console works extremely slow (ypan and mtrr:3 doesn't help).
Thank you for your answer. Well, next time before hardware upgrade i should look for videocard with good drivers support.