NVIDIA/open-gpu-kernel-modules #367: Tearing on high GPU load or in game Wayland and Xorg

MiMillieuh, 2022-09-11:
### NVIDIA Open GPU Kernel Modules Version

515.65.01

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora release 36 (Thirty Six)

### Kernel Release

Linux fedora 5.19.7-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Sep 5 14:50:12 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 Laptop GPU (UUID: GPU-9cd4d6b7-b1a2-2074-3e94-2ea82c4396b6)

### Describe the bug

Since I’m using an optimus laptop, I encounter a huge tearing issue when my GPU hit the 70 degrees.

I already had this issue in the past with a desktop and it just needed to disable an option in the Nvidia panel but on my laptop, there is almost no option for my GPU.

Here is my configuration :

GPU : RTX 3050 laptop
CPU : Ryzen 7 6800HS creator edition
RAM : 32GB ram DDR5

Fedora 36
Proprietary driver from RPMfusion
Kernel 5.19.7

A video that describe that bug : [NvidiaTearingBug](https://www.youtube.com/watch?v=24dpIEntPLU)

Does anyone knows how to fix that kind of tearing without the option from the Nvidia panel since they aren’t avalible to me on both Xorg and Wayland ?

Thanks.

my panel on Wayland AND Xorg are the same and lack a lot of option compared to my GTX 1070 desktop GPU : 
[image]
[image]
[image]
[image]

### To Reproduce

just launch a game or anything that has high GPU demand and wait for it to heat and goes to tearing

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9542211/nvidia-bug-report.log.gz)

### More Info

The laptop manufacture is LENOVO
and the model is the Yoga slim 7 pro X

changing performance mode in bios doesn't affect this issue

MiMillieuh, 2022-09-11:
Update on this : it's related to thermal throttling as my GPU critical temp is 75 degrees
But it isn't supposed to heat that much and also the fan aren't spinning that much according to the noise.

I guess there are some issue in the way it's managed.

NBZion, 2022-09-30:
You're reporting this with proprietary drivers, this repo is for the open sourced version of the drivers

MiMillieuh, 2022-09-30:
> You're reporting this with proprietary drivers, this repo is for the open sourced version of the drivers

I have the same issue on both drivers...

I've installed the open and the closed and the issue still the same.

kindagoose, 2022-10-13:
on X11 I have tearing in any vulkan/dxvk game even tho __GL_SHOW_GRAPHICS_OSD=1 shows that vsync is on. This issue existed even since at least 510. Both open and closed driver regardless if compositor is present or not. Don't recommend to use composition pipeline, it causes terrible stutters in any game. 

kernel 5.19
kernel 6.0
nvidia 510
nvidia 515
nvidia 520
3060ti

MiMillieuh, 2022-10-14:
> on wayland I only get stutters, but on X11 I have tearing in any vulkan/dxvk game even tho __GL_SHOW_GRAPHICS_OSD=1 shows that vsync is on. This issue exists since at least 510. Both open and closed driver regardless if compositor is present or not. Please don't recommend to use composition pipeline, it causes terrible stutters in any game. 
> 
> kernel 5.19
> kernel 6.0
> nvidia 510
> nvidia 515
> nvidia 520
> 3060ti 

Well even without it's awful...

Although disableing V-sync in DXVK conf get rides of the sutters but only if you have more FPS than your monitor refresh rate...

kindagoose, 2022-10-14:
I forgot to mention that I have 60hz screen.
Locking fps with strangle and using decimals like 59.97, 59.94 etc to try to prevent vsync buffer piling up doesn't help either.
Same behavior in Dark Souls 3, Valheim, Elden Ring. Two of these games are hard locked at 60fps, and tearing still happens in X11.

kindagoose, 2022-11-15:
https://github.com/Plagman/gamescope/issues/495

kindagoose, 2022-11-15:
I think this issues are connected:
https://github.com/Plagman/gamescope/issues/495

kakra, 2022-11-15:
> Please don't recommend to use composition pipeline, it causes terrible stutters in any game.

For me, the composition pipeline actually works properly since I'm running on a 3060 12GB (previously 1660 Ti). But only, if I disable vsync in games. If vsync is enabled, it terribly stutters. If vsync is disabled, and games render below 60fps, it also stutters (but a lot less). If vsync is disabled, and games render above 60fps, it's perfectly smooth (except some microstutters every now and then from background system activity probably, depends on the game). In both latter cases, there's no tearing. Instead of vsync on, I'm now using fps limiting if the games support it. Usually it works fine when limiting to 90fps or higher, limiting below 90fps still introduces some barely noticeable stutter. For some reason, there's no stutter if the game itself falls below 90fps but stays above 60fps.

With the composition pipeline off, there's chaos all over the place: stutters, tearing, out-of-order presentation...

kindagoose, 2022-11-15:
I noticed one more thing. So when the game is in its native fullscreen mode and vsync is ON, then it works (No FCP). But tearing can appear sometimes, for instance it just appeared as soon as I started recording with OBS and went away when I closed OBS, opening OBS again did bring tearing back. (Tearing doesn't appear on recorded gameplay)

MiMillieuh, 2022-11-18:
Even tought I managed to fix a huge part of that issue using some wired tricks... Optimus has a lot of issues and NVIDIA SHOULD REALLY pay some attention to it. like the constant usage of the GPU even when not in use with a software... (3W used for nothing)

Also Support Wayland, Xorg is good for legacy hardware but recent hardware play more nicely with Wayland usually.