NVIDIA/open-gpu-kernel-modules #377: Nvidia 515.76 dkms is failing with llvm15

ptr1337, 2022-09-21:
### NVIDIA Open GPU Kernel Modules Version

515.76

### Operating System and Version

CachyOS (Archlinux based)

### Kernel Release

5.19.10 / 6.0rc6

### Build Command

Building dkms

### Terminal output/Build Log

[nvidia-dkms.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9615414/nvidia-dkms.log)

### More Info

LLVM 15 was two weeks ago released. Most distro's still dont updated it, anyways this issue exist since llvm 15 got developed.
Following commit does break nvidia-dkms:
https://github.com/llvm/llvm-project/commit/7d644e1215b376ec5e915df9ea2eeb56e2d94626

Issue at llvm:
https://github.com/llvm/llvm-project/issues/55820

Patching the nvidia-dkms with this patch helps to build it correctly:

```
From 6007f992254bb4d1933df02d0bf528f4164336df Mon Sep 17 00:00:00 2001
From: Peter Jung <admin@ptr1337.dev>
Date: Tue, 2 Aug 2022 16:51:12 +0200
Subject: [PATCH] fix-llvm-15

---
 kernel-open/Kbuild      | 4 ++--
 kernel-open/conftest.sh | 2 +-
 2 files changed, 3 insertions(+), 3 deletions(-)

diff --git a/kernel/Kbuild b/kernel/Kbuild
index d8d376a..a341332 100644
--- a/kernel/Kbuild
+++ b/kernel/Kbuild
@@ -70,7 +70,7 @@ $(foreach _module, $(NV_KERNEL_MODULES), \
 
 EXTRA_CFLAGS += -I$(src)/common/inc
 EXTRA_CFLAGS += -I$(src)
-EXTRA_CFLAGS += -Wall -MD $(DEFINES) $(INCLUDES) -Wno-cast-qual -Wno-error -Wno-format-extra-args
+EXTRA_CFLAGS += -Wall -MD $(DEFINES) $(INCLUDES) -Wno-cast-qual -Wno-implicit-function-declaration -Wno-strict-prototypes -Wno-error -Wno-format-extra-args
 EXTRA_CFLAGS += -D__KERNEL__ -DMODULE -DNVRM
 EXTRA_CFLAGS += -DNV_VERSION_STRING=\"515.57\"
 
@@ -98,7 +98,7 @@ ifeq ($(ARCH),arm64)
 endif
 
 ifeq ($(ARCH),x86_64)
- EXTRA_CFLAGS += -mno-red-zone -mcmodel=kernel
+ EXTRA_CFLAGS += -mno-red-zone -mcmodel=kernel -Wno-implicit-function-declaration -Wno-strict-prototypes
 endif
 
 ifeq ($(ARCH),powerpc)
diff --git a/kernel/conftest.sh b/kernel/conftest.sh
index dae07d2..6d5f231 100755
--- a/kernel/conftest.sh
+++ b/kernel/conftest.sh
@@ -185,7 +185,7 @@ test_headers() {
 }
 
 build_cflags() {
-    BASE_CFLAGS="-O2 -D__KERNEL__ \
+    BASE_CFLAGS="-O2 -Wno-implicit-function-declaration -Wno-strict-prototypes -D__KERNEL__ \
 -DKBUILD_BASENAME=\"#conftest$$\" -DKBUILD_MODNAME=\"#conftest$$\" \
 -nostdinc -isystem $ISYSTEM"
 
-- 
2.37.1

```

Since some upstream distro's want to update their llvm, it would be a idea to include this patch or provide another solution, so they can provide the updated llvm

mtijanic, 2022-09-21:
Hi there, thanks for the detailed report and the patch. For my own understanding, is the change to `Kbuild` files actually necessary?

From what I can tell, the issue is just in the conftest which now fails and doesn't generate the correct definitions. And I'm not sure the conftest patch is correct, as disabling the warning could make it pass when it should otherwise fail, and again generate incorrect definitions for the driver to consume.

We'll have to take a closer look at this. Thanks!

ptr1337, 2022-09-21:
> Hi there, thanks for the detailed report and the patch. For my own understanding, is the change to `Kbuild` files actually necessary?
>
> From what I can tell, the issue is just in the conftest which [...]

It is actually not the best solution, but the only solution how I get it to work right now. I think the llvm commit, I linked above should help to adjust the conftest.sh to the changes, which llvm15 did bring.

https://github.com/llvm/llvm-project/commit/7d644e1215b376ec5e915df9ea2eeb56e2d94626

ptr1337, 2022-09-21:
@mtijanic 

I just tested it, with patching only the conftest.sh, and it seems that it is required to patch also the Kbuild right now.
It had way less errors, as without patching anything, but still fails. Log is attached

[nvidia-dkms.log](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9615697/nvidia-dkms.log)

mtijanic, 2022-09-21:
Thanks for confirming and thanks for that log. This is something to be fixed in the driver code.

There would probably be many more such errors if you were building nv-kernel.o from source too.

mtijanic, 2022-09-22:
For future reference, this is tracked internally as bug 3803365

ionenwks, 2022-09-26:
On gentoo, I just added the missing (void) to fix this rather than do -Wno-strict. Patch could probably be taken as-is.

Note some other things like nvidia-settings, nvidia-modprobe, etc... will also break when clang16 comes around because -Werror=strict-prototypes will be a hard default.

https://github.com/gentoo/gentoo/blob/master/x11-drivers/nvidia-drivers/files/nvidia-drivers-470.141.03-clang15.patch (applies to all drivers from 470 to 515 including open ones)
https://github.com/gentoo/gentoo/blob/master/x11-drivers/nvidia-drivers/files/nvidia-drivers-390.154-clang15.patch (legacy)
https://github.com/gentoo/gentoo/blob/master/x11-drivers/nvidia-drivers/files/nvidia-drivers-390.154-clang15-x86.patch (uvm-less version)
https://github.com/gentoo/gentoo/blob/master/x11-drivers/nvidia-drivers/files/nvidia-extras-470.141.03-clang16.patch
https://github.com/gentoo/gentoo/blob/master/x11-drivers/nvidia-drivers/files/nvidia-extras-390.154-clang16.patch (other stuff)

ptr1337, 2022-09-29:
It seems like the error will be downgraded for now. So it is probably working with the nvidia drivers, as soon 15.0.2 has been officially released.
Here the commit:
https://github.com/llvm/llvm-project/commit/c0141f3c300fbc002cf79404fa0b82b4cb1191df#diff-ec770381d76c859f5f572db789175fe44410a72608f58ad5dbb14335ba56eb97

amrit1711, 2022-12-27:
We were able to root cause the issue, shall inform once driver is released publicly.

ptr1337, 2023-01-05:
Seems still not to be fixed in 525.78.01 closed drivers...

xuanruiqi, 2023-01-19:
Is there an ETA? This (-Wstrict-prototypes) is killing me now.

closed 2023-01-19 by ptr1337