NVIDIA/open-gpu-kernel-modules #382: High power consumption in idle / unable to turn GPU off

reibengu, 2022-10-04:
### NVIDIA Open GPU Kernel Modules Version

515.76

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Fedora 36

### Kernel Release

5.19.12-200.fc36.x86_64

### Hardware: GPU

NVIDIA GeForce RTX 3050 Ti Laptop GPU

### Describe the bug

If Nvidia drivers are loaded `powertop` reports ~14W usage on idle while `nvidia-smi ` shows "No running processes found"

If I blacklist nouveau, nvidia and remove the device using udev rules `powertop` reports ~7W usage on idle.

Note: From time to time it can boot with Nvidia drivers loaded and show ~7W usage on idle, this is rare but possible.

### To Reproduce

Install nvidia drivers using this official fedora guide:
https://rpmfusion.org/Howto/NVIDIA

### Bug Incidence

Always

### nvidia-bug-report.log.gz

N/A

vans163, 2022-10-05:
Same problem https://forums.developer.nvidia.com/t/idle-power-usage-stuck-at-10-20watts-after-running-an-app/217520

niv, 2022-10-27:
Can you please attach the output of `nvidia-smi -q`, before/after running a graphics app, as well?

Specifically looking for info on the Performance/boost state flag, but all the power info can be useful.

scaledteam, 2022-12-16:
I have similar problem in 525.60.11 driver,  same videocard (RTX 3050 Ti laptop). I have to use this option to be able to use this driver.
```
options nvidia NVreg_OpenRmEnableUnsupportedGpus=1
```
I tried to follow the instructions in documentation, but no success.
https://download.nvidia.com/XFree86/Linux-x86_64/525.60.11/README/dynamicpowermanagement.html

Also, my card doesn't show status and everything. And it also ignores temperature limit which was set by manufacturer, and boosts to higher clocks compared to proprietary driver. 75-80 degrees was max temperature in proprietary driver, in open-source one it's is 85-88 degrees.
```sh
$ cat /proc/driver/nvidia/gpus/0000\:01\:00.0/power
Runtime D3 status:          ?
Video Memory:               ?

GPU Hardware Support:
 Video Memory Self Refresh: ?
 Video Memory Off:          ?

$ nvidia-smi -q

==============NVSMI LOG==============

Timestamp                                 : Fri Dec 16 13:27:00 2022
Driver Version                            : 525.60.11
CUDA Version                              : 12.0

Attached GPUs                             : 1
GPU 00000000:01:00.0
    Product Name                          : NVIDIA GeForce RTX 3050 Ti Laptop GPU
    Product Brand                         : GeForce
    Product Architecture                  : Ampere
    Display Mode                          : Disabled
    Display Active                        : Disabled
    Persistence Mode                      : Disabled
    MIG Mode
        Current                           : N/A
        Pending                           : N/A
    Accounting Mode                       : Disabled
    Accounting Mode Buffer Size           : 4000
    Driver Model
        Current                           : N/A
        Pending                           : N/A
    Serial Number                         : N/A
    GPU UUID                              : GPU-ce39c7d8-66ee-e4ef-cf66-4400546963ed
    Minor Number                          : 0
    VBIOS Version                         : 94.07.3B.00.B4
    MultiGPU Board                        : No
    Board ID                              : 0x100
    Board Part Number                     : N/A
    GPU Part Number                       : 25A0-775-A1
    Module ID                             : 0
    Inforom Version
        Image Version                     : G001.0000.03.03
        OEM Object                        : 2.0
        ECC Object                        : N/A
        Power Management Object           : N/A
    GPU Operation Mode
        Current                           : N/A
        Pending                           : N/A
    GSP Firmware Version                  : 525.60.11
    GPU Virtualization Mode
        Virtualization Mode               : None
        Host VGPU Mode                    : N/A
    IBMNPU
        Relaxed Ordering Mode             : N/A
    PCI
        Bus                               : 0x01
        Device                            : 0x00
        Domain                            : 0x0000
        Device Id                         : 0x25A010DE
        Bus Id                            : 00000000:01:00.0
[... 101 lines cut ...]
    Deferred Clocks
        Memory                            : N/A
    Max Clocks
        Graphics                          : 2100 MHz
        SM                                : 2100 MHz
        Memory                            : 5501 MHz
        Video                             : 1950 MHz
    Max Customer Boost Clocks
        Graphics                          : N/A
    Clock Policy
        Auto Boost                        : N/A
        Auto Boost Default                : N/A
    Voltage
        Graphics                          : 631.250 mV
    Fabric
        State                             : N/A
        Status                            : N/A
    Processes
        GPU instance ID                   : N/A
        Compute instance ID               : N/A
        Process ID                        : 1297
            Type                          : G
            Name                          : /usr/bin/gnome-shell
            Used GPU Memory               : 1 MiB
 ```

scaledteam, 2022-12-16:
> What makes you believe that this is a problem
Proprietary driver says `GPU Max Operating Temp: 75 C` . But open-source one says `87 C`, which is strange.

> However it's not clear how is your problem related to the issue posted here
My GPU doesn't go to sleep in idle, but in proprietary driver it working as expected. Probably i should create separate post, because OP says that for him it happens with proprietary driver too, but for me it's only happens with open-source kernel modules.

voidpointertonull, 2022-12-16:
> Proprietary driver says `GPU Max Operating Temp: 75 C`

That's surely a better description of the problem, but then if the proprietary driver goes up to 80 *C anyways, then even if it shows the limit in the VBIOS, it's suspicious as earlier mentioned that it doesn't really obey that, while the open source driver may have the wrong limit, but it successfully maxes out the likely not exactly great laptop cooler.

> My GPU doesn't go to sleep in idle, but in proprietary driver it working as expected.

What indicates the sleeping status you are looking for? It surely sounds like you have a different issue, but I'm curious that aside from apparently power status reporting not being supported, what makes you believe it's not idling as well as it can.
I'm not familiar with the laptop exclusive features as poor power management and poor support of standards just made me disable Nvidia GPUs on work laptops, but your 7 W idle power consumption matches the OP's desired target, and your GPU seems to be P8 state, so I'd expect this to be as good as it gets, but then you imply that the proprietary driver gets better results.

scaledteam, 2022-12-16:
> What indicates the sleeping status you are looking for?
In proprietary driver GPU transparently goes to sleep after roughly 20 seconds of inactivity, and after that, when i trying to access GPU, program (nvidia-smi, blender, cuda programs) freezes for a second and then starting working normally. I can't even track that, because every request to GPU wakes it up with notisable 1 second freeze. Theoretically it possible to track via this command, or with powertop, but i not tested that, battery life was good in proprietary drivers.
```
cat /sys/bus/pci/devices/0000\:01\:00.0/power_state
```

voidpointertonull, 2022-12-16:
I wouldn't expect the GPU to go to such a deep sleep with a desktop environment running. If you also have an iGPU, then I wouldn't be surprised if you were missing graphics duties being handed over to that so the dGPU could get powered off, but that's a completely different can of worms I'd only dream of opening after dGPU-only operation is already painless.

While I recommend going through with the earlier plan of opening a new issue for your problem, it's unlikely to have a resolution any soon. Aside from the obvious "OpenRmEnableUnsupportedGpus" part making your setup officially not supported, power efficiency appears to be one of the lowest priority issues lately.

scaledteam, 2022-12-16:
It's sad, because this open kernel modules is only driver that works with my external RTX 3090 TI via thunderbolt. =( Proprietary driver doesn't work with it at all, because of RmInitAdapter failed error, but open-source one works fine. But it lacks some features for my integrated 3050 TI, like sleep support and power saving features.

Is there an analogue of the OpenRmEnableUnsupportedGpus parameter for a proprietary driver? It might solve all my problems.

voidpointertonull, 2022-12-16:
I believe you have a misunderstanding about the purpose of OpenRmEnableUnsupportedGpus. This driver is officially meant to be used only on data center GPUs even if there are way more features than what's needed for that, so you are forced to use a flag as an explicit step of pretty much stating that you understand that even if your setup mostly works, it's officially not supported.

We got really far from OP's problem though, so please really consider opening separate issues, you apparently have even two, the lack of (deep) sleep support which appears to be an issue with this driver, and the problem with the lack of support for our Thunderbolt setup which is not really supposed to be here, but then the Github issue tracker is so much better than the official forum with intentionally broken native browser search support, you could try being cheeky here, just provide a really detailed report then.

To stay somewhat on topic though, if you do have an iGPU and you don't need the extra compute power on the go, then I'd disable the built-in Nvidia dGPU which should give you better power efficiency, and would use the Thunderbolt setup with this driver when stationary.

ejrydhfs, 2024-10-24:
i used to have this issue on fedora linux 40 running kernel version 6.10.10. i have upgraded to kernel version 6.11.4 and the issue has gone away but nvidia-smi still reports it is consuming power, yet powertop shows it is not consuming any power.

ejrydhfs, 2024-10-24:
i tested it using driver version 560.35.03

ejrydhfs, 2024-10-24:
i have also tested the same driver verson on Ubuntu 24.04 running kernel version 6.8 and the issue doesn't seem to be present in Ubuntu either. My graphics card is an RTX 3060 mobile