NVIDIA/open-gpu-kernel-modules #384: No graphics in aarch64 (SolidRun Honeycomb LX2K) Ubuntu 22.04 using NVidia drivers

carlosedp, 2022-10-08:
### NVIDIA Open GPU Kernel Modules Version

515.76

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Ubuntu 22.04.1 LTS

### Kernel Release

5.15.0-50-generic #56-Ubuntu SMP Tue Sep 20 13:23:50 UTC 2022 aarch64

### Hardware: GPU

GPU 0: NVIDIA GeForce GTX 1660 Ti

### Describe the bug

I have a SolidRun Honeycomb LX2K ARM64 system and recently installed a GTX1660Ti GPU on it. I get no video on Desktop (monitor stays blank as there is no signal to it).

The initial install of Ubuntu 22.04 worked fine due to it's use of Noveau drivers. I then installed the NVidia drivers (stock) and also open drivers and lost the GUI (Gnome).

The install steps I took are listed below:

- Downloaded 515.76 drivers from <https://www.nvidia.com/Download/driverResults.aspx/193096/en-us/>
- Installed it without Kernel modules (`sudo ./NVIDIA-Linux-aarch64-515.76.run --no-kernel-modules`)
- Cloned this repo and built according to instructions (make modules/make modules_install). Got [this error](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/271) in the modules_install but the drivers were installed. Build the driver with the patch I got after talking to Solidrun support <https://gist.github.com/jnettlet/226c0787ae9744082493accc512a9a6b>.

- Blacklisted Noveau (otherwise Ubuntu would boot with it)
- On first boot, I saw these errors in Kernel logs:

```
10.0.20.203: kern: warning: [2022-09-02T19:03:16.922855838Z]: NVRM: Open nvidia.ko is only ready for use on Data Center GPUs.
10.0.20.203: kern: warning: [2022-09-02T19:03:16.923935838Z]: NVRM: To force use of Open nvidia.ko on other GPUs, see the
10.0.20.203: kern: warning: [2022-09-02T19:03:16.924966838Z]: NVRM: 'OpenRmEnableUnsupportedGpus' kernel module parameter described
10.0.20.203: kern: warning: [2022-09-02T19:03:16.926121838Z]: NVRM: in the README.
10.0.20.203: kern: warning: [2022-09-02T19:03:17.235131838Z]: NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x63:0x0:1905)
10.0.20.203: kern: warning: [2022-09-02T19:03:17.236812838Z]: NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number
```

Also nvidia-smi showed a message that no devices were found.

- Created `/etc/modprobe.d/nvidia.conf` and added `options nvidia NVreg_OpenRmEnableUnsupportedGpus=1` to it. Run `sudo update-initramfs -u` and rebooted.

- Linux detected the GPU bus still no graphics (nothing, my monitor stays in standby as before like there is no signal to it). nvidia-smi shows:

[image]

I could successfully run some ML workloads like Pytorch but no graphics. I tried each and every setting to X11 I found with no success until now.

Sometimes I see these messages in Kernel logs:

```
[ 1741.048380] [drm:nv_drm_atomic_commit [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00040100] Flip event timeout on head 0
[ 1741.252831] nvidia-modeset: ERROR: GPU:0: Timed out waiting for core channel idle.
```

### To Reproduce

Steps in the description above.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9739728/nvidia-bug-report.log.gz)

carlosedp, 2022-10-08:
The PR https://github.com/NVIDIA/open-gpu-kernel-modules/pull/324/files solves the `modules_install` certificate error. Problem persists tho.

improvisor40, 2022-12-09:
Hello, did you get that gpu working, plus can you pls give a brief info on what gpus work.