NVIDIA/open-gpu-kernel-modules #387: Unable to determine the device handle for GPU

DanielCoelho112, 2022-10-11:
### NVIDIA Open GPU Kernel Modules Version

520.61.05

### Does this happen with the proprietary driver (of the same version) as well?

Yes

### Operating System and Version

Open Suse Leap 15.2

### Kernel Release

Linux deeplar 5.3.18-lp152.106-default #1 SMP Mon Nov 22 08:38:17 UTC 2021 (52078fe) x86_64 x86_64 x86_64 GNU/Linux

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 2080 Ti (UUID: GPU-67a571b1-79ee-1016-8ea3-c1fff3f1c1d2) GPU 1: NVIDIA GeForce RTX 2080 Ti (UUID: GPU-3e34e62f-d88e-f448-415e-be2cbf2c6178) GPU 2: NVIDIA GeForce RTX 2080 Ti (UUID: GPU-2d540dd6-583b-cf11-80dd-cc28edc1dd62) GPU 3: NVIDIA GeForce RTX 2080 Ti (UUID: GPU-80be5c9a-eb7d-2388-ad7c-7d3ee8f87235)

### Describe the bug

Sometime when using the GPU, the GPU 1 falls out of the bus, and the command nvidia-smi outputs the folllowing error: Unable to determine the device handle for GPU 0000:0B:00.0: Unknown Error.

I checked the temperatures and power usage of GPU 1 before the failure and didn't get anything out of the usual. 

### To Reproduce

Run GPU intensive process and then the error will arise.

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/NVIDIA/open-gpu-kernel-modules/files/9757795/nvidia-bug-report.log.gz)

DanielCoelho112, 2022-10-19:
Fixed by reseating GPU.

closed 2022-10-19 by DanielCoelho112

saiful9379, 2023-05-18:
Hi @DanielCoelho112 , I am fetching same problem GPU 3090 Ti ,  Driver Version: 470.182.03  CUDA Version: 11.4 .
When i train Model the GPU show that  "Unable to determine the device handle for GPU 000:08.0.0": unknown Error.  so this problem raise after 7 or 9, 10 epoch. if you write your approach here it will helpful for me.

DanielCoelho112, 2023-05-19:
Hi @saiful9379,

To fix my issue, I had to physically detach the defective GPU, eliminate the accumulated dust, and reinsert it into its original position.

Lucca-cherries, 2023-11-01:
> Hi @saiful9379,
> 
> To fix my issue, I had to physically detach the defective GPU, eliminate the accumulated dust, and reinsert it into its original position.

Hi @DanielCoelho112 , I am exactly having the same issue when I train a model. It seems that a certain conda environment and a certain dataset will always lead to this problem. Do you have some ideas? I tried rebooting the server and the error was gone, but next time I run some other processes, it will happen again.

lmsasu, 2023-11-14:
In my case the GPU overheated. I could see its temperature raising to 97 C via nvitop (pip install nvitop), then the error message occured.

Lucca-cherries, 2023-11-14:
> In my case the GPU overheated. I could see its temperature raising to 97 C via nvitop (pip install nvitop), then the error message occurred.

That's true. My gpu 0 seems to have problem because I can see it tends to get an extremely high temperature, with fan MAX, through nvitop. So I suppose there is something wrong with the hardware setting.

lmsasu, 2023-11-14:
> That's true. My gpu 0 seems to have problem because I can see it tends to get an extremely high temperature, with fan MAX, through nvitop. So I suppose there is something wrong with the hardware [...]

For sure it will happen for a GPU with passive cooling, as it happened in my case. It is a Tesla T4 which works in a blade server with airflow, but not in a workstation case.